evcc-io/evcc #446: Integration von SolarView für Überschussermittlung

HT83, 2020-11-17:
Hallo zusammen :-)

ich würde gerne die Daten aus SolarView nehmen um damit eine Wallbox zu steuern, damit diese nur bei PV-Überschuss lädt.
Funktionen von SolarView: https://www.solarview.info/solarview_linux.aspx

Viele werten damit auch den Eigenverbrauch aus. Über die D0-Schnittstelle wird der Zweirichtungszähler permanent ausgewertet. Der D0-Lesekopf hängt dann über USB am RPi.

[image]

Daten aus SolarView können mit MQTT oder über TCP abefragt werden.
Doku: https://www.solarview.info/solarview-fb_Installieren.pdf
MQTT Publishing Seite 46 oder https://www.photovoltaikforum.com/thread/138419-solarview-f%C3%BCr-linux-unterst%C3%BCtzt-nun-mqtt
TCP Server Seite 45 

Es macht wohl Sinn die Daten über MQTT abzurufen, oder?^

Zugriff auf einen SolarView Datenlogger kann ich bereit stellen!

Bei Rückfragen stehe ich gerne zu Verfügung.
Dankeee :-)

andig, 2020-11-17:
Also:
- unter https://github.com/andig/evcc#meter und https://github.com/andig/evcc#plugins findest Du die Doku zu Zählern und Plugins
- unter https://github.com/andig/evcc-config#meters gibt es jede Menge Konfigurationsbeispiele

Damit kannst Du loslegen ;)

HT83, 2020-11-17:
Ich bin ja recht fit in solchen Dingen, aber MQTT ist für mich noch ein Fremdwort. Werde ich mich mal durchkämpfen müssen oder mir noch weitere Mitstreiter suchen ;)

andig, 2020-11-17:
Du musst ja nicht mqtt nehmen wenn du auch http zur Verfügung hast!

closed 2020-11-21 by andig