evcc-io/evcc #447: Audi API not working

AdrianW84, 2020-11-17:
trying to get information from myaudi.com about charging state

evcc.yaml:

vehicles:
- name: audi
  type: audi
  title: E-Tron
  capacity: 86 # kWh
  user: myaudi.com useraccount
  password: password
  vin: WAUZZZGExxxxxxxxxxx
  cache: 5m

log:
[main  ] INFO 2020/11/17 16:01:09 evcc 0.31 (feb564d)
[main  ] INFO 2020/11/17 16:01:09 using config file /root/evcc/evcc.yaml
[main  ] INFO 2020/11/17 16:01:09 listening at 0.0.0.0:7070
[lp-1  ] INFO 2020/11/17 16:01:10 charger enabled
[lp-1  ] DEBUG 2020/11/17 16:01:10 set charge current: 6A
[site  ] INFO 2020/11/17 16:01:10 site config:
[site  ] INFO 2020/11/17 16:01:10   grid ✓
[site  ] INFO 2020/11/17 16:01:10   pv ✓
[site  ] INFO 2020/11/17 16:01:10   battery —
[site  ] INFO 2020/11/17 16:01:10   grid meter config:
[site  ] INFO 2020/11/17 16:01:10     power ✓
[site  ] INFO 2020/11/17 16:01:10     energy —
[site  ] INFO 2020/11/17 16:01:10     currents ✓
[lp-1  ] INFO 2020/11/17 16:01:10 loadpoint 1 config:
[lp-1  ] INFO 2020/11/17 16:01:10   vehicle —
[lp-1  ] INFO 2020/11/17 16:01:10   charge ✓
[lp-1  ] INFO 2020/11/17 16:01:10   charge meter config:
[site  ] INFO 2020/11/17 16:01:10     power ✓
[site  ] INFO 2020/11/17 16:01:10     energy —
[site  ] INFO 2020/11/17 16:01:10     currents ✓
[lp-1  ] INFO 2020/11/17 16:01:10   charger config:
[lp-1  ] INFO 2020/11/17 16:01:10     power —
[lp-1  ] INFO 2020/11/17 16:01:10     energy —
[lp-1  ] INFO 2020/11/17 16:01:10     currents —
[lp-1  ] INFO 2020/11/17 16:01:10     timer —
[lp-1  ] INFO 2020/11/17 16:01:10   mode: pv
[site  ] DEBUG 2020/11/17 16:01:10 ----
[site  ] DEBUG 2020/11/17 16:01:10 pv power: -5800W
[site  ] DEBUG 2020/11/17 16:01:10 grid power: -290W
[site  ] DEBUG 2020/11/17 16:01:10 site power: -290W
[lp-1  ] DEBUG 2020/11/17 16:01:10 charge power: 2800W
[lp-1  ] DEBUG 2020/11/17 16:01:10 charger status: C
[lp-1  ] INFO 2020/11/17 16:01:10 start charging ->
[lp-1  ] DEBUG 2020/11/17 16:01:10 detected phases: 3p [1 2 3]A
[lp-1  ] DEBUG 2020/11/17 16:01:10 max charge current: 6A = 6A + 0A (-290W @ 3p)
[lp-1  ] DEBUG 2020/11/17 16:01:10 pv timer reset
[lp-1  ] DEBUG 2020/11/17 16:01:10 target charge current: 6A

EVCC is showing "—" as vehicle and 0kWh as capacity.
Any ideas? any better log options than "debug" as shown above?

Thanks alot

andig, 2020-11-17:
Versuch bitte mal:

    evcc vehicle -c <conf>

AdrianW84, 2020-11-17:
./evcc vehicle -c evcc.yaml
[main  ] INFO 2020/11/17 16:54:46 evcc 0.31 (feb564d)
[main  ] INFO 2020/11/17 16:54:46 using config file evcc.yaml
State: 76%
Charge status: A
Finish time: 2020-12-30 05:41:00 +0000 UTC
Climate active: false
Outside temp: 10.6°C
Target temp: 20.5°C

hier klappt es,
und der Fehler ist oben ersichtlich
[lp-1 ] INFO 2020/11/17 16:01:10 vehicle —

unter Loadpoint was in der Dummyconfig kein Vehicle angegeben

Vielen Dank

closed 2020-11-17 by AdrianW84