evcc-io/evcc #448: Hausverbrauch in der UI anzeigen

mpucherSmIT, 2020-11-17:
**Describe the solution you'd like**
Irgendwo zwischen Erzeugung und Einspeisung/Batteriladung noch den Hausverbrauch anzeigen (mit nem schönen Häuschen Symbol)

**Describe alternatives you've considered**
Kann man sich rausrechnen, aber wäre halt schöner, wenn es direkt auch angezeigt wird.

andig, 2020-11-17:
Der Hausverbrauch ist für EV/PV Laden nicht wirklich relevant- im Sinne eines cleanen UIs würde ich das tendenziell eher nicht machen? /cc @TheRealKerni

mpucherSmIT, 2020-11-17:
Darüber kann man anderer Meinung sein. Im Sinne von "Energieflüsse" kann das schon relevant (oder zumindest interessant) sein

cathiele, 2020-11-17:
+1 für cleanes UI. Hausverbrauch habe ich noch nicht vermisst in der UI

TheRealKerni, 2020-11-17:
Bei EVCC geht es um das PV geführte Laden. Und das möchten wir so einfach wie möglich gestalten. Hierbei stören zu viele Zahlen und Daten eher. Wir zeigen jetzt schon zu viele Daten an.

Es ist verständlich, dass der Technik-interessierte da evtl. etwas mehr sehen möchte. Aber dafür gibt es auch schon andere Lösungen die das ermöglichen.

andig, 2020-11-17:
Also fürs Erste: won't fix. Wir können uns aber später überlegen, ob es dafür mal einen Expertenmodus geben sollte oder sogar die aktuelle Darstellung zum Expertenmodus wird.

closed 2020-11-17 by andig

userjp123, 2020-11-17:
mM: Unbedingt so lassen. 
Ich habe noch nichtmal die PV-Erzeugung in der Anzeige. Die habe ich an anderer Stelle mit HausVerbrauch etc. 
Bei evcc reicht mir Bezug/Einspeisung.

mpucherSmIT, 2020-11-17:
Ich kann das mit dem clean UI Argument verstehen. Irgendwann kommt einer, und will seine Kesseltemperatur angezeigt bekommen ;-). Der Hausverbrauch ist mMn aber schon eine relevante Größe für die Überschusssteuerung.

Vielleicht kann man das UI ja mal später über die Konfig steuern? Dann könnte sich das jeder so anpassen, wie er es braucht.

userjp123, 2020-11-17:
Der Hausverbrauch ist eben **keine** relevante Größe, sondern nur der Überschuss/Bezug.