evcc-io/evcc #451: Konfiguration Home Assistant Plugin

atze09, 2020-11-18:
Config with data direct from home assistant.
Ich Habe das Plugin erfolgreich auf der aktuellsten home assistant Version installiert.
Nun habe ich einen Solaredge WEchselrichter von dem ich mir bereits alle Daten vom Modbus TCP hole und diese sind bereits als Entity vorhanden (Doppelabfrage des Modbus TCP is soweit ich weiß nicht möglich).
In der Readme habe ich zwar jede Menge Wege  bzw. kommunikationsprotokolle gefunden, jedoch würde ich in diesem Fall nur ungerne den Weg über MQTT gehen.
Nun zu meiner Fragen kann dich die Daten der Entity's auch irgendwie über ein Plugin direkt an EVCC zur Verfügung stellen?
Oder muss ich den Weg über MQTT oder HTTP gehen?

andig, 2020-11-18:
Wie Du willst!

atze09, 2020-11-18:
> Wie Du willst!

Das heißt Umweg über MQTT oder HTTP?
Ein direktes eingeben des Enitity's funktioniert nicht? Ist das geplant?

andig, 2020-11-18:
Was ist ein Entity?

atze09, 2020-11-18:
> Was ist ein Entity? Was willst Du an „eingeben“? Hast Du die README gelesen? Plugins?

Ja ich habe die Readme inklusive Plugine gelesen, ich habe die Sachen vom PV Wechselrichter in Entity's integriert.
Somit sind die ganzen Info's in sogenannte Sensoren vorhanden. Es wäre somit cool so einen Sensor direkt in deine config zu integrieren.
[image]

andig, 2020-11-18:
Sorry, was sind diese Entitäten?

atze09, 2020-11-18:
In diesem Fall virtuelle Sensoren in home assistant, dahinter steht dann jeweils eine Variable in diesem Fall Strom, Spannung pro Phase und so weiter..
Deswegen, wäre es sehr smart diese direkt in dein Plugin per Plugin integrieren zu können.
Ich weiß aber programmiertechnisch nicht wie die Schnittstelle zu EVCC aussieht bzw. wie das hier übergeben werden kann.

Caibuk, 2020-11-18:
Ich glaube, hier geht es um eine home assistant **Integration** (also ein Plugin, dass es einem in home assistant ermöglicht auf (andigs) evcc zuzugreifen. So etwas gibt es aber noch nicht. Es wird allerdings an einem **Hassio-Addon** gearbeitet. Damit wird keine Anbindung im Sinne einer Datenschnittstelle zwischen evcc <-> home assistant hergestellt.

Zur weiteren Verwirrung trägt sicherlich bei, dass es tatsächlich bereits eine Custom Integration für einen Phoenix-Lade-Controller mit Namen "Electric Vehicle Charge Control" gibt. Diese Integration hat aber NICHTS mit diesem evcc hier zu tun.
[image]

BTW: Entitäten sind in home assistant Objekte mit einem Status und/oder Wert, die zB Geräte eines Smarthomes repräsentieren (ganz plump gesprochen).

atze09, 2020-11-18:
> Ich glaube, hier geht es um eine home assistant **Integration** (also ein Plugin, dass es einem in home assistant ermöglicht auf (andigs) evcc zuzugreifen. So etwas gibt es aber noch nicht. Es wird [...]

Ja genau, aber ich meine schon die EVCC hier.
Aber danke für das Update, schade, aber dann schaue ich mal wie ich das hinbekomme, wahrscheinlich über irgend einen Umweg.

andig, 2020-11-18:
Der „Umweg“ ist überhaupt kein Umweg. Wenn die Daten in MQTT liegen können wir die einfach von da abholen. Alles kein Problem!

atze09, 2020-11-18:
Die Datenschnittstelle könnte ja über MQTT sichergestellt werden.

Caibuk, 2020-11-18:
@atze09 ich habe evcc und ha laufen. Bei mir fragen sowohl ha, als auch evcc (und noch mbmd - um noch weitere Kürzel einzuwerfen) ziemlich rege per Modbus TCP meinen Kostal Plenticore WR ab. Das ist überhaupt kein Problem. Die Übertragung ist performant und aufgrund des TCP-Protokolls auch in der Übertragung abgesichert (da gehen keine Pakete verloren - Handshake und so weiter, im Gegensatz zu UDP).
Du könntest, wie @andig schreibt, natürlich auch MQTT nutzen. Oder die Werte in ha als web endpoints per http zur Verfügung stellen. Es gibt so viele Möglichkeiten. evcc lässt einem da die Wahl.

atze09, 2020-11-18:
> Der „Umweg“ ist überhaupt kein Umweg. Wenn die Daten in MQTT liegen können wir die einfach von da abholen. Alles kein Problem!

Genau aber ich muss schauen wie ich die Entity's (Sensoren) in den MQTT Broker bekomme.
Ich halte euch auf dem laufenden.

andig, 2020-11-18:
Ich verstehs einfach nicht. Wie hast Du sie denn in Hassio _rein_ bekommen?

atze09, 2020-11-18:
> Ich verstehs einfach nicht. Wie hast Du sie denn in Hassio _rein_ bekommen?

Noch gar nicht.

> @atze09 ich habe evcc und ha laufen. Bei mir fragen sowohl ha, als auch evcc (und noch mbmd - um noch weitere Kürzel einzuwerfen) ziemlich rege per Modbus TCP meinen Kostal Plenticore WR ab. Das ist [...]

So werde ich es auch machen..

closed 2020-11-18 by atze09