evcc-io/evcc #452: Make Phoenix EV-CC-AC1-M3 enabled by default

L-Online, 2020-11-19:
Hello,

Thanks for this great project!
I have a DIY portable charger, based on the EV-CC-AC1-M3 from Phoenix Contact.
The charge controller is great, but has one big disadvantage for my use-case.
When configuring register 4000 for remote enabling via RS485, the default state after power on is "disabled" (which probably makes sense for public chargers).
Default state "charging enabled" after power-up of the EV-CC-AC1-M3 is much better because:
- using as portable charger is possible (the Raspberry is not inside the charger)
- when the EVCC system is not working for some reason, charging is still possible, when the admin (me) is not at home ;)
- In the winter, when solar power is barely enaugh for base load, the Raspberry can be shut down, to save power.

My proposed workaround is pretty simple: The charger is disabled by switching the digital OUT of the EV-CC-AC1-M3 on ("0" to register 23000).

On the HW side, this requires a relay, controlled by the OUT signal, with NC contacts between EN and 12V

The relevant registers have to be configured:
4000    -> 1 (remanent)
5500    -> 0 (remanent)
20000  -> dont care
23000  -> disable charging

What do you think?
How do you recommend to implement that? (I am new to Github whith some C and Matlab programming, but I guess I would try it)

BR Ludwig

[image]
ys in

andig, 2020-11-19:
Before we start: which one to you have- the `EV-CC-AC1-M3-CBC-SER` or `EV-CC-AC1-M3-CBC-RCM-ETH`? I've just realized, the the current implementation is actually a mixture of both and that- although they are both `EV-CC-AC1-M3`, the register sets are totally different. The implementation covers the `SER` variant.

L-Online, 2020-11-19:
EV-CC-AC1-M3-CBC-SER-PCB is the exact model.

andig, 2020-11-19:
Right now, evcc doesn't configure anything on startup:

- is that necessary or a one-time configuration?
- if necessary: shouldn't 4000 be 3?

In other words: I'm unclear what you're trying to achieve and if it's evcc's task ;)

L-Online, 2020-11-19:
My problem is: when register 4000 is set to 3, (which is necessary, for enable charger via modbus normally), the default state of the charger is "disabled" (reg 20000 = 0). 
That means, when the RS485 connector is unplugged, the charger cannot be enabled.

For my suggestion, evcc only has to take care of register 23000 
0 -> enable charger (default after startup), 
1-> disable charger

The relay inverts the signal OUT and feeds it to the EN contact of the EV-CC-AC1-M3-CBC-SER-PCB.
Since OUT is 0 after startup, the EN contact is activated by the "Normally Closed" contact of the relay.

Register 4000 -> 1 and 5500 -> 0 is factory default and has to stay that way or have to be configured back to this states manually.
That is another advantage for beginners -> no configuring of registers necessary for use with evcc.

The easiest way, to fulfill my wish, is adding to every "reg 20000 = 0" in the code a "reg 23000 = 1" and vice versa.
The nicer way would be, that the evcc.yaml file has a variable to set for "enable via reg 23000 and inverted OUT contact".
So if anyone uses the OUT signal for something else, this function does not cause problems.

andig, 2020-11-19:
Your suggestion would _only_ work with the additional relay? In that case it would be won't fix.

L-Online, 2020-11-19:
Correct, works only with relay.

andig, 2020-11-19:
This is a very specific requirement, but I'd be happy to accept a PR as long as the changes remain compatible with the current implementation.

L-Online, 2020-11-20:
Okay, thanks! I'll fork the project and see, if I can handle that 👍

andig, 2020-11-20:
Great. Let me know if you need help. That part of the code ist pretty stable, so you can just modify the original file. Alternatively create a new „charger“ type for your purpose by copying.

closed 2020-11-20 by andig