evcc-io/evcc #454: HW Detection delivers wrong results

mclane, 2020-11-19:
Auto detection will now start to scan the network for available devices.  
    We're focusing on devices that are commonly used that are detectable with reasonable efforts.
    On successful detection, suggestions for EVCC configuration can be made.
    Full auto configuration is probably no achievable.

    [main  ] INFO 2020/11/19 13:52:21 my ip: 192.168.0.134
    [main  ] INFO 2020/11/19 13:52:21 ip: 127.0.0.1 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.38 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.39 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.27 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.29 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.42 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.45 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.2 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.40 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.40 task: tcp_502 ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.1 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.47 task: ping ok
    [main  ] INFO 2020/11/19 13:52:21 ip: 192.168.0.24 task: ping ok
    [main  ] INFO 2020/11/19 13:52:23 ip: 192.168.0.50 task: ping ok
    [main  ] INFO 2020/11/19 13:52:23 ip: 192.168.0.27 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:23 ip: 192.168.0.2 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:23 ip: 192.168.0.29 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:23 ip: 192.168.0.38 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:23 ip: 192.168.0.52 task: ping ok
    [main  ] INFO 2020/11/19 13:52:23 ip: 192.168.0.24 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:24 ip: 192.168.0.52 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:24 ip: 192.168.0.47 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:26 ip: 192.168.0.50 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:26 ip: 192.168.0.40 task: sunspec ok
    [main  ] INFO 2020/11/19 13:52:26 ip: 192.168.0.72 task: ping ok
    [main  ] INFO 2020/11/19 13:52:26 ip: 192.168.0.73 task: ping ok
    [main  ] INFO 2020/11/19 13:52:26 ip: 192.168.0.74 task: ping ok
    [main  ] INFO 2020/11/19 13:52:26 ip: 192.168.0.48 task: ping ok
    [main  ] INFO 2020/11/19 13:52:27 ip: 192.168.0.76 task: ping ok
    [main  ] INFO 2020/11/19 13:52:28 ip: 192.168.0.77 task: ping ok
    [main  ] INFO 2020/11/19 13:52:28 ip: 192.168.0.72 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:28 ip: 192.168.0.72 task: go-e ok
    [main  ] INFO 2020/11/19 13:52:28 ip: 192.168.0.72 task: evsewifi ok
    [main  ] INFO 2020/11/19 13:52:28 ip: 192.168.0.72 task: powerwall ok
    [main  ] INFO 2020/11/19 13:52:30 ip: 192.168.0.40 task: inverter ok
    [main  ] INFO 2020/11/19 13:52:41 ip: 192.168.0.40 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:45 ip: 192.168.0.100 task: ping ok
    [main  ] INFO 2020/11/19 13:52:45 ip: 192.168.0.104 task: ping ok
    [main  ] INFO 2020/11/19 13:52:45 ip: 192.168.0.103 task: ping ok
    [main  ] INFO 2020/11/19 13:52:45 ip: 192.168.0.102 task: ping ok
    [main  ] INFO 2020/11/19 13:52:46 ip: 192.168.0.102 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:46 ip: 192.168.0.103 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:46 ip: 192.168.0.115 task: ping ok
    [main  ] INFO 2020/11/19 13:52:46 ip: 192.168.0.100 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:52:47 ip: 192.168.0.120 task: ping ok
    [main  ] INFO 2020/11/19 13:52:52 ip: 192.168.0.134 task: ping ok
    [main  ] INFO 2020/11/19 13:52:52 ip: 192.168.0.137 task: ping ok
    [main  ] INFO 2020/11/19 13:52:54 ip: 192.168.0.140 task: ping ok
    [main  ] INFO 2020/11/19 13:53:08 ip: 192.168.0.1 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:53:10 ip: 192.168.0.167 task: ping ok
    [main  ] INFO 2020/11/19 13:53:10 ip: 192.168.0.169 task: ping ok
    [main  ] INFO 2020/11/19 13:53:10 ip: 192.168.0.170 task: ping ok
    [main  ] INFO 2020/11/19 13:53:11 ip: 192.168.0.174 task: ping ok
    [main  ] INFO 2020/11/19 13:53:11 ip: 192.168.0.178 task: ping ok
    [main  ] INFO 2020/11/19 13:53:11 ip: 192.168.0.179 task: ping ok
    [main  ] INFO 2020/11/19 13:53:11 ip: 192.168.0.169 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:53:11 ip: 192.168.0.170 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:53:13 ip: 192.168.0.167 task: tcp_80 ok
    [main  ] INFO 2020/11/19 13:53:13 ip: 192.168.0.194 task: ping ok
    [main  ] INFO 2020/11/19 13:53:13 ip: 192.168.0.195 task: ping ok
    [main  ] INFO 2020/11/19 13:53:14 ip: 192.168.0.198 task: ping ok
    [main  ] INFO 2020/11/19 13:53:16 ip: 192.168.0.200 task: ping ok
    [main  ] INFO 2020/11/19 13:53:16 ip: 192.168.0.202 task: ping ok
    [main  ] INFO 2020/11/19 13:53:17 ip: 192.168.0.203 task: ping ok
[... 56 lines cut ...]
    +               +---------------------+           +--------------------------------+
    |               |                     |           | {SlaveID:3 Model:103 Point:W   |
    |               |                     |           | Value:22}                      |
    +               +---------------------+           +--------------------------------+
    |               |                     |           | {SlaveID:71 Model:103 Point:W  |
    |               |                     |           | Value:23}                      |
    +               +---------------------+-----------+--------------------------------+
    |               |                     | sunspec   | {SlaveID:71 Model:1 Point:Mn   |
    |               |                     |           | Value:KACO new energy}         |
    +---------------+---------------------+-----------+--------------------------------+
    | 192.168.0.72  | GongDGZentral.local | go-e      |                                |
    +               +                     +-----------+--------------------------------+
    |               |                     | evsewifi  |                                |
    +               +                     +-----------+--------------------------------+
    |               |                     | powerwall |                                |
    +---------------+---------------------+-----------+--------------------------------+

    Please open https://github.com/andig/evcc/issues/new in your browser and copy the
    results above into a new issue. Please tell us:

            1. Is the scan result correct?
            2. If not correct: please describe your hardware setup.`

Der Lader (.225) und der WR (.40) werden richtig erkannt. go-e, evsewifi und powerall (.72) sind falsch. GongDGZentral ist ein RasPi, der (wie der Name schon sagt) ein Gong-mp3 abspielt wenn an der Tür geklingelt wird. Ausserdem läuft da noch ein reverse proxy, der eine ioborker hue-emulation auf den port 80 umbiegt. Das hat mit evcc nichts zu tun.
Es gibt in unserem Netz noch 2 MQTT Broker, die in früheren Versionen von evcc detect auch richtig erkannt wurden, nun aber nicht mehr erscheinen.

andig, 2020-11-19:
Was liefert 

    curl -v -H "Content-type: application/json" 192.168.0.72

andig, 2020-11-19:
Die MQTT Broker sind fürs erste nicht mehr in der Discovery damit wir und auf die Geräte konzentrieren können.

mclane, 2020-11-19:
[uli@Libra ~]$ curl -v -H "Content-type: application/json" 192.168.0.72
*   Trying 192.168.0.72:80...
* Connected to 192.168.0.72 (192.168.0.72) port 80 (#0)
> GET / HTTP/1.1
> Host: 192.168.0.72
> User-Agent: curl/7.73.0
> Accept: */*
> Content-type: application/json
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: nginx/1.6.2
< Date: Thu, 19 Nov 2020 14:48:11 GMT
< Content-Type: application/json; charset=utf-8
< Content-Length: 2
< Connection: keep-alive
< X-Powered-By: Express
< Access-Control-Allow-Origin: *
< Access-Control-Allow-Methods: POST, GET, OPTIONS, PUT, DELETE, HEAD
< Access-Control-Allow-Credentials: true
< Access-Control-Max-Age: 3600
< Access-Control-Allow-Headers: Origin, X-Requested-With, Content-Type, Accept
< ETag: W/"2-vyGp6PvFo4RvsFtPoIWeCReyIC8"
< 
* Connection #0 to host 192.168.0.72 left intact

andig, 2020-11-19:
Ich hab den Branch nochmal um mehr Output erweitert- vielleicht liefert das Aufschluss. Bei mir kein Fehler reproduzierbar :/

andig, 2020-11-19:
Fixed :)

closed 2020-11-19 by andig