evcc-io/evcc #455: Auto-detect multiple connected vehicle - nach welcher Logik wird entschieden?

martinez81, 2020-11-20:
Da seit gestern Abend endlich bei meinen Leaf die SOC-Abfrage klappt, wäre es schon super die "Auto-detect multiple connected vehicle" Option zu nutzen.
Aber wie funktioniert das?
Ich habe dazu gestern beide Autos (Leaf und Model 3) korrekt eingetragen.
Es wird aber immer nur das 1. Auto in der GUI angezeigt.
Nach welcher Logik wird entschieden, welches Fahrzeug gerade anschlossen ist?
Gruß

andig, 2020-11-20:
Statt:

    vehicle: car1

jetzt:

    vehicles:
    - car 1
    - car 2

martinez81, 2020-11-20:
Ja, das habe ich so auch umgesetzt.
car 1 war aber nicht angeschlossen. car 2 war dran und hat geladen.
Es wurde aber nur der name und soc von car 1 angezeigt.

andig, 2020-11-20:
Da bräuchte ich ein Logfile. Die Umschaltung ist allerdings träge- sie kann erst greifen wenn der API Cache abgelaufen ist (Default 5min) damit zu sehen ist, dass das erste Auto nicht mehr verbunden ist.

martinez81, 2020-11-20:
> 
> 
> Da bräuchte ich ein Logfile. Die Umschaltung ist allerdings träge- sie kann erst greifen wenn der API Cache abgelaufen ist (Default 5min) damit zu sehen ist, dass das erste Auto nicht mehr verbunden ist.

Das bedeutet also Cache: 5m  --> 5Min 😄 
Das haben wir uns die ganze Zeit gefragt...

So lange haben wir nicht gewartet. Das wird heute Nachmittag getestet!

Noch eine Frage dazu: Wie, oder durch was merkt EVCC welches Auto angeschlossen ist? Das verstehe ich noch nicht.

premultiply, 2020-11-20:
An der Stelle wäre natürlich zu überlegen ob man das Caching iegendwie dynamischer machen kann.
Kurze Intervalle (1m) um Start/Stop/Stastuswechsel rum und dafür lange Intervalle (15m) während des Ladevorgangs. Dann wäre auch der Estimator genauer ;)

andig, 2020-11-20:
> Wie, oder durch was merkt EVCC welches Auto angeschlossen ist? Das verstehe ich noch nicht.

Heuristik. Wenn das "angeschlossene" Auto "unplugged" sagt, der Charger aber "connected" dann schaut evcc ob das zweite Auto vielleicht auch "connected" sagt. Dafür brauchts aber eben die API Ergebnisse...

andig, 2020-11-20:
> Kurze Intervalle (1m) um Start/Stop/Stastuswechsel rum und dafür lange Intervalle (15m) während des Ladevorgangs. Dann wäre auch der Estimator genauer ;)

Wenn Du mir sagst wie wir das tun ohne dass Kia und Co wieder die Anwender aussperren? Mir ist das zu heiß- einfach ein kleinwenig Geduld mitbringen ;)

martinez81, 2020-11-20:
> Heuristik. Wenn das "angeschlossene" Auto "unplugged" sagt, der Charger aber "connected" dann schaut evcc ob das zweite Auto vielleicht auch "connected" sagt. Dafür brauchts aber eben die API [...]

Ah, okay! unplugged und connected wird im Moment bei meiner Lösung über das Script nicht übergeben.
Sicher könnte ich diesen Zustand über Dartnissanconnect auslesen...
Wie kann ich das dann per Script übergeben?
Wie sollte die Generisch (Script) aussehen?
Über cmd wird mein SOC übergeben. Über welchen Ausdruck kann ich die Verbindung übergeben?

```
- type: default
  title: Auto # display name for UI
  capacity: 50 # kWh
  charge:
    type: script # use script plugin
    cmd: /bin/sh -c "echo 50" # actual command
    timeout: 3s # kill script after 3 seconds
  cache: 5m # cache duration
```

andig, 2020-11-21:
Ich verstehe die Frage nicht. Was willst Du tun?

martinez81, 2020-11-22:
> Ich verstehe die Frage nicht. Was willst Du tun?

Du hast oben geschrieben:
> > Wie, oder durch was merkt EVCC welches Auto angeschlossen ist? Das verstehe ich noch nicht.
> 
> Heuristik. Wenn das "angeschlossene" Auto "unplugged" sagt, der Charger aber "connected" dann schaut evcc ob das zweite Auto vielleicht auch "connected" sagt. Dafür brauchts aber eben die API Ergebnisse...

Dadurch bin ich davon ausgegangen, dass das Auto "angeschlossen" oder "unplugged" mitteilen muss, also per script den Zustand vom Auto an EVCC übergeben werden muss.
Im Moment wird ja nur der SOC übergeben, nicht ob das Auto angeschlossen ist, oder nicht.

andig, 2020-11-22:
> Dadurch bin ich davon ausgegangen, dass das Auto "angeschlossen" oder "unplugged" mitteilen muss, also per script den Zustand vom Auto an EVCC übergeben werden muss.

Ah, verstanden. Fast. Das API heisst `status` analog `charge` und muss den Status analog der Wallbox als Buchstaben zurück geben:
- `A`: getrennt
- `B`: plugged
- `C`: charging

andig, 2020-11-22:
Ich mache hier mal zu. Mit #465 hat sich das erledigt ;)

closed 2020-11-22 by andig

themuck, 2022-07-12:
> Statt:
> 
> ```
> vehicle: car1
> ```
> 
> jetzt:
> 
> ```
> vehicles:
> - car 1
> - car 2
> ```

Hallo, ist das so noch aktuell? Ich bekomme da einen Fehler...

```Jul 12 08:19:07 evcc evcc[213945]: [main  ] FATAL 2022/07/12 08:19:07 failed configuring loadpoint: 1 error(s) decoding:
Jul 12 08:19:07 evcc evcc[213945]: * 'vehicle' expected type 'string', got unconvertible type '[]interface {}', value: '[SuperB]'
```

```
loadpoints:
- title: Scheune
  charger: wallbox1
  vehicle:
  - SuperB
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
```
   
```
vehicle: SuperB
```
Die letzte Variante geht...