evcc-io/evcc #461: Zappi Einbindung

xerion3800, 2020-11-20:
Hallo bin gerade dabei mein PV aufzurüsten und in die Elektromobilität einzusteigen. Bin dabei auf dieses Project gestoßen. Habe das auch schon mehr oder weniger im Docker Container am laufe bekommen und bin sehr dankbar für eure Arbeit. Momentan bin ich mir noch nicht sicher welche Wallbox ich mir anschaffen werden. Ich finde die myEnergi Zappi ganz interessant ist da schon was zwecks Einbindung der API geplant?

andig, 2020-11-21:
Ich kenne die Zappi niicht.  Warum nicht eine d er unterstützten Boxen nehmen? Hat Zappi ein Api?

xerion3800, 2020-11-21:
Hier der Link der API https://github.com/twonk/MyEnergi-App-Api

Die Zappi hat den Vorteil daß man, über Stromklemmen die Werte drahtlos zur Zappi senden kann ist sehr interessant für den Nachrüstbereich falls der Schaltschrank kein Platz mehr bietet für einen MID Zähler oder z.B wenn eine kabelgebundene Installation nicht möglich ist.

andig, 2020-11-21:
Sieht gut aus. Willst Du einen PR versuchen?

andig, 2020-11-21:
Die Zappi scheint ja selbst schon einige Features von EVCC zu haben- welche Vorteile siehst Du denn wenn EVCC die Zappi auch unterstützen würde ggü der reinen Zyppi Nutzung?

Caibuk, 2020-11-21:
Zappi kann zB keinen SoC auslesen. Man kann nur Energiemengen vorgeben.

xerion3800, 2020-11-21:
Also grundsätzlich wurde mir der Zappi von meinem Installateur empfohlen. Wir erweitern gerade die PV Anlage und das EV ist auch schon bestellt. Für die meisten Wallboxen ist eine PV abhängige Regelung keine große Herausforderung mehr. Aber immer nur durch zusätzliche Hardware. Zusätzlich ist es in Kombination mit einem Speicher noch schwieriger. Durch den Speicher sind ja schon diverse Messeinrichtungen vorhanden die ich aber z.B. mit den Zappi nicht nutzen kann. Mit evvc kann ich dann über mein Home-Server (FHEM) die Daten des Speichers per Mqtt an evcc senden das Auto wird auch unterstützt und jetzt fehlt nur noch eine passende WB. Ich favorisiere momentan eigentlich den go-e Charger muss aber nich den Installateur überzeugen. Bei go-e stört mich nur dass es dort kein festes Kabel dran gibt und eine App auch noch nicht vorhanden ist. Das gibt es wiederum bei Zappi. Wollte nur Mal hören ob Zappi schon auf der ToDo Liste stand da die Zappi ansich sehr gut bewertet wird. Ich würde gerne diese Projekt dafür nutzen wollen um unabhängig bleiben slzu können und um die schon vorhandene Messtechnik nutzen zu können.

cathiele, 2020-11-21:
>... Ich favorisiere momentan eigentlich den go-e Charger muss aber nich den Installateur überzeugen. Bei go-e stört mich nur dass es dort kein festes Kabel dran gibt und eine App auch noch nicht vorhanden ist. ...

Was meinst du hier mit „App“? für android / iOS gibt es für den Go-eCharger Apps.

xerion3800, 2020-11-21:
> Was meinst du hier mit „App“? für android / iOS gibt es für den Go-eCharger Apps.

Ja stimmt hatte ich gerade nicht auf den Schirm...

andig, 2020-11-21:
Und wofür eine App wenns EVCC gibt?

xerion3800, 2020-11-21:
Ohne VPN die Ladestation abfragen. Aber ist nicht so wichtig. Die Steuerung der WB anhand des PV Überschuss und Berücksichtigung des Speichers hat die höchste Priorität.

andig, 2020-11-21:
> Ohne VPN die Ladestation abfragen.

Kannst Du mit evcc auch- "einfach" Google Auth davor schalten. Bzgl. Steuerung: die Zappi will offensichtlich immer über die Cloud gehen. Wenns Internet mal weg ist wirds dann schwierig mit Auto laden...

andig, 2020-11-21:
Also Summary: Zappi ist möglich, aktuell aber nicht unterstützt. Mangels Kapazität mache ich fürs erste mal mit wontfix zu, PR gerne.

closed 2020-11-21 by andig

metalux2030, 2023-04-18:
Gibt es hier mittlerweile eine Unterstützung für die Zappi?
Danke im voraus für die Info.

premultiply, 2023-04-18:
Nein. Geschlossenes System.

wuedan, 2023-07-27:
Zappa unterstützt neuerdings OCPP - vielleicht lässt sie sich damit ja manuell an EV-CC anbinden?

rgo1337, 2023-11-16:
Moin,

bin Laie wenn es um die Programmierung geht aber ich kann im HomeAssistant die Wallbox bedienen wie es mir passt. Auch mit anderen Integrationen ist das möglich. 
Dennoch würde ich gerne evcc nutzen. Evtl über diesen weg nutzbar? (da evcc im HA als Addon lauffähig ist, könnte man darüber die Steuerung übernehmen).
Gruß

andig, 2023-11-16:
@rgo1337 oben steht dass sie Ocpp kann- also einfach ausprobieren?

Sabnock2210, 2024-03-05:
Mahlzeit. Hat schon jemand versucht die Zappi einzubinden? Wäre auch für mich interessant

rgo1337, 2024-12-02:
Hallo zusammen,

ich pack das Thema aus der Mottenkiste hoch. Da ja [Marq24](https://github.com/marq24/ha-evcc) eine nettes HA-CustomComp gebaut hat, ist es ja nun möglich Sensoren direkt aus HA abzugreifen. Das Ganze würde natürlich super mit dem Abgreifen von weiteren Entitäten passen.

die alte Zappi (mit SNR startend mit 1 - also "alte" Version) kann leider kein OCPP, daher entfällt der Weg. Nur die neuen Modelle unterstützen dies.

Allerdings, wenn die Kommunikation zwischen HA und EVCC so funktioniert - wäre es wert das Ganze für HA einzubinden? Die Vorarbeit müsste in HA natürlich getätigt werden (Start, Stop, Modi).

andig, 2024-12-02:
Kann die Neue Ocpp? Dann brauchts auch kein HA.

rgo1337, 2024-12-02:
wenn meine die neue wäre, dann würd ich es ja machen .. aber hab ja eben die alte :/

wuedan, 2024-12-02:
Die Zappi V2.1 unterstützt zwar OCPP - aber nur über die Cloud, sprich, die Wallbox selbst spricht kein OCPP sondern würde die Befehle nur über den MyEnergi (Hersteller) Server erhalten - in sofern ist keine Netzwerkinterne Kommunikation mit der Zappi via OCPP möglich, man müsste (unverschlüsselt) über die Cloud gehen. Aus dem Grund ist die OCPP Konfiguration der Zappi auch nur im Webportal möglich - und hier eben keine lokalen Adressen.
 
[image]

guckmaster, 2024-12-22:
Hallo, wäre cool, wenn man die Zappi einbinden könnte.
Die Zappi ist über HA steuerbar.

andig, 2024-12-22:
Ist sie? Gibts ne Doku dazu? Wie wäre es- alternativ- sie über HA einzubinden (`custom` Charger)?

rgo1337, 2024-12-22:
> Ist sie? Gibts ne Doku dazu? Wie wäre es- alternativ- sie über HA einzubinden (`custom` Charger)?

Hi,

Ich verstehe deine „ist sie?“ - Frage nicht.

Doku wäre hier https://github.com/CJNE/ha-myenergi

führst du einen Feldzug gegen HomeAssistant? Was man eigentlich nur sagen möchte ist, über eine CustomIntegration für HA wäre die Zappi Steuerung möglich, man möchte aber Deine EVCC Lösung nutzen - es geht aber nicht. 

Vllt würde ja eine Adaption der CustomIntegration für Dich hier nützlich sein, damit die Leute, die leider eine Zappi haben auch das EVCC nutzen könnten

andig, 2024-12-22:
> Doku wäre hier

Prima

> führst du einen Feldzug gegen HomeAssistant

Wie kommst du da drauf? 

> es geht aber nicht.

Was geht denn nicht? Ein bisschen konkreter müsstest du schon werden.

andig, 2024-12-22:
Als Zappi Kunde würde ich Druck machen, eine standardkonforme OCPP Implementierung mit Smart Charging Profile zu bekommen. Das würde alle Probleme lösen...

rgo1337, 2024-12-22:
Meine 2.0er kann leider kein OCPP. Ob die 2.1er das macht und welche Wege sie gehen muss, keine Ahnung. 
Hast recht, sie ist recht dumm. War „damals“ als pv Lösung als heißer scheiss beworben worden-das es nicht so ist weiß ich heute auch. 

der Support ist fürn arsch - Man könnte fast dubios sagen. 

Die Wunsch Lösung, die soweit Smart zu machen damit mit damit leben kann das sie nur Fast(11kW), Eco+ (1p,3p,automatisch bei pv-only) oder Eco (pv only + variabler Prozentbereich Netzstrom zu ziehen - Beispiel: 6kw kommt vom Dach und 20% vom netz also 1,2kw =9,2kw gesamt ins auto) wäre, ein einfacher Boolean ein aus. 
wenn EVCC erkennt das Strom da ist und wieviel (alles vorhanden) einfach ne variable befüttern. 

So könnte jeder, der HA z.B. nutzt die variable abgreifen und seine Wallbox aus HA mit eigenen Mitteln starten / Modi auswählen etc.

das mir sogar der SoC vom Auto fehlt, fang ich garnicht erst an :(

mfuchs1984, 2024-12-22:
@rgo1337 auf Basis deiner Links oben würde ich sagen, die Wallbox ist zwar steuerbar, aber bietet nicht das von evcc benötigte Interface, da weder die Ladeströme noch die Leistung gesteuert werden kann.

mfuchs1984, 2024-12-22:
Im Übrigen kannst du evcc jederzeit mit Home Assistant verbinden und dort verfügbare Geräte auslesen und steuern, sowohl über mqtt als auch über die Home Assistent API.