evcc-io/evcc #467: Export metrics via Prometheus Exporter

mweinelt, 2020-11-22:
**Describe the solution you'd like**
Prometheus is a popular time series database that pulls data from exporters. EVCC could expose an [open metrics](https://openmetrics.io/) endpoint to support it and other time series databases compatible with it.

**Describe alternatives you've considered**
Influxdb already exists, but InfluxSQL and Flux are both alot more complicated than PromQL. For example I have had issues with doing calculations between two different series in Influxdb, which to me is mind blogging.

**Additional context**
An open metrics endpoint is a HTTP endpoint for which all of the boilerplate is provided by the Prometheus instrumentation library (https://github.com/prometheus/client_golang), so this would likely be low maintenance.

The prometheus website has a short guide on how instrumentation with Go projects works: https://prometheus.io/docs/guides/go-application/.

Either way, I know adding support for additional external tooling when a similar one exists is a big ask, so thank you for taking the time to consider this request.

andig, 2020-11-22:
Well-written requirement 👍🏻. As for Influx: use flux instead of influxql. As for metrics: while certainly doable I‘d lean on declining as its and entirely technical thing that no user understands. The audience is diametrically different. On the other side: it wouldn‘t cost much (is there a default path so it wouldn‘t even need to make it into the readme?)- so I‘ve be happy to accept a PR.

mweinelt, 2020-11-22:
> is there a default path so it wouldn‘t even need to make it into the readme?

For the http endpoint? That would be  `/metrics`.
For the port the exporter listens on? There would be a [non-conflicting default port](https://github.com/prometheus/prometheus/wiki/Default-port-allocations), that could be overriden by end users

So the smallest amount of information a prometheus user would need is

> Prometheus endpoint at `http://evcc-host:9999/metrics`

andig, 2020-11-22:
Looking through the spec metrics need to have a type. Currently that information is not available in the codebase and would require maintenance. 

Taking the audience of this application into account, I'm closing as wontfix for time being.

Thank you for an interesting request!

closed 2020-11-22 by andig