evcc-io/evcc #474: ID.3 Verbindung nicht möglich

fsdschmidt, 2020-11-24:
Versuche mit einer Config laut Beispiel die Verbindung zum ID.3 aufzubauen. Verwende dafür das aktuelle binary 0.32 für OSX. Leider bekomme ich eine Fehlermeldung:

cannot create vw vehicle: Get "": unsupported protocol scheme ""

Vermute dass es sich um eine Inkompatibilität zu OSX handelt. Kann es sein, dass für irgendeine URL das Protokoll nicht angegeben ist?

**To Reproduce**
./evcc -c evcc.yaml

**Expected behavior**
Verbindung zu ID.3 ist möglich.

**EVCC details:**
Show output of `evcc -v` between the quotes below:

```
evcc version 0.32 (f458524)
```

Show evcc configuration file `evcc.yaml` between the quotes below:

```
vehicles:
- name: ID.3
  type: vw
  title: ID.3
  capacity: 58 # kWh
  user: ***
  password: ***
  vin: ***
  cache: 5m
```

andig, 2020-11-24:
Hast Du ein Logfile?

fsdschmidt, 2020-11-25:
Hallo,

war mir nicht so ganz sicher, wie sensibel die Daten im Trace sind, deswegen wollte ich sie im github nicht posten. Anbei ein log.

Danke,
Micha

andig, 2020-11-25:
Kein Log anbei. Wenn privat gerne an cpuidle@gmx.de.

andig, 2020-11-25:
Kannst Du Dich im Portal direkt anmelden?

fsdschmidt, 2020-11-25:
Ja, bei myVolkswagen kann ich mich anmelden, sehe dort den ID.3 und auch die aktuelle Restreichweite.

PR #478 (2020-11-25): Trigger meaningful error when VW consent screen is observed
Fix #474
closed 2020-11-25 by andig