evcc-io/evcc #475: hems configuration

ilosuna, 2020-11-24:
**Describe the bug**
Old yaml configuration parameter 

`hems: sma`

produces the following error:

```
[main  ] FATAL 2020/11/24 20:05:31 failed parsing config file /path/to/evcc.yaml: 1 error(s) decoding:
* 'HEMS' expected a map, got 'string'
```
New notation (since release 0.32) according to README
```
hems: 
- type: sma
```
produces this error:

```
[main  ] FATAL 2020/11/24 20:09:32 failed parsing config file /path/to/evcc/evcc.yaml: 1 error(s) decoding:
* 'HEMS' expected a map, got 'slice'
```
**To Reproduce**
Steps to reproduce the behavior:
1. Edit evcc.yaml and add option for [hems](https://github.com/andig/evcc#home-energy-management-system)
2. Start evcc

**EVCC details:**
Show output of `evcc -v` between the quotes below:

```
evcc version 0.32 (f458524)
```

Show evcc configuration file `evcc.yaml` between the quotes below:

```
uri: 0.0.0.0:7070
interval: 10s

log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

#hems: sma
hems:
- type: sma

meters:
- name: sma
  type: sma
  uri: 192.168.178.69

chargers:
- name: keba
  type: keba
  uri: 192.168.178.59:7090

site:
- title: Haus
  meters:
    grid: sma
    pv: sma
    
loadpoints:
- title: Keba
  charger: keba
  meters:
    charge: sma
```

Show evcc log output with `--log trace` between the quotes below:

```
[main  ] INFO 2020/11/24 20:21:09 evcc 0.32 (f458524)
[main  ] INFO 2020/11/24 20:21:09 using config file /path/to/evcc.yaml
[main  ] FATAL 2020/11/24 20:21:09 failed parsing config file /home/mark/evcc/evcc.yaml: 1 error(s) decoding:

* 'HEMS' expected a map, got 'slice'
```

ilosuna, 2020-11-24:
OK, I think I figured it out - it should be without hyphen/minus:

```
hems:
  type: sma
```

martinez81, 2020-11-24:
Hi!
Das Problem hatte ich auch.
Deine Lösung ist richtig. 👍🏼

andig, 2020-11-24:
Doku korrigiert, danke!

closed 2020-11-24 by andig