evcc-io/evcc #480: SMA/Keba: Frontend zeigt zu niedrige Werte

ilosuna, 2020-11-26:
**Describe the bug**
Im Frontend werden zu niedrige Werte für Einspeisung, Erzeugung und Ladeleistung angezeigt. In diesem Moment (2020/11/26 13:24:00) wurde mit etwa 5 kW geladen, das Frontend zeigt nur 32 W:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Öffne das Frontend unter http://host:7070
2. Vergleiche die angezeigten Werte z.B. mit denen im Sunny-Portal

**Expected behavior**
Die Anzeige der Werte zumindest in ähnlicher Größenordnung.

**EVCC details:**
Show output of `evcc -v` between the quotes below:

```
evcc version 0.33 (403e953)
```

Show evcc configuration file `evcc.yaml` between the quotes below:

```
uri: 0.0.0.0:7070
interval: 10s

log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

hems:
  type: sma

meters:
- name: shm
  type: sma
  uri: 192.168.178.69

chargers:
- name: keba
  type: keba
  uri: 192.168.178.59:7090

site:
- title: Haus
  meters:
    grid: shm
    pv: shm
    
loadpoints:
- title: Keba
  charger: keba
  mode: minpv
  meters:
    charge: shm
```

Show evcc log output with `--log trace` between the quotes below:

```
[main  ] INFO 2020/11/26 13:24:08 evcc 0.33 (403e953)
[main  ] INFO 2020/11/26 13:24:08 using config file /opt/evcc/evcc.yaml
[main  ] INFO 2020/11/26 13:24:08 listening at 0.0.0.0:7070
[semp  ] WARN 2020/11/26 13:24:08 SEMP_BASE_URL unspecified, using http://192.168.178.60:7070 instead
[cache ] TRACE 2020/11/26 13:24:08 title: Keba
[cache ] TRACE 2020/11/26 13:24:08 minCurrent: 6
[cache ] TRACE 2020/11/26 13:24:08 maxCurrent: 16
[cache ] TRACE 2020/11/26 13:24:08 phases: 1
[cache ] TRACE 2020/11/26 13:24:08 activePhases: 1
[... 1032 lines cut ...]
"Enable user": 1,
"Max curr": 13000,
"Max curr %": 216,
"Curr HW": 16000,
"Curr user": 13000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 0,
"Input": 0,
"Serial": "18980178",
"Sec": 755138
}
[lp-1  ] DEBUG 2020/11/26 13:25:48 detected phases: 3p [5.21 5.05 4.97]A
[lp-1  ] DEBUG 2020/11/26 13:25:48 max charge current: 13A = 13A + 0A (-51W @ 3p)
[cache ] TRACE 2020/11/26 13:25:48 chargeCurrents: [5.205 5.05 4.972]
[cache ] TRACE 2020/11/26 13:25:48 activePhases: 3
[lp-1  ] DEBUG 2020/11/26 13:25:48 pv timer reset
[lp-1  ] DEBUG 2020/11/26 13:25:48 target charge current: 13A
[cache ] TRACE 2020/11/26 13:25:48 remoteDisabled: 
[keba  ] TRACE 2020/11/26 13:25:49 recv from 192.168.178.59:7090 {"E pres": 104759}
[keba  ] TRACE 2020/11/26 13:25:49 recv: listener blocked
```

premultiply, 2020-11-26:
pv und grid shm?
Da kann doch was nicht stimmen.

ilosuna, 2020-11-26:
Ja, für beides den Sunny Home Manager. Habe ich das falsch verstanden?

premultiply, 2020-11-26:
Ja.
Du musst 3 Meter konfigurieren:
- Netzanschluss (grid). Bei dir wohl der SHM.
- PV-Anlage (pv). Bei dir wahrscheinlich ein SMA Wechselreichter via Modbus TCP und sunspec Modell, sonst weglassen.
- Ladestation (charge): Hat die Keba ein internes Meter? Keine Ahnung. Sonst Extrazähler oder weglassen.

ilosuna, 2020-11-26:
Ach so, ich dachte der Wechselrichter (Sunny Tripower) läuft auch über den SHM.

ilosuna, 2020-11-26:
OK, habe es mal so konfiguriert:

```
meters:
- name: shm
  type: sma
  uri: 192.168.178.69
- name: stp
  type: modbus
  uri: 192.168.178.70:502 
  id: 126
  model: sma-sunspec
  power: Power
  energy: Sum

# ...

site:
- title: Haus
  meters:
    grid: shm
    pv: stp
```
Die Werte scheinen jetzt besser zu stimmen, es ist gerade aber nicht mehr so viel Sonnenenergie (werde das in den nächsten Tagen beobachten).

Wer auch einen Sunny Tripower hat: Modbus muss aktiviert sein. Bei mir war das nicht der Fall. Musste mich erst als Installateur (Passwort war zum Glück wie Benutzer) in die Admin-Oberfläche einloggen und Modbus aktivieren (Externe Kommunikation --> Modbus --> TCP-Server=ja).

andig, 2020-11-26:
Für Werte einfach mal ein `evcc meter -c <config>` aufrufen

ilosuna, 2020-11-26:
Ah, gut zu wissen! Ich bekomme

```
shm
---
Power:          187W
Current L1..L3: 0.8A 0.6A 0.5A

stp
---
Power:  0W
Energy: sunspec: Sum not found
```
Ich hoffe mal "Sum not found" kommt, weil es dunkel ist.

ilosuna, 2020-11-26:
Gerade im Slack erfahren: `energy: Sum` muss raus!

andig, 2020-11-27:
Nein, das kommt weil Dein WR das nicht implementiert hat (und nicht braucht). Muss raus- Energy ist nur für Ladezähler relevant und wird sonst ohnehin nicht verwendet.

andig, 2020-11-27:
Ich mache hier mal zu- denke das Problem ist gelöst.

closed 2020-11-27 by andig