evcc-io/evcc #485: ID.3 gibt 403 Fehler

StefanSchoof, 2020-11-27:
**Describe the bug**
Ich habe einen ID.3 den ich versuche in evcc einzubinden. Ich dem ich den Consent auf https://www.portal.volkswagen-we.com gegeben habe, bekomme ich die Meldung:
```
State: unexpected status: 403
Charge status: unexpected status: 403
Finish time:   unexpected status: 403
Climater:      unexpected status: 403
```
Auf https://www.portal.volkswagen-we.com wird kein aktives Fahrzeug angezeigt. Wenn ich versuche das Fahrzeug dort hinzufügen bekomme ich die Meldung "Das Fahrzeug konnte nicht hinzugefügt werden
Die von Ihnen eingegebene FIN kann zu einem Fahrzeug gehören, für das die mobilen Online-Dienste nicht genutzt werden können. Bitte prüfen Sie zur Sicherheit noch mal die Groß- und Kleinschreibung der eingegebenen FIN und achten Sie auf die Verwechslungsmöglichkeit der Zahl „0“ und des Buchstabens „O“."

Wird also noch ein Schritt benötigt, um den ID.3 einzufügen?

**To Reproduce**
1. Neuen ID.3 zu evcc.yaml hinzufügen
2. Consent auf https://www.portal.volkswagen-we.com geben
3. `evcc vehicle`
4. Fehlermeldung 403

**Expected behavior**
Das die Werte ausgelesen werden oder das eine Fehlermeldung mit Informationen, was zu tun ist um das zum Laufen zu bekommen.

**EVCC details:**
Show output of `evcc -v` between the quotes below:

```
evcc version 0.33 (403e953)
```

Show evcc configuration file `evcc.yaml` between the quotes below:

```
vehicles:
- type: vw
  title: ID.3
  capacity: 58 # kWh
  user: <user>
  password: <password>
  vin: <VIN>
  cache: 5m
```

Show evcc log output with `--log trace` between the quotes below:

```
Erstelle ich wenn benötigt (aber da sind Token und co drin)
```

**If using Docker:**
Show output of `docker run andig/evcc -v` between the quotes below:

```
<- paste here
```

Show evcc log output with `docker logs <container id>` between the quotes below:

```
<- paste here
```

andig, 2020-11-27:
Ähnliches Problem wie https://github.com/andig/evcc/issues/474. Es betrifft anscheinend einen großen Teil der neuen ID Fahrzeuge. Teilweise sind sie im Portal sichtbar, aber nicht im API, teilweise- wie hier bei Dir- sind sie auch nicht im Portal sichtbar.

Hier gibt es nichts das wir tun könnten- frag doch mal VW wie Du "die mobilen Online-Dienste" nutzen kannst? Freu mich auf Feedback!

StefanSchoof, 2020-11-27:
@fsdschmidt Siehst du jetzt nach dem du den Consent auf https://www.portal.volkswagen-we.com/ dein Auto dort oder nur unter https://www.volkswagen.de/de/besitzer-und-nutzer/myvolkswagen.html?

fsdschmidt, 2020-11-27:
Ich sehe es nur unter volkswagen.de. Unter portal.volkswagen-we.com wird es nicht angezeigt. Wenn ich dort versuche es hinzuzufügen, bekomme ich die Meldung:

Die von Ihnen eingegebene FIN kann zu einem Fahrzeug gehören, für das die mobilen Online-Dienste nicht genutzt werden können. 

Nach meinem Wissen sind bei mir sowohl im Auto und auch unter volkswagen.de alle Online-Funktionen aktiviert. Die We-Connect-ID App funktioniert auch mit dem bisher bescheidenen Funktionsumfang. Kann mir also nur vorstellen, dass der ID.3 irgendwie nicht mehr kompatibel zu dem alten Portal ist ...

StefanSchoof, 2020-11-27:
Und kannst du jetzt evcc nutzen?

fsdschmidt, 2020-11-27:
Nein ich bekomme ebenfalls den 403 in evcc. Habe jetzt mal bei der VW-Hotline angerufen, da bekomme ich auch nur den Verweis auf die neue WeConnect-ID App. Man muss zwingend diese nutzen, der ID.3 ist nicht mehr zum bisherigen WeConnect kompatibel und dazu zählt dann wohl auch portal.volkswagen-we.com.

Dann kommen wir also hier nicht weiter, wenn VW nicht für die ID-Familie ein Pendant zu dem portal anbietet?

StefanSchoof, 2020-11-27:
Es gibt noch die URL `https://cardata.apps.emea.vwapps.io/vehicles/<FIN>/fuel/status` die über myVolkswagen auf der Details Seite aufgerufen wird:
```json
{
    "data": [
        {
            "id": "primaryEngine",
            "carCapturedTimestamp": "2020-11-26T21:04:50Z",
            "properties": [
                {
                    "name": "engineType",
                    "value": "electric"
                },
                {
                    "name": "remainingRange_km",
                    "value": "250"
                },
                {
                    "name": "currentSOC_pct",
                    "value": "75"
                }
            ]
        }
    ]
}
```
Aber wenn ich das richtig sehe kommt nur Soc.

mark-sch, 2020-11-27:
Willkommen im Club, VW hat einfach derzeit massive Probleme die sie nicht in den Griff bekommen. Es betrifft auch nicht ID Fahrzeuge wie den e-Golf. Bei mir ist es aktuell so, dass ich Werte bekomme sobald der Wagen im WLAN ist. Wird dagegen die OCU unit mit interner eSim genutzt, dann deaktivieren sich die Dienste bis im WLAN alles neu in Betrieb genommen wird. Mit Odis wurde getestet und ein Ticket aufgemacht, Wolfsburg hat sich bisher noch nicht zu dem Problem geäußert, es gibt Überlegungen aus Verzweiflung die 5F und OCU unit komplett zu tauschen.

andig, 2020-11-27:
@StefanSchoof hab ich gesehen- dieses API scheint aber einen anderen Loginflow zu erfordern als den den wir heute haben? Oder kommst Du (Kommandozeile, curl....) mit einem vorhandenen Access Token da irgendwie ran? Wie?

StefanSchoof, 2020-11-27:
Habe mal ein bisschen Code geschrieben, um den Order Status abzurufen und in der Hoffnung, dass es irgendwann für evcc gebraucht sogar in Go:
https://github.com/StefanSchoof/vwidentity 

Damit sollte man auch den fuel status abfragen können.

andig, 2020-11-27:
Genial! Das bekommen wir doch irgendwie integriert? Wissen wir wann welches API genutzt werden muss? Trial and error?

StefanSchoof, 2020-11-28:
Die `fuel/status` URL sieht man, wenn man den Netzwerk Tab auf myvolkswagen bebochtet. Ich konnte da aber auch keine anderen hilfreiche URLs beobachten. Ich habe irgendwo gelesen, dass VW angeblich für die IDs keine Webseite in Planung hat sondern nur die App. Um alle api URLs zu bekommen müsste man die App beobachten. Aber das ist ja durch Verschlüsselung und System Zertifikate nicht mehr ganz einfach.

goebelmeier, 2020-11-28:
@andig wenn dir das irgendwie hilft nutz auch mal die dir bekannten Credentials, ob damit vielleicht auch die "legacy" Fahrzeuge funktionieren. Wenn ich was testen soll kannst du auch gerne rufen.

andig, 2020-11-29:
@StefanSchoof check, habs gefunden. Wie bist Du auf die `w1hub-backend` gekommen? Die finde ich im VW Portal nirgendwo?

andig, 2020-11-29:
Nahe dran, aber noch nicht ganz. Ich schaffe es nciht den Portalrequest zu imitieren:

    GET /vehicles/WVW.../fuel/status HTTP/1.1
    Host: cardata.apps.emea.vwapps.io
    Accept: applicaton/json
    Authorization: Bearer eyJraWQiOiJ...
    Referer: https://www.volkswagen.de/
    User-Id: a30800b5-...

    HTTP/2.0 403 Forbidden
    Content-Length: 101
    Access-Control-Allow-Origin: *
    Content-Type: application/json;version=1
    Date: Sun, 29 Nov 2020 11:43:05 GMT
    Vcf-Trace-Id: ce91a8a4-...
    X-Vcap-Request-Id: e5ce64d4-...

    {"error":{"code":4004,"message":"The performed call cannot be executed due to missing user consent"}}

StefanSchoof, 2020-11-29:
Auf der myvolkswagen Seite auf die Fahrzeuge und dann zu dem ID.3 und dann auf weitere Details. Angezeigt werden da keine der Infos.

andig, 2020-11-29:
> Auf der myvolkswagen Seite auf die Fahrzeuge und dann zu dem ID.3 und dann auf weitere Details. Angezeigt werden da keine der Infos.

Gefunden. Es scheint als würde dieser Aufruf mittels des Access Tokens funktionieren. Für weitere APIs erfolgt allerdings nochmal ein Token Exchange unter `myvw-idk-token-exchanger.apps.emea.vwapps.io`. Den bekomme ich nicht nachgestellt- landet immer in `403 Forbidden`. Hat jemand eine Idee?

andig, 2020-11-29:
Denke ich habs gefunden. Stay tuned....

andig, 2020-11-30:
Wollt ihr den PR mal testen? Wir müssten rausfinden was passiert wenn das Token ausläuft- dazu habe ich leider kein API gefunden.

obiwan007, 2020-11-30:
Wie kann ich denn von deinem feature branch installieren?

obiwan007, 2020-12-01:
Ich kann das testen, habe aber leider keinen Charger/Wallbox. Gibt es eine Config die ich da nehmen kann um nur den SOC abzurufen vom Fahrzeug?

andig, 2020-12-01:
evcc vehicle

fsdschmidt, 2020-12-01:
Gibt es ein osx Assembly von dem build? Kann ihn nicht finden und hab hier leider keine build Umbegung für das Projekt.

andig, 2020-12-01:
> Gibt es ein osx Assembly von dem build?

Nein, PRs musst Du leider lokal bauen. Wie ist in der README beschrieben.

obiwan007, 2020-12-01:
I got it compiled on my rock64. 
Das access_token konnte abgerufen werden.
Aber natürlich alle folgecalls nicht mehr da hier das VW connect api und nicht connect ID genutzt wird scheinbar.

TRACE 2020/12/01 10:15:53 POST https://mbboauth-1d.prd.ece.vwg-connect.com/mbbcoauth/mobile/oauth2/v1/token
{"access_token":"eyJraWQiOiJNQkIwMSIsImFsZyI6IlJ...,"expires_in":3599,"scope":"fal"}

[vw    ] TRACE 2020/12/01 10:15:55 GET https://msg.volkswagen.de/fs-car/bs/batterycharge/v1/VW/DE/vehicles/WVWZZZE1ZL******/charger
{"error":{"errorCode":"batterycharge.auth.forbidden","description":"The referenced vehicle with VIN: WVWZZZ***** could not be found or has no association to the given User"}}

Hier meine config.yaml
vehicles:
- name: vw
  type: vw
  title: ID.3
  capacity: 60 # kWh

StefanSchoof, 2020-12-01:
Der PR fügt den `type: id` hinzu. Das musst du in deiner config anpassen.

obiwan007, 2020-12-01:
[id    ] TRACE 2020/12/01 11:13:18 GET https://cardata.apps.emea.vwapps.io/vehicles//fuel/status
301 Moved Permanently
<a href="/vehicles/fuel/status">Moved Permanently</a>.
[id    ] TRACE 2020/12/01 11:13:18 GET https://cardata.apps.emea.vwapps.io/vehicles/fuel/status
404 Not Found
404 page not found
Capacity:      60kWh
State:         unexpected status: 404
Vehicle range: unexpected status: 404

obiwan007, 2020-12-01:
https://cardata.apps.emea.vwapps.io/vehicles/WVWZZZE1Z*****/fuel/status
wird vom myvolkswagen portal genutzt. Die FIN fehlt hier scheinbar.

fsdschmidt, 2020-12-01:
Kann es leider nicht testen, bekomme die Buildumgebung nicht zum Laufen. Liegt vielleicht daran, dass ich bei den MacPorts nur Go 1.15 finde und das geforderte 1.13 nur noch recht aufwendig installiert werden kann. NodeJs hab ich in Version 14.15.1 installiert. Das make von evcc schlägt bei mir fehl: 

rm -rf dist/
go install github.com/mjibson/esc
go: downloading github.com/mjibson/esc v0.2.0
go: downloading golang.org/x/tools v0.0.0-20200925191224-5d1fdd8fa346
go: downloading github.com/pkg/errors v0.9.1
go: downloading golang.org/x/mod v0.3.0
go: downloading golang.org/x/xerrors v0.0.0-20200804184101-5ec99f83aff1
go install github.com/golang/mock/mockgen
go: downloading github.com/golang/mock v1.4.4
npm ci

Usage: npm <command>

where <command> is one of:
    access, adduser, bin, bugs, c, cache, completion, config,
    ddp, dedupe, deprecate, dist-tag, docs, edit, explore, get,
[...]

Falls noch wer einen Tipp hat, wie ich das make zum Laufen bekomme, kann ich es versuchen. Ansonsten kann ich den Test leider nicht unterstützen.

andig, 2020-12-01:
@obiwan007 könntest Du mir Zugangsdaten unter cpuidle@gmx.de zur Verfügung stellen? Das Fahrzeug sollte tatsächlich auch ohne Angabe der VIN erkannt werden.

obiwan007, 2020-12-01:
Wenn ich 
`v := &ID{
		embed:    &embed{cc.Title, cc.Capacity},
		Helper:   request.NewHelper(log),
		user:     cc.User,
		password: cc.Password,
		VIN:      cc.VIN,
	}`

einbaue klappt es.

andig, 2020-12-01:
> https://github.com/andig/evcc/issues/485#issuecomment-736391403 Usage: npm

/cc @naltatis

andig, 2020-12-01:
> Wenn ich ... einbaue klappt es.

Was/ wo/ warum/ um welches Problem zu lösen?

obiwan007, 2020-12-01:
@fsdschmidt du solltest mal die nodejs version checken. Die ist möglicherweise bei dir zu alt. Du kannst auch mit go 1.15 das Ganze zum Laufen bringen.
Ich habe auch so diverse buildfehler (etc, mockgen) gehabt und durch hinzufügen von $GOPATH/bin zu $PATH fixen können.
Es reicht auch wenn du "make build" eingibst (keine Ahnung ob die assets wirklich gebraucht werden).

obiwan007, 2020-12-01:
> > Wenn ich ... einbaue klappt es.
> 
> Was/ wo/ warum/ um welches Problem zu lösen?

Im file id.go damit die FIN gefunden wird und damit  https://cardata.apps.emea.vwapps.io/vehicles/WVWZZZE1ZLP001643/fuel/status korrekt befüllt wird. 
Ansonsten wird vehicles//fuel/status requested.
v.VIN ist leer.

fsdschmidt, 2020-12-01:
Stimmt "make build" funzt. Jetzt hab ich den branch lauffähig. Danke für den Tipp!

Hab jetzt type:id in meiner config und bekomme nun keinen 403 mehr, sondern einen 404. Und auch ich hab das Moved Permantently. 

Aber mit der zusätzlichen VIN-Erweiterung von @obiwan007, boom hab ich meinen aktuellen SOC im Webinterface und auf der Konsole sehe ich auch die korrekte Reichweite. Klasse!

andig, 2020-12-01:
> Wenn ich ... einbaue klappt es.

Ahhh. Du kannst auch einfach

    vin: XYZ

in der Config machen. Das heisst, dass die automatische Fahrzeugerkennung beim ID nicht funktioniert. 
Ich bräuchte Testdaten um rauszufinden warum.

StefanSchoof, 2020-12-01:
Ich denke das Problem ist das in https://github.com/andig/evcc/pull/492/files#diff-278cb1e6e4d76d244a4d5bf469346004111535cbf4132cac8abe5b0a1db893a0R85 `v.VIN` nur geschrieben wird wenn keine vin in der config ist. Wenn eine vin in der config ist wird `v.VIN` nirgends geschrieben

obiwan007, 2020-12-01:
@andig sorry, aber ich kann dir leider so eben mal nicht den Account aufmachen bei mir. Da sind so viele Dienste drauf registriert, das ich vermutlich nicht mal mehr beim ID3 die Tür aufbekomme wenn ich dort das Passwort ändere :)

andig, 2020-12-01:
> Ich denke das Problem ist das in https://github.com/andig/evcc/pull/492/files#diff-278cb1e6e4d76d244a4d5bf469346004111535cbf4132cac8abe5b0a1db893a0R85 v.VIN nur geschrieben wird wenn keine vin in der config ist. Wenn eine vin in der config ist wird v.VIN nirgends geschrieben

Wir in der Folgezeile geschrieben: https://github.com/andig/evcc/pull/492/files#diff-278cb1e6e4d76d244a4d5bf469346004111535cbf4132cac8abe5b0a1db893a0R86

Ich hab noch andere Credentials und schaue später nochmal.

StefanSchoof, 2020-12-01:
Aber das nur wenn `cc.VIN == ""`, wenn die Config die VIN kommt er da nicht hin.

andig, 2020-12-01:
Treffer versenkt! Danke 🙏🏻

fsdschmidt, 2020-12-01:
> Wollt ihr den PR mal testen? Wir müssten rausfinden was passiert wenn das Token ausläuft- dazu habe ich leider kein API gefunden.

Um mal auf die ursprüngliche Frage zurück zu kommen. Eine Stunde läuft der Abruf wunderbar, dann kommt ein 401. Hier ein Ausschnitt aus dem Trace:

[cache ] TRACE 2020/12/01 15:09:03 socCharge: 70
[cache ] TRACE 2020/12/01 15:09:03 chargeEstimate: 17h25m3s
[cache ] TRACE 2020/12/01 15:09:03 chargeRemainingEnergy: 6444.444444444445
[cache ] TRACE 2020/12/01 15:09:03 range: 265
[exec  ] TRACE 2020/12/01 15:09:03 /bin/sh -c echo 1: 1
[lp-1  ] WARN 2020/12/01 15:09:03 charger out of sync
[exec  ] TRACE 2020/12/01 15:09:03 /bin/sh -c echo false: false
[lp-1  ] DEBUG 2020/12/01 15:09:03 max charge current: 4A = 6A + -2A (370W @ 1p)
[lp-1  ] DEBUG 2020/12/01 15:09:03 target charge current: 0A
[cache ] TRACE 2020/12/01 15:09:03 remoteDisabled: 
[modbus] TRACE 2020/12/01 15:09:08 modbus: closing connection due to idle timeout: 5.004071387s
[modbus] TRACE 2020/12/01 15:09:08 modbus: closing connection due to idle timeout: 5.004336051s
[site  ] DEBUG 2020/12/01 15:09:13 ----
[modbus] TRACE 2020/12/01 15:09:13 modbus: send 01 7f 00 00 00 06 47 03 9c 88 00 32
[modbus] TRACE 2020/12/01 15:09:13 modbus: recv 01 7f 00 00 00 67 47 03 64 00 9f 00 32 00 37 00 35 ff fe ff ff ff ff ff ff 09 20 08 f0 08 eb ff ff 00 eb 00 00 13 87 ff fe 01 71 00 00 01 1b 00 00 03 e8 ff fd 00 20 8c 0c 00 00 ff ff 00 00 ff ff 00 00 01 34 00 00 01 9b 01 13 01 29 00 00 ff ff 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2020/12/01 15:09:13 Power: 235.00W
[site  ] DEBUG 2020/12/01 15:09:13 pv power: 235W
[cache ] TRACE 2020/12/01 15:09:13 pvPower: 235
[modbus] TRACE 2020/12/01 15:09:13 modbus: send 02 ee 00 00 00 06 47 03 9c 87 00 69
[modbus] TRACE 2020/12/01 15:09:13 modbus: recv 02 ee 00 00 00 d5 47 03 d2 80 00 00 84 00 6c 00 59 ff fe 80 00 5b 33 59 f2 59 44 80 00 80 00 80 00 80 00 ff fe 13 86 ff fe 00 25 00 1a 00 03 00 07 00 01 00 2d 00 1b 00 13 00 0f 00 01 ff e7 00 07 ff ed ff f3 00 01 03 40 03 c3 00 b0 01 ec ff fd ff f4 9b a5 ff fc f1 7a ff fa b7 d6 ff fa a5 b8 00 11 8b 1a 00 0d 4e 40 00 04 8d 93 00 01 fb e3 00 00 ff f2 d6 42 ff fc 4a 1a ff f9 f3 0e ff f9 ef b4 00 14 02 bf 00 0d b7 36 00 07 b1 8d 00 04 09 25 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2020/12/01 15:09:13 Power: 370.00W
[site  ] DEBUG 2020/12/01 15:09:13 grid power: 370W
[site  ] DEBUG 2020/12/01 15:09:13 site power: 370W
[cache ] TRACE 2020/12/01 15:09:13 gridPower: 370
[cache ] TRACE 2020/12/01 15:09:13 mode: pv
[modbus] TRACE 2020/12/01 15:09:13 modbus: send 02 ef 00 00 00 06 47 03 9c 87 00 69
[modbus] TRACE 2020/12/01 15:09:13 modbus: recv 02 ef 00 00 00 d5 47 03 d2 80 00 00 84 00 6c 00 59 ff fe 80 00 5b 33 59 f2 59 44 80 00 80 00 80 00 80 00 ff fe 13 86 ff fe 00 25 00 1a 00 03 00 07 00 01 00 2d 00 1b 00 13 00 0f 00 01 ff e7 00 07 ff ed ff f3 00 01 03 40 03 c3 00 b0 01 ec ff fd ff f4 9b a5 ff fc f1 7a ff fa b7 d6 ff fa a5 b8 00 11 8b 1a 00 0d 4e 40 00 04 8d 93 00 01 fb e3 00 00 ff f2 d6 42 ff fc 4a 1a ff f9 f3 0e ff f9 ef b4 00 14 02 bf 00 0d b7 36 00 07 b1 8d 00 04 09 25 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2020/12/01 15:09:13 Power: 370.00W
[lp-1  ] DEBUG 2020/12/01 15:09:13 charge power: 370W
[cache ] TRACE 2020/12/01 15:09:13 chargePower: 370
[cache ] TRACE 2020/12/01 15:09:13 chargeCurrent: 6
[cache ] TRACE 2020/12/01 15:09:13 chargedEnergy: 440.6168524704445
[cache ] TRACE 2020/12/01 15:09:13 chargeDuration: 1h2m0s
[exec  ] TRACE 2020/12/01 15:09:13 /bin/sh -c echo C: C
[lp-1  ] DEBUG 2020/12/01 15:09:13 charger status: C
[cache ] TRACE 2020/12/01 15:09:13 connected: true
[cache ] TRACE 2020/12/01 15:09:13 charging: true
[id    ] TRACE 2020/12/01 15:09:13 GET https://cardata.apps.emea.vwapps.io/vehicles/***/fuel/status
401 Unauthorized
{"error":{"code":4003,"message":"The performed call is unauthorized"}}
[lp-1  ] WARN 2020/12/01 15:09:13 updating soc failed: unexpected status: 401
[lp-1  ] ERROR 2020/12/01 15:09:13 vehicle error: unexpected status: 401
[cache ] TRACE 2020/12/01 15:09:13 socCharge: -1
[cache ] TRACE 2020/12/01 15:09:13 chargeEstimate: -1ns
[cache ] TRACE 2020/12/01 15:09:13 range: -1
[exec  ] TRACE 2020/12/01 15:09:13 /bin/sh -c echo 1: 1
[lp-1  ] WARN 2020/12/01 15:09:13 charger out of sync
[exec  ] TRACE 2020/12/01 15:09:13 /bin/sh -c echo false: false
[lp-1  ] DEBUG 2020/12/01 15:09:13 max charge current: 4A = 6A + -2A (370W @ 1p)
[lp-1  ] DEBUG 2020/12/01 15:09:13 target charge current: 0A
[cache ] TRACE 2020/12/01 15:09:13 remoteDisabled:

andig, 2020-12-01:
Bitte nochmal testen. Token refresh sollte (tm) jetzt auch klappen.

obiwan007, 2020-12-02:
Nach einer Stunde kommt leider wieder

[lp-1  ] DEBUG 2020/12/02 10:01:10 charge power: 0W
[lp-1  ] DEBUG 2020/12/02 10:01:10 charger status: B
[lp-1  ] WARN 2020/12/02 10:01:10 updating soc failed: unexpected status: 401
[lp-1  ] ERROR 2020/12/02 10:01:10 vehicle error: unexpected status: 401
[lp-1  ] DEBUG 2020/12/02 10:01:10 max charge current: 0A = 0A + 0A (0W @ 3p)
[lp-1  ] DEBUG 2020/12/02 10:01:10 target charge current: 6A

andig, 2020-12-02:
Bitte nochmal neuer Versuch

obiwan007, 2020-12-02:
Sieht so aus als ob das refreshing nun funktioniert. Bislang keine Fehler.
Hast Du eine Docu gefunden für die ID API? Die aktuelle WeConnect ID app gibt ja schon Ausblick af neue Funktionen.
Spannend wäre z.B. noch das Laden zu starten/stoppen bzw. den Ladezustand abzufragen.

fsdschmidt, 2020-12-02:
Bei mir gibt es noch ein Problem mit dem refreshing. Immer zum Zeitpunkt des refresh nach einer Stunde funktionieren die Abfragen für ca. 5min nicht. Dann fängt es sich wieder und es funktioniert wieder für eine Stunde.

andig, 2020-12-02:
Logfile?

fsdschmidt, 2020-12-02:
Das hatte ich dir schon per Mail geschickt.

fsdschmidt, 2020-12-03:
Hab es die ganze Nacht laufen lassen, funzt jetzt dauerhaft, bis auf die stündlichen Verbindungsabbrüche. 

@obiwan007: Hast du das Problem nicht?

Mal eine andere Frage am Rande: Gibt es zufällig in der REST Schnittstelle einen Endpunkt um SOC und Restreichweite eines Fahrzeugs abzufragen? Laut Doku nicht, aber vielleicht kann evcc ja schon mehr. :)

andig, 2020-12-03:
In der Doku gibts das /state api?!

obiwan007, 2020-12-03:
@fsdschmidt doch ich hab die Abbrüche auch. Sehe ich im IOBroker mit Flotedit ganz gut. Jede Stunde werden die Werte mit 0 geschrieben.
Die SOC und Restreichweite wird per MQTT schon weitergegeben. Ob per REST weiss ich nicht.
mqtt.0.evcc.loadpoints.1.socCharge
mqtt.0.evcc.loadpoints.1.range

Der Range ist fast immer mit dem SOC synchron. Hier läuft offenbar nur eine leichte zeitversetzte Mittelung im Fahrzeug.
Auch ändert sich der SOC nur in 10% Schritten. Denke das Fahrzeug sendet das auch nur so. Funktioniert erstaunlicherweise auch in der Tiefgarage mit schlechtem Empfang.

andig, 2020-12-03:
Für soc in 10% Schritten könnt ihr mal estimate: true setzen.

fsdschmidt, 2020-12-03:
Ach man sollte die API auch einfach mal abrufen. Klar im state Endpunkt wird ja ein JSON mit allen möglich Infos geliefert. Sorry, hatte irgendwie nicht erwartet dass da so viel beim state zurück kommt. Danke! Genau was ich gesucht habe. :)

andig, 2020-12-03:
Ok. Man sollte auch einen neuen Token nicht einfach nur abrufen, sonder auch benutzen. Jetzt sollte es gehen!

obiwan007, 2020-12-03:
[main  ] FATAL 2020/12/03 08:53:33 cannot create vehicle 'ID.3': cannot create type 'id': unexpected status: 401

fsdschmidt, 2020-12-03:
@obiwan007: Hatte ich gestern Nachmittag auch. Melde dich mal unter volkswagen.de an, da gab es gestern Nachmittag eine Änderung und die brauchte erneute Zustimmung. Ist vielleicht auch etwas worauf evcc noch mit einem Hinweis reagieren könnte.

Danke @andig, aktuellen Stand probiere ich aus.

StefanSchoof, 2020-12-03:
Das mit den Änderungen an den Terms and conditions scheint aktuell öfter vor zu kommen. Gefühlt alle 2 Wochen gibt es eine neue Fassung.

andig, 2020-12-03:
> https://identity.vwgroup.io/signin-service/v1/4fb52a96-2ba3-4f99-a3fc-583bb197684b@apps_vw-dilab_com/terms-and-conditions?relayState=9d6b24d3178870a59d6171780d132c84a55a6d9e&updatedDataPrivacy=true

Ich möchte die nicht für Euch akzeptieren, aber eine Warning könnte gehen.

StefanSchoof, 2020-12-03:
Ja, das eine Warnung finde ich gut.

[22 more replies]

PR #492 (2020-12-08): Add VW ID api
Fix #485. Usage:

    type: id

Token refresh is still missing
closed 2020-12-08 by andig