evcc-io/evcc #488: neue Nissan Leaf API aktualisiert sich nicht

martinez81, 2020-11-27:
**Describe the bug**
Der SOC und der Status vom Auto aktualisiert sich nicht von selbst.
Erst wenn man die App (vom Smartphone) vom Auto öffnet und diese aktualisiert, wird auch der SOC und der Status in EVCC aktualisiert.

Im konkreten Fall war der Leaf **nicht** angesteckt. Das Model 3 wurde geladen. Da EVCC aber beim Leaf den Status B erkannt hat, war laut EVCC der Leaf angesteckt. Dies wurde auch auf der Webseite angezeigt.

**Expected behavior**
Wenn das Auto an die Wallbox angesteckt wird sollte das SOC und der Status ca. alle 5 Min neu vom Auto übertragen werden.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
LEAF war nicht angesteckt (Model 3 hat geladen)
[lp-1  ] DEBUG 2020/11/27 20:02:58 charge power: 2240W
[lp-1  ] DEBUG 2020/11/27 20:02:59 charger status: C
[lp-1  ] DEBUG 2020/11/27 20:02:59 vehicle status: B (Leaf)
[lp-1  ] DEBUG 2020/11/27 20:02:59 vehicle soc: 45%

LEAF in der App aktualisiert!
DEBUG 2020/11/27 20:04:11 charge power: 0W
[lp-1  ] DEBUG 2020/11/27 20:04:11 charger status: B
[lp-1  ] DEBUG 2020/11/27 20:04:12 vehicle status: A (Leaf)
[lp-1  ] DEBUG 2020/11/27 20:04:12 vehicle soc: 45%
```

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions -test-
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: sma-grid
  type: sma
  serial: 300613xxxx
- name: sma-pv
  type: modbus
  model: sunspec
  uri: 192.168.0.165:502
  id: 126

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: GO-eCharger
  type: go-e # Go-e charger
  uri: http://192.168.0.131

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
- name: tesla
  type: tesla
  title: Model 3
  capacity: 79
  clientid: xxx
  clientsecret: xxx
  user: xxx # your email address used on Tesla website
  password: xxx # your password used on Tesla website password
  vin: xxx # the VIN of you car (can be obtained from Tesla website)
  cache: 5m

- name: nissan
  type: nissan
  title: Leaf # display name for UI
  capacity: 40 # kWh
  user: xxx # user
  password: xxx # password
  region: DE # carwings region, leave empty for Europe
  cache: 5m # cache API response

# site describes the EVU connection, PV and home battery
site:
  title: xxx # display name for UI
  meters:
    grid: sma-grid # grid meter
    pv: sma-pv # pv meter
#    battery: battery # battery meter
#  prioritySoC: 60 # give home battery priority up to this soc (0 to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: GO-eCharger # display name for UI
  charger: GO-eCharger # charger
#  meters:
#    charge: charge # charge meter
#  vehicle: audi
  vehicles: # use if multiple vehicles allowed to charge on this loadpoint
  - nissan
  - tesla
#  - nissan
  mode: now
  soc:
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 80 # always charge to 100%
    alwaysUpdate: false # set true to update vehicle soc even when disconnected
    estimate: false # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 30
    - 50
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: now # switch back to pv mode
    targetSoC: 80 # charge to 100%
  phases: 3 # ev phases (default 3)
  sensitivity: 1 # current raise/lower step size (default 10A)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 500 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

# EVCC can integrate itself with Home Energy Management Systems.
# At this time, the SMA Home Manager (SHM) is the only supported
# system. To enable add
hems: 
  type: sma

# mqtt message broker
mqtt:
  # broker: localhost:1883
  # topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>

</pre></details>

Kann ich noch irgend welche Daten liefern?

Gruß
Martin

andig, 2020-11-27:
> Im konkreten Fall war der Leaf nicht angesteckt. Das Model 3 wurde geladen. Da EVCC aber beim Leaf den Status B erkannt hat, war laut EVCC der Leaf angesteckt. Dies wurde auch auf der Webseite angezeigt.

Tja.

> Wenn das Auto an die Wallbox angesteckt wird sollte das SOC und der Status ca. alle 5 Min neu vom Auto übertragen werden.

Das ist der Fall. 

Die Konfiguration wäre interessant, ist im Ticket aber leider zerschossen.

martinez81, 2020-11-27:
Hier das yaml File.

[evcc.yaml.zip](https://github.com/andig/evcc/files/5609415/evcc.yaml.zip)

Wo finde ich das Logfile?

mark-sch, 2020-11-27:
> Hier das yaml File.
> 
> [evcc.yaml.zip](https://github.com/andig/evcc/files/5609415/evcc.yaml.zip)
> 
> Wo finde ich das Logfile?

./evcc --log trace

andig, 2020-11-27:
Wie startest Du‘s denn? Systemd? Docker? Ohne die notwendigen Informationen ist das ein Suchspiel...

martinez81, 2020-11-28:
Ich starte auf ein Raspberry per systemctl.
```
$ sudo systemctl start evcc.service
```
<details><summary>sudo nano /etc/systemd/system/evcc.service<code>evcc.yaml</code>:</summary><br><pre>

[Unit]
Description=evcc
After=syslog.target
[Service]
ExecStart=/usr/local/bin/evcc --config /etc/evcc.yaml --log error
Restart=always
[Install]
WantedBy=multi-user.target

</pre></details>

Für den Trace starte ich jetzt einmal so:
(Da kommt extrem viel)
<details><summary>$ sudo evcc --config /etc/evcc.yaml --log trace</code>:</summary><br><pre>

pi@Docker:~ $ sudo evcc --config /etc/evcc.yaml --log trace
[main  ] INFO 2020/11/28 09:15:54 evcc 0.33 (403e953)
[main  ] INFO 2020/11/28 09:15:54 using config file /etc/evcc.yaml
[main  ] INFO 2020/11/28 09:15:54 listening at 0.0.0.0:7070
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 01 00 00 00 06 7e 03 9c 40 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 01 00 00 00 07 7e 03 04 53 75 6e 53
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 02 00 00 00 06 7e 03 9c 42 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 02 00 00 00 07 7e 03 04 00 01 00 42
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 03 00 00 00 06 7e 03 9c 86 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 03 00 00 00 07 7e 03 04 00 0b 00 0d
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 04 00 00 00 06 7e 03 9c 95 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 04 00 00 00 07 7e 03 04 00 0c 00 62
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 05 00 00 00 06 7e 03 9c f9 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 05 00 00 00 07 7e 03 04 00 67 00 32
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 06 00 00 00 06 7e 03 9d 2d 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 06 00 00 00 07 7e 03 04 00 78 00 1a
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 07 00 00 00 06 7e 03 9d 49 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 07 00 00 00 07 7e 03 04 00 79 00 1e
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 08 00 00 00 06 7e 03 9d 69 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 08 00 00 00 07 7e 03 04 00 7a 00 2c
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 09 00 00 00 06 7e 03 9d 97 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 09 00 00 00 07 7e 03 04 00 7b 00 18
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 0a 00 00 00 06 7e 03 9d b1 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 0a 00 00 00 07 7e 03 04 00 7c 00 18
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 0b 00 00 00 06 7e 03 9d cb 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 0b 00 00 00 07 7e 03 04 00 7e 00 40
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 0c 00 00 00 06 7e 03 9e 0d 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 0c 00 00 00 07 7e 03 04 00 7f 00 0a
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 0d 00 00 00 06 7e 03 9e 19 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 0d 00 00 00 07 7e 03 04 00 80 00 0e
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 0e 00 00 00 06 7e 03 9e 29 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 0e 00 00 00 07 7e 03 04 00 83 00 40
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 0f 00 00 00 06 7e 03 9e 6b 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 0f 00 00 00 07 7e 03 04 00 84 00 40
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 10 00 00 00 06 7e 03 9e ad 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 10 00 00 00 07 7e 03 04 00 a0 00 80
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 11 00 00 00 06 7e 03 9f 2f 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 11 00 00 00 07 7e 03 04 00 81 00 3c
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 12 00 00 00 06 7e 03 9f 6d 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 12 00 00 00 07 7e 03 04 00 82 00 3c
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 13 00 00 00 06 7e 03 9f ab 00 02
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 13 00 00 00 07 7e 03 04 ff ff 00 00
[modbus] TRACE 2020/11/28 09:15:54 modbus: send 00 14 00 00 00 06 7e 03 9c 44 00 42
[modbus] TRACE 2020/11/28 09:15:54 modbus: recv 00 14 00 00 00 87 7e 03 84 53 4d 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 53 6f 6c 61 72 20 49 6e 76 65 72 74 65 72 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 33 34 37 00 00 00 00 00 00 00 00 00 00 00 00 33 2e 31 30 2e 31 31 2e 52 00 00 00 00 00 00 00 33 30 30 36 35 34 37 36 32 33 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 80 00
[nissan] TRACE 2020/11/28 09:15:56 POST https://prod.eu.auth.kamereon.org/kauth/json/realms/root/realms/a-ncb-prod/authenticate
200 
{"authId":"eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJvdGsiOiJzN2VyOXQ0MGxiMDIyOTRrcmhtbjAydWZ1diIsInJlYWxtIjoibz1hLW5jYi1wcm9kLG91PXNlcnZpY2VzLGRjPWNvbmZpZyxkYz1rYW1lcmVvbixkYz1vcmciLCJzZXNzaW9uSWQiOiJBUUlDNXdNMkxZNFNmY3pLeUhaSFFhd1dLOWtNeTc0VFp4WlRoM183QUJvXzV3Zy4qQUFKVFNRQUNNRElBQWxOTEFCTXpNRE16TmpRMk16UXlOVEUxTmpFeU5qZ3lBQUpUTVFBQ01ERS4qIn0.0aOY6BZvw6-H_NrDbB505IXGWqjNqoXA64bwweHWBYk","template":"","stage":"LDAP1","header":"Sign in","callbacks":[{"type":"NameCallback","output":[{"name":"prompt","value":"User Name:"}],"input":[{"name":"IDToken1","value":""}]},{"type":"PasswordCallback","output":[{"name":"prompt","value":"Password:"}],"input":[{"name":"IDToken2","value":""}]}]}
[... cut ...] c6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"","rnm":"","rne":"","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":0,"mce":0,"mcs":"test.mosquitto.org","mcp":1883,"mcu":"","mck":"","mcc":0}
[cache ] TRACE 2020/11/28 09:16:38 chargeCurrent: 11
[cache ] TRACE 2020/11/28 09:16:38 remoteDisabled: 

</pre></details>

Hier nochmal als File:
[trace.txt.zip](https://github.com/andig/evcc/files/5610251/trace.txt.zip)

andig, 2020-11-29:
@martinez81 der Hinweis mit dem Logfile hat etwas in die Irre geführt. Was wir bräuchten wäre ein Logfile das zu dem Vorgang oben passt (genau der Zeitpunkt). Wenn Du systemd nutzt solltest Du an das Log mit `journalctl` kommen. Könntest Du nochmal schauen ob Du ein dazu passendes findest? Ohne wird's schwer.

martinez81, 2020-11-29:
Also über **journalctl** kommt meiner Meinung nach nichts sinnvolles. Tut mir leid.

[image]

Reproduzieren kann man das immer.
Der Status vom Leaf wird ohne das man über die (Smartphone) App die Daten aktualisiert nicht geändert. Und egal was der Leaf macht, wird nie das 2. Auto (Model 3) in EVCC angezeigt.
Gibt es Leute die 2 Autos an einer Wallbox betreiben?

andig, 2020-11-29:
@martinez81 wenn ich dich richtig verstehe, geht es hier nicht darum, dass es ein Problem mit mehrere Fahrzeugen gibt, sondern dass das Nissan api nicht aktualisiert?

Trifft das auf dartnissanconnect dann auch zu? Dann hab ich leider keine Idee 😫

Alex-ander-s, 2020-11-30:
@andig, @martinez81:
Ich selbst hole mir die Daten zum SOC mit evcc via script aus meiner postgres DB und verwende zum Befüllen dieser Datenbank meinen eigenen fork von https://github.com/jdhorne/pycarwings2.
Ich kann bestätigen, dass bei mir die SOC Daten in der Datenbank zum Teil auch nicht ankommen, wenn die Nissan Server, ich nenne es Mal "spezielle" Probleme haben.
Wenn ich die SOC Informationen in der Situation dann z.B. über die Android App "My Leaf" aktualisiere, kommen danach die Daten wieder an.
Die Nissan Server sind furchtbar buggy und die API verhält sich immer wieder mal anders.
Ich kann bestätigen, dass es vor allem gestern und heute wieder ziemliche Probleme mit der Nissan API gab.
Die postgres DB verwende ich um genau dieses Verhalten zu dokumentieren, da ich schon etliche Male Probleme mit den Nissan Servern hatte (deshalb auch mein Fork von pycarwings2)
Als Notiz am Rande...

premultiply, 2020-11-30:
Dass sich Vehicle-API komisch verhalten und unzuverlässig funktionieren gehört offensichtlich zum guten Ton bei den Fahrzeugherstellern. 🙄

andig, 2020-11-30:
OT @Alex-ander-s magst Dich mal bei Slack melden? Dein Feedback war interessant ;)

martinez81, 2020-11-30:
> @martinez81 wenn ich dich richtig verstehe, geht es hier nicht darum, dass es ein Problem mit mehreren Fahrzeugen gibt, sondern dass das Nissan api nicht aktualisiert?
> 
> Trifft das auf dartnissanconnect dann auch zu? Dann hab ich leider keine Idee 😫

Ja, die Nissan API funktioniert nicht wie erwartet.

andig, 2020-11-30:
> Ja, die Nissan API funktioniert nicht wie erwartet.

Ok, in dem Fall kann ich nichts tun :(. Du könntest nur beobachten, ob das ein temporäres Problem ist (gut) oder ein prinzipielles (schlecht). In letzterem Falle bliebe wieder nur rauszufinden wie die Nissan App dieses Problem eigentlich umgeht... (decompile, unpin certificates, trace...)

junibart, 2020-12-04:
> @martinez81 wenn ich dich richtig verstehe, geht es hier nicht darum, dass es ein Problem mit mehreren Fahrzeugen gibt, sondern dass das Nissan api nicht aktualisiert?
> Trifft das auf dartnissanconnect dann auch zu? Dann hab ich leider keine Idee 😫

Hallo, 
habe auch einen Leaf vom Ende 2019, also schon mit der neuen Abfragemethode. Ich kann das Problem aus dem ersten Beitrag bestätigen. Ich habe allerdings EVCC auf einem Raspi nur zu Testzwecken installiert, den Leaf als einziges vehicle konfiguriert und  die Fahrzeugdaten mit `./evcc vehicle` abgefragt. 
```
Capacity:      40kWh
State:         67%
Vehicle range: 167km
Charge status: B
Finish time:   not available
```
Das Ergebnis (state) ist in der Tat das, was nach der letzten, vorhergehenden Abfrage mit der Hersteller-App oder alternativ MyLeaf vom Fahrzeug abgerufen wurde. Offenbach wird das in der Nissan-Cloud gecached und das Neueinlesen vom Fahrzeug muss gesondert getriggert werden. Dafür spricht auch der Zeitablauf: 

- Das Ergebnis von `./evcc vehicle`  wird nach ca. 6 Sekunden ausgegeben. 

- Starte ich MyLeaf, wird nach ca. 5 Sekunden ebenfalls ein alter Wert ausgegeben, ein rotierendes Symbol zeigt jedoch, dass im Hintergrund noch eine Abfrage läuft. Erst nach über einer Minute (letzter Test: 63 Sekunden) wird der SoC auf den richtigen Wert aktualisiert und das rotierende Symbol bleibt stehen. 

- Gelegentlich nutze ich auch "Leaf Steuerung und Überwachung" (https://android.chk.digital/de/lcc/). Auch hier klappt das Abfragen des tatsächlichen SoC zuverlässig, dauert aber auch ca. 30 Sekunden pro Abfrage. Ich hatte mal mit dem Entwickler Kontakt, auch er setzt auf den bekannten  Libs auf. 

In der Nissan Connect App wird ebenfalls zunächst der alte Wert gezeigt (ggf. auch der, den MyLeaf einige Minuten vorher aus dem Fahrzeug geholt hat), hier muss man allerdings mehrfach mit einer Wischgeste die Daten aktualisieren, bis ein neuer Wert, der dann auch frisch vom Fahrzeug geholt wurde, angezeigt wird. 

Das ganze ist langwierig, aber ich habe bisher nicht bewusst erlebt, dass von den Nissan-Servern keine Daten kamen. Ich vermute, dass das Auslesen aktueller Daten aus dem Fahrzeug gesondert getriggert und das Ergebnis geduldig abgewartet werden muss. 
MyLeaf bekommt das zuverlässig hin, und die Sourcen sind zugänglich: https://gitlab.com/tobiaswkjeldsen/carwingsflutter.
Soweit ich das verstehe wird für ältere Fahrzeuge die dartcarwings-Lib, für Nordamerika die dartnissanconnectna -lib und für Fahrzeuge ab Mai 2019 die dartnissanconnect-lib verwendet... 
Leider enden da meine Kenntnisse, bin kein Entwickler.  Ich denke, man sollte die Sourcen nochmal durchsehen, wie in MyLeaf die Abfrage des Fahrzeugs angestoßen wird. 

Soweit erstmal, 

Gunnar

andig, 2020-12-05:
Danke @junibart das hilft! Bei https://gitlab.com/tobiaswkjeldsen/carwingsflutter findet sich:

    Future<bool> requestBatteryStatusRefresh() async {
      var response = await session.requestWithRetry(
          endpoint:
              '${session.settings['EU']['car_adapter_base_url']}v1/cars/$vin/actions/refresh-battery-status',
          additionalHeaders: <String, String>{
            'Content-Type': 'application/vnd.api+json'
          },
          params: {
            'data': {'type': 'RefreshBatteryStatus'}
          });

      return response.statusCode == 200;
    }

    Future<NissanConnectBattery> requestBatteryStatus() async {
      var response = await session.requestWithRetry(
          endpoint:
              '${session.settings['EU']['car_adapter_base_url']}v1/cars/$vin/battery-status',
          method: 'GET');

      return NissanConnectBattery(response.body);
    }

Damit ist auch klar, wie die Aktualisierung anzustossen ist. Da der Vorgang arg lange dauert würde ich das mal so einbauen, dass wir das Update im Hintergrund anstossen und später (d.h. nach Ablauf des Caches) den neuen Wert bekommen. Geht sicher auch eleganter, wäre aber erstmal einfach zu machen. Damit hängen wir immer 5min hinterher.

Jetzt die spannende Frage: wie oft darf man das aufrufen, ohne dass Nissan das als Abuse betrachtet? Wirklich alle 5min?!

andig, 2020-12-05:
Könnt ihr den PR mal probieren? Wenn die Cachezeit (5min) abgelaufen ist, wird ein Update vom Server angefordert und im Anschluss der Cache gelöscht. In meinen Tests kommt die Updateanforderung allerdings immer sehr schnell zurück- ich bin nicht sicher ob da auch schon der richtige Wert im API zur Verfügung steht?

martinez81, 2020-12-05:
Bzgl. der Abfrage:
abgefragt wird aber nur wenn ein Fahrzeug am Ladepunkte lädt.
Nicht das wirklich alle 5 min das Fahrzeug abgefragt wird. Das Fahrzeug sollte „schlafen“ können, wenn nicht geladen wird.

andig, 2020-12-05:
> Nicht das wirklich alle 5 min das Fahrzeug abgefragt wird. Das Fahrzeug sollte „schlafen“ können, wenn nicht geladen wird.

Das Verhalten ist immer identisch. Abgefragt wird wenn das Fahrzeug verbunden ist. Woher soll evcc wissen, dass der Zustand nur während des Ladens abgefragt werden soll? Wie finden wir dann raus ob überhaupt geladen werden sollte (minSoC etc)? 

Du kannst aber jederzeit der cache Wert erhöhen. Die Frage ist also: was ist sinnvoll/ gewünscht und funktioniert für alle Fahrzeuge?

martinez81, 2020-12-05:
> Das Verhalten ist immer identisch. Abgefragt wird wenn das Fahrzeug verbunden ist. Woher soll evcc wissen, dass der Zustand nur während des Ladens abgefragt werden soll? Wie finden wir dann raus ob [...]

Vorschlagen würde ich folgende Logik:
EVCC prüft ob ein Ladekabel am Ladepunkt und Auto angeschlossen ist und ob ein Ladestrom ≥ 6A vorhanden ist.
Nur dann wird der SOC vom Fahrzeug abgefragt.
So könnte das Fahrzeug auch bei angeschlossen Ladekabel, aber bei 0A  Ladestrom (z.B. weil keine Sonne, oder soll SOC erreicht) wieder schlafen. Der Abgefragte SOC verbleibt dann im Speicher des EVCC.

Ich kann vom Model 3 sprechen.
Wenn da die ganze Nacht das Ladekabel angeschlossen ist, und alle 5 oder 15 min der SOC abgefragt wird, obwohl nicht geladen wird, verliert das Model 3 2-5% SOC/Nacht. Nur wenn das Model 3 schläft ist der Computer aus.
Wenn das Model 3 durch SOC Abfragen im 
Standby gehalten wird, verbraucht das System ca. 300 Watt. Das ist über 12 Stunden ne Menge Holz.

Meine Erfahrung entnehme ich aus Teslamate. Damit logge ich das Model 3.

Ob z.B. der Leaf oder andere E-Autos ähnlich reagieren weiß ich nicht.

VG
Martin

andig, 2020-12-05:
Damit würden wir leider nie erfahren ob z.b. der minSoc unterschritten ist und wir einschalten müssten. Es fehlt also noch eine weitere Regel, wie oft im Status B abgefragt werden sollte.

martinez81, 2020-12-05:
> Damit würden wir leider nie erfahren ob z.b. der minSoc unterschritten ist und wir einschalten müssten. Es fehlt also noch eine weitere Regel, wie oft im Status B abgefragt werden sollte.

Meinst du damit: wenn das Auto ein paar Wochen am Lader hängt und dann irgendwann unter den minSOC fällt?
Dieser Zustand sollte ja eigentlich extrem selten auftreten.
Ggf. Könnte man 1x pro Woche den SOC abfragen und dann entsprechen nachladen.

Oder verstehe ich den minSOC falsch?

martinez81, 2020-12-05:
Nachtrag:
Mein Ladekabel anschließen sollte natürlich immer einmal der SOC abgefragt werden.

junibart, 2020-12-06:
> Könnt ihr den PR mal probieren? Wenn die Cachezeit (5min) abgelaufen ist, wird ein Update vom Server angefordert und im Anschluss der Cache gelöscht. In meinen Tests kommt die Updateanforderung [...]

Hallo, hab den PR installiert und getestet. 
Zum Test habe ich meine Testkonfiguration genutzt: nur das Fahrzeug konfiguriert und `./evcc vehicle` aufgerufen. 

Die Ausgabe des Kommandos kommt wie bisher nach ca. 6 Sekunden. Die erste Ausgabe zeigt den SoC, der irgendwann vorher vom Fahrzeug abgerufen wurde. 
Wird  `./evcc vehicle`  nach etwa einer Minute erneut aufgerufen, wird ebenfalls nach 6 Sekunden ein aktueller SoC ausgegeben, ohne dass zwischenzeitlich mit einer App extern "nachgeholfen" werden musste. Das Anstoßen des Abrufs vom Fahrzeug im Hintergrund funktioniert offenbar.  

Der PR funktioniert also grundsätzlich. 

Zur derzeit diskutierten Thematik "Häufigkeit SoC-Abruf" wäre mein Standpunkt: Beim Anstecken des Fahrzeugs einmalig, und während des Ladens nur alle 15 Minuten.  Wenn nicht geladen wird, gar nicht.

premultiply, 2020-12-06:
Hier muss man jetzt nur noch gucken wie man insbesondere das initiale Delay so handhabt, dass auch wirklich "aktiv" bzw. mit schnellem Polling auf das Eintreffen des aktuellen Status nach dem Verbinden gewartet wird diese Meldung nicht dann erst nach 15 Minuten eintrudelt.
Denn z. B. bei der ZOE dauert es auch etwas bis das Fahrzeug sein Statusupdate an das Backend übermittelt hat und dieser dann wiederum vom Client (hier EVCC) abgerufen werden kann. Hier ist der einzige Unterschied, dass das Fahrzeug bei bestimmten Events automatisch eine Meldung absetzt und beim Leaf dies offenbar zunächst ein Update angefordert werden muss, welches dann asynchron eintrifft.
Das Problem ist dort also nicht die API-Abfrage an sich, dann diese wird ja in diesem Fall doppelt gecached (im Backend und in EVCC), sondern das Kommando an das Fahrzeug die aktuellen Daten an das Backend zu senden.
Ich hab bei mir daher das Abfrageintervall während das Fahrzeug angeschlossen ist derzeit auf 1 Minute stehen um die Latenz durch den EVCC-internen Cache zu minimieren.
Hier gibt es daher bestimmt noch Optimierungspotenzial.
Mir scheint es aber sinnvoll bei bestimmten Events (z. B. Kabel angeschlossen/getrennt, Ladevorgang gestartet/beendet) seitens EVCC in jedem Fall neue Daten anzufordern und vielleicht für dann jeweils 5 Minuten im Minutentakt die API nach neuen Daten zu polllen.

martinez81, 2020-12-06:
> >
> Zur derzeit diskutierten Thematik "Häufigkeit SoC-Abruf" wäre mein Standpunkt: Beim Anstecken des Fahrzeugs einmalig, und während des Ladens nur alle 15 Minuten. Wenn nicht geladen wird, gar nicht.

Finde ich gut.

martinez81, 2020-12-06:
> Mir scheint es aber sinnvoll bei bestimmten Events (z. B. Kabel angeschlossen/getrennt, Ladevorgang gestartet/beendet) seitens EVCC in jedem Fall neue Daten anzufordern und vielleicht für dann jeweils 5 Minuten im Minutentakt die API nach neuen Daten zu polllen.

Ja, genau in diesen Zuständen ist das Auto sowieso wach, hier kann und sollte der SOC abgefragt werden.

andig, 2020-12-06:
> Die Ausgabe des Kommandos kommt wie bisher nach ca. 6 Sekunden. Die erste Ausgabe zeigt den SoC, der irgendwann vorher vom Fahrzeug abgerufen wurde.
Wird ./evcc vehicle nach etwa einer Minute erneut aufgerufen, wird ebenfalls nach 6 Sekunden ein aktueller SoC ausgegeben, ohne dass zwischenzeitlich mit einer App extern "nachgeholfen" werden musste. Das Anstoßen des Abrufs vom Fahrzeug im Hintergrund funktioniert offenbar.

Mit der neuen Logik aus #505 werde ich das so umbauen, dass die Aktualisierung dann "inline" erfolgt, das wird sich also nochmal ändern.

premultiply, 2020-12-06:
> Mit der neuen Logik aus #505 werde ich das so umbauen, dass die Aktualisierung dann "inline" erfolgt, das wird sich also nochmal ändern.

Ich gehe davon aus, dass es nicht ausreicht direkt nach dem lokalen Event die API abzufragen da das Fahrzeug im schlimmsten Fall auch erstmal aufwachen und dann seinen aktualisierten Zustand übermitteln muss. Diese Zeitspanne ist wohl nicht vorhersehbar. Nicht nur beim Leaf.
Daher der Vorschlag mit dem zeitbegrenzten Polling im Minutentakt nach einem lokalen Event.

andig, 2020-12-06:
Das Problem können wir immer noch lösen wenn es eintritt. Bis dahin gilt KISS.

premultiply, 2020-12-06:
Naja, es tritt heute schon ein. Daher musste ich das Intervall ja bislang generell auf Minimum reduzieren.
Hier wäre eine adaptive Lösung die dies nur für eine kurze Zeit nach Events tut daher besser.

andig, 2020-12-07:
PR ist so aktualisiert, dass die Aktualisierung jetzt "inline" erfolgt. In Anbetracht von #505 würde ich dennoch erst mergen wenn die Gefahr der API/Batterieüberlastung gebannt ist.

andig, 2020-12-07:
Wer es sich zutraut könnte jetzt #504 zusammen mit #505 testen.

andig, 2020-12-08:
Ich hab jetzt mal alles gemerged. Wenn es nicht funktioniert bitte hier melden, dann mach ich wieder auf.

closed 2020-12-08 by andig

joshTA2022, 2022-11-18:
Hallo, habe auch ein Problem beim Abrufen von SOC, 

ERROR 2022/11/18 22:15:17 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'carwings': login failed: Post "https://gdcportalgw.its-mo.com/api_v200413_NE/gdc/InitialApp_v2.php": dial tcp: lookup gdcportalgw.its-mo.com on 1.1.0.1:53: read udp 10.0.0.91:47365->1.1.0.1:53: i/o timeout

Bei der Konfiguration waren die Login Daten bei der Prüfung ok und es gab keine Fehlermeldung. 
Das ganze läuft auf einer Qnap NAS als Container. Liegt eventuell da das Problem....

andig, 2022-11-18:
Du verwendest Docker und hast vmtl. /etc gemounted; damit ist DNS kaputt. Hat nichts mit Nissan zu tun.