evcc-io/evcc #494: Homebattery SoC not displayed in UI when battery is empty

rivengh, 2020-11-30:
**Describe the bug**
Homebattery SoC not displayed in UI, when battery is empty. Note: Battery SoC is correctly displayed, if SoC is >= 1 %.

**Expected behavior**
Homebattery SoC displayed in UI with value  %.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.32 (f458524)
```

premultiply, 2020-11-30:
Den Effekt habe ich auch.

rivengh, 2020-11-30:
Ursache ist vermutlich das if-Statement `v-if="state.batterySoC"` im übergeordneten Div in assets/index.html:

```
  <div class="col-6 col-md-3 mt-3" v-if="state.batterySoC">
    <div class="mb-2 value">SoC
      <fa-icon class="text-primary" icon="battery-three-quarters" v-bind:class="{'text-primary':state.batteryPower>0,'text-muted':state.batteryPower<0}"></fa-icon>
    </div>
    <h2 class="value">
      {{state.batterySoC}} <small class="text-muted">%</small>
    </h2>
  </div>
```

andig, 2020-11-30:
Können wir annehmen- sofern eine Batterie konfiguriert ist- dass auch der SoC vorhanden wäre? Würde niemand ohne machen, oder?

premultiply, 2020-11-30:
Ja, ich denke davon kann man ausgehen.

PR #497 (2020-11-30): Don't treat battery soc 0 as missing value
Fix #494
closed 2020-11-30 by andig