evcc-io/evcc #495: Load management problem in "Sofort" mode

vic911, 2020-11-30:
**Describe the bug**
Load management in "Sofort" mode not working. Charge current not set according the grid limits and actual gride/site power consumption. Charge power shows a false value, not the one that received back from evse-wifi.

**To Reproduce**
Steps to reproduce the behavior:
1. config grid meter currents and power, loadpoint min and max current. no PV meter.
2. start "sofort" mode in evcc
3. check current at evse-wifi gui modbus registers (1000 and 1001)
4. check charge power in evcc gui or log

**Expected behavior**
evcc sends a current value to evse-wifi according to grid meter currents value and house limit. evcc shows charge power according the value received back from evse-wifi.

**EVCC details:**
Using config file: /config/evcc.yaml
starting evcc: 'evcc --config /config/evcc.yaml'
[main ] INFO 2020/11/30 11:20:06 evcc 0.33 (8ddfaa9)
[main ] INFO 2020/11/30 11:20:06 using config file /config/evcc.yaml
[main ] INFO 2020/11/30 11:20:06 listening at 0.0.0.0:7070
[mqtt ] INFO 2020/11/30 11:20:06 connecting evcc-174035675 at 192.168.1.200:1883
[mqtt ] DEBUG 2020/11/30 11:20:06 192.168.1.200:1883 connected
[site ] INFO 2020/11/30 11:20:07 site config:
[site ] INFO 2020/11/30 11:20:07 meters: grid ✓ pv — battery —
[site ] INFO 2020/11/30 11:20:07 grid: power ✓ energy — currents ✓
[lp-1 ] INFO 2020/11/30 11:20:07 loadpoint 1:
[lp-1 ] INFO 2020/11/30 11:20:07 mode:
[lp-1 ] INFO 2020/11/30 11:20:07 charger: power — energy — currents — timer ✓
[lp-1 ] INFO 2020/11/30 11:20:07 meters: charge —
[lp-1 ] INFO 2020/11/30 11:20:07 vehicles: ✓
[lp-1 ] INFO 2020/11/30 11:20:07 car 0: estimate — status — climate —
[site ] DEBUG 2020/11/30 11:20:07 ----
[site ] DEBUG 2020/11/30 11:20:07 grid power: -1288W
[site ] DEBUG 2020/11/30 11:20:07 site power: -1288W
[lp-1 ] DEBUG 2020/11/30 11:20:07 charge power: 0W
[lp-1 ] DEBUG 2020/11/30 11:20:07 charger status: B
[lp-1 ] DEBUG 2020/11/30 11:20:07 vehicle soc: 45%
[site ] DEBUG 2020/11/30 11:20:17 ----

grid/site power is minus because of PV.
mincurrent and maxcurrent is set to 6A and 18A in config at loadpoints section.

Starting the charge in mode "sofort" EVCC sets the evse wifi to 6A.

[site ] DEBUG 2020/11/30 11:20:47 grid power: -1310W
[site ] DEBUG 2020/11/30 11:20:47 site power: -1310W
[lp-1 ] DEBUG 2020/11/30 11:20:47 charge power: 0W
[lp-1 ] DEBUG 2020/11/30 11:20:47 charger status: C
[lp-1 ] INFO 2020/11/30 11:20:47 start charging ->
[lp-1 ] DEBUG 2020/11/30 11:20:47 vehicle soc: 45%
[site ] DEBUG 2020/11/30 11:20:57 ----
[site ] DEBUG 2020/11/30 11:20:57 grid power: -1301W
[site ] DEBUG 2020/11/30 11:20:57 site power: -1301W
[lp-1 ] DEBUG 2020/11/30 11:20:57 charge power: 12420W
[lp-1 ] DEBUG 2020/11/30 11:20:57 charger status: C
[lp-1 ] DEBUG 2020/11/30 11:20:57 vehicle soc: 45%

Why it is not set to 18A, since current limit of the grid/site allows it?
Charge power shows that it tried to set it to 18A (18A230V3Ph=12420W) but I've checked on evse wifi GUI , the register 1000 (Configured Current) and 1001 (Actual Current) was set to 6A.
using http getParameters at evsewifi returns 6A for actualCurrent, so it seems evcc not using it when calculating the charge power.

After updating the modbus register manually to 18A the grid power shows, that now it is ok. (maybe the battery is too cold, so the car capped it to 12A) 12A3Ph230V-PV=6299W
Charge power still shows 12420W in log.

[site ] DEBUG 2020/11/30 11:40:37 grid power: 6299W
[site ] DEBUG 2020/11/30 11:40:37 site power: 6299W
[lp-1 ] DEBUG 2020/11/30 11:40:37 charge power: 12420W
[lp-1 ] DEBUG 2020/11/30 11:40:37 charger status: C
[lp-1 ] DEBUG 2020/11/30 11:40:37 vehicle soc: 52%

andig, 2020-11-30:
> Why it is not set to 18A, since current limit of the grid/site allows it?

    [lp-1 ] DEBUG 2020/11/30 11:20:57 charge power: 12420W

Looks like 18A.

>  but I've checked on evse wifi GUI , the register 1000 (Configured Current) and 1001 (Actual Current) was set to 6A.

Could you share the entire log file (`--log trace`) and config? Please use the issue template!

vic911, 2020-11-30:
> Could you share the entire log file (`--log trace`) and config? Please use the issue template!

I will try it. I'm using the HA addon, how can I input this trace setting?

andig, 2020-11-30:
> I will try it. I'm using the HA addon, how can I input this trace setting?

Use the `evcc.yaml` config file. Set `log: trace`.

vic911, 2020-12-01:
thanks,
can't test charging at the moment, but
looking the trace log it seems that currents received from MBMD always a positive number.
Here +3.653 on L3 instead of -3.653. (PV is connected to L3)
Should I correct the sign somehow according to whether it is export to or import from the grid in?

[mqtt  ] TRACE 2020/12/01 09:49:38 recv mbmd/sdm1-1/Sum: '4799.809'
[mqtt  ] TRACE 2020/12/01 09:49:38 recv mbmd/sdm1-1/Power: '-788.390'
[mqtt  ] TRACE 2020/12/01 09:49:38 recv mbmd/sdm1-1/Current/L1: '0.144'
[mqtt  ] TRACE 2020/12/01 09:49:38 recv mbmd/sdm1-1/Current/L2: '0.000'
[mqtt  ] TRACE 2020/12/01 09:49:38 recv mbmd/sdm1-1/Current/L3: '3.653'

andig, 2020-12-01:
> looking the trace log it seems that currents received from MBMD always a positive number.

That's irrelevant.

Could you please show the relevant log part for the charge current/power issue?

vic911, 2020-12-01:
sure
Since I don't know where to get the whole log, I could only try to capture it from the log window. Hope nothing relevant has been missed.
Attached it in txt format. I've also attached my config file, evcc.yaml.

[evcc.yaml.txt](https://github.com/andig/evcc/files/5621931/evcc.yaml.txt)
[evcc_log.txt](https://github.com/andig/evcc/files/5621932/evcc_log.txt)

it seems that at 11:21:02 there is 2 setCurrent command

[evse  ] TRACE 2020/12/01 11:21:02 GET http://192.168.1.120/setCurrent?current=18
200 OK
S0_set current to given value
[cache ] TRACE 2020/12/01 11:21:02 chargeCurrent: 18
[evse  ] TRACE 2020/12/01 11:21:02 GET http://192.168.1.120/setCurrent?current=6
200 OK
S0_set current to given value

> looking the trace log it seems that currents received from MBMD always a positive number.
>>That's irrelevant.

I'm really curios, how can you calculate the available current per phase (and then get the minimum of them) if the sign of the actual current is not known. I could not see any calculations in the log.

I appreciate your kind support,
thanks

andig, 2020-12-01:
Du meinst die Stelle hier:

    [evse  ] TRACE 2020/12/01 11:21:02 GET http://192.168.1.120/getParameters
    200 OK
    {"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":20,"actualCurrent":0,"actualPower":0,"duration":197143,"alwaysActive":true,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0,"useMeter":false}]}
    [evse  ] TRACE 2020/12/01 11:21:02 GET http://192.168.1.120/setCurrent?current=18
    200 OK
    S0_set current to given value
    [cache ] TRACE 2020/12/01 11:21:02 chargeCurrent: 18
    [evse  ] TRACE 2020/12/01 11:21:02 GET http://192.168.1.120/setCurrent?current=6
    200 OK
    S0_set current to given value
    [lp-1  ] DEBUG 2020/12/01 11:21:02 charger enable
    [cache ] TRACE 2020/12/01 11:21:02 chargeCurrent: 18

Ich glaub das ist ein Bug in der evse Implementierung.

> I'm really curios, how can you calculate the available current per phase 

We don't. At present we're only looking at the power but not at the available currents ;)

vic911, 2020-12-01:
> Du meinst die Stelle hier:
> ```
> [evse  ] TRACE 2020/12/01 11:21:02 GET http://192.168.1.120/getParameters
> 200 OK [...]

is it a bug on evcc side or in SimpleEVSE-WiFi firmware at CurtRod?
 
> > I'm really curios, how can you calculate the available current per phase
> 
> We don't. At present we're only looking at the power but not at the available currents ;)

I see. It is fine for 1P environment, but not obvious in 3P. Are there any plan to implement this enhancement?

sorry, for answering in English, but my German is far worse.
regards

andig, 2020-12-01:
> sorry, for answering in English, but my German is far worse.

No problem!

> I see. It is fine for 1P environment, but not obvious in 3P. 

Why not? At least in Germany the sum of the phases is relevant, hence looking at power is correct.

> Are there any plan to implement this enhancement?

Which one?

vic911, 2020-12-01:
> > I see. It is fine for 1P environment, but not obvious in 3P.
> 
> Why not? At least in Germany the sum of the phases is relevant, hence looking at power is correct.
 
Ah that's it. In Hungary the limit is given per phase, so house main circuit breakers are sized according to this per phase limit.  

> > Are there any plan to implement this enhancement?
> 
> Which one?
to calculate and set charging current according to the per phase current data. (like in openWB)

premultiply, 2020-12-01:
Siehe #150

andig, 2020-12-01:
> to calculate and set charging current according to the per phase current data. 

I see. It's in the backlog but not top priority. I'm wondering though how I'd do that if the per phase currents don't carry the required sign. They'd have to?

> is it a bug on evcc side or in SimpleEVSE-WiFi firmware at CurtRod?

I need to double-check. I suspect evcc.

vic911, 2020-12-01:
> > to calculate and set charging current according to the per phase current data.
> 
> I see. It's in the backlog but not top priority. I'm wondering though how I'd do that if the per phase currents don't carry the required sign. They'd have to?

I'm not an engineer but the sign of the Power Factor could do the job.
[image]

andig, 2020-12-01:
> I'm not an engineer but the sign of the Power Factor could do the job.

I must admit that I can't remember. But lets discuss this as part of #150 als @premultiply mentioned. I've found the EVSE bug (which didn't manifest itself earlier) and will file a PR asap.

andig, 2020-12-01:
@vic911 are you able to test? If not I'll merge without your test.

vic911, 2020-12-01:
> @vic911 are you able to test? If not I'll merge without your test.

I would test is with pleasure, if test version will be available as HA add-on.

andig, 2020-12-01:
Can't do that for PRs. Once CI has finished building you should be able to update to `:latest`. Please let me know if fixed, if not another log file would help.

closed 2020-12-01 by andig