evcc-io/evcc #496: Allow mqtt for more vehicle data

vic911, 2020-11-30:
**Is your feature request related to a problem? Please describe.**
Renault server is very rigorous for querying data too often. Quota exceeds and the server locks out the user for a while. If more apps query the server for data the quota exceeds very fast.

**Describe the solution you'd like**
mqtt source for SOC for car is allowed, but I could not find any parameters for others. (remaining time, charge power, range)
this would make it possible to query data only by one app (ie. home assistant) and publish it to mqtt server, where all other app can reach what it needs.

premultiply, 2020-11-30:
Just define your vehicle in evcc configuration and you can get the requested data from evcc's own API interfaces which are also available through mqtt. I am pretty sure the Renault-API interface inside EVCC is done very well (and other car APIs too).

vic911, 2020-11-30:
thanks. the problem is, that I prefer querying it via Home Assistant, since afaik eg. evcc can not control the HVAC while HA does. Also HA can query some other sensor values, what evcc doesn't.

andig, 2020-11-30:
You can do everything on the HA side if you want and use generic vehicle/ plugins to feed the data in to evcc, too.

vic911, 2020-11-30:
Only one questions remained. how?
in my understanding, "charge" option is to feed evcc with SOC value. 
what is the code for the other? or evcc also do not use remaining time, range, charging power gathered via own renault api?

vehicles:
- name: renault
  title: Zoe
  capacity: 40 # kWh
  type: default
  charge:
    type: mqtt
    topic: homeassistant/renault/soc

andig, 2020-11-30:
Please check the readme and the examples in the evcc-config repo.

andig, 2020-11-30:
Sorry, I realize this is not obvious. You can set:

    charge:
    - type
    - ...
    status: # optional
    - type
    - ...
    range: # optional
    - type
    - ...

Please use Markdown formatting for your posts!

vic911, 2020-11-30:
works like charm. thanks for your support

closed 2020-11-30 by vic911