evcc-io/evcc #498: Remove active vehicle from configuration when API shows inconsistent results

andig, 2020-12-01:
Active vehicle should be removed after period of repeated errors or inconsistent readings (vehicle status != charger status) even if only single vehicle configured.
Should help with https://github.com/andig/evcc/issues/488.

mark-sch, 2020-12-01:
findActiveVehicle in der Loadpoint.go funktioniert nicht richtig, sobald die vehicle api keine valide Antwort liefert (403). Sind ein oder mehrere Fahrzeuge konfiguriert und wird dann ein nicht konfiguriertes Fremdfahrzeug (ohne Api) angesteckt, so wird fälschlicherweise das erste konfigurierte vehicle als verbunden gemeldet (ist es aber nicht).

andig, 2020-12-04:
Refs https://github.com/mark-sch/evcc/commit/03450ae0eca98ff87ce4100b6dcd70d258524c6f

andig, 2020-12-07:
Depends on #505

andig, 2020-12-16:
Wir haben noch keine konkreten Vorschläge wie das umgesetzt werden sollte, insbesondere in Anbetracht der jetzt selteneren Vehicle Updates. Wenn's keine Ideen gibt würde ich hier erstmal als "works as implemented" zumachen.

closed 2021-02-11 by stale[bot]