evcc-io/evcc #500: check go-e-charger firmware version

Optic00, 2020-12-02:
Since the go-e-charger now supports a temporary power control that is not written to flash (to prevent a flash failure) and evcc writes to that new API command, any go-e-charger with an old firmware will now not be power controlled.

It would be helpful to get a notice either in debug or gui that the firmware of the go-e-charger is too old (<040.0 in this case)

the api status for this is "fwv"

while I personally didn't run into this issue I noticed someone else did and maybe someone who doesn't know (or read all the changelogs ;-) ) won't notice. 

probably a low priority thing, although in a few month(?) almost everybody should be on fw 040.0 or later.

andig, 2020-12-02:
Version is written in the release notes and README. I currently don‘t have capacity to implement this, sorry. PR welcome.

tmeidinger, 2020-12-07:
I thought about implementing a firmware check as well. But since go-e has changed the format of the fwv string a couple of times I'd rather not go down that slope...

andig, 2020-12-07:
Then let’s close this issue. It will become obsolete over time anyway.

closed 2020-12-07 by andig