evcc-io/evcc #505: Redesign vehicle SoC updates

andig, 2020-12-06:
Die Praxis har gezeigt, dass einige Fahrzeuge/ Hersteller APIs nicht dafür geeignet sind, sich häufig zu aktualisieren. Mindestens Hyundai Kona und anscheinend auch Tesla zeigen dabei hohen Verbraucht, schlimmstenfalls entleerte Batterie.

Folgende Änderungen sind in Diskussion:
- Defaultwert für Cache von 5m auf 15m. Häufiger kann in keinem Fall abgefragt werden.
- Warnung falls kürzere Zeit eingestellt
- Einschränkung der Updates auf den Ladevorgang über einen `poll` Modus, per default nur wenn `charging`
- Falls `poll: connected` Abfrageintervall nur 1x je Stunde. Warnung falls häufiger
- Falls `poll: always` Abfrageintervall 1x je Stunde. Ebenfalls Warnung

premultiply, 2020-12-06:
Ich habe mal meinen erweiterten Vorschlag hierzu versucht nochmals besser auszuformulieren: https://github.com/andig/evcc/issues/488#issuecomment-739517148

andig, 2020-12-06:
> Ich habe mal meinen erweiterten Vorschlag hierzu versucht nochmals besser auszuformulieren

Hinzugefügt:

> - Reset `poll` timeout on (dis)connect and charger enable/disable events

Lokal gibts das schon, suche den letzten Testfehler...

andig, 2020-12-07:
Ready for test. /cc @caibuk

PR #508 (2020-12-08): Make vehicle soc polling more secure
Fix #505
closed 2020-12-08 by andig