evcc-io/evcc #506: Add Easee charger support

ste-ta, 2020-12-06:
Hi,

I am a user of an Easee Wallbox that can be controlled via APP and API. I already have an HomeAssistant integration but would love to use it together with EVCC.
The wallbox supports variable currents from 6-32 A as well as remote adoption of phases to be used.
There is a python wrapper library thats wraps the easee api so integration should be straightforward.

**Info Easee **
https://easee-international.com/de/zuhause-laden/

**Easee API:**
https://api.easee.cloud/index.html

**Python wrapper:**
https://pypi.org/project/easee/

**Documentation:** 
https://fondberg.github.io/pyeasee/pyeasee/

andig, 2020-12-06:
Sorry for not having a better replay to a good issue: at this time support for additional chargers is not a priority.

Letting alone the fact that I personally would not want a cloud-based API for my home charger, I cannot currently spend the time for implementing and supporting an additional charger. If the vendor wants to fund this on a regular basis he is welcome.

I'd also accept PRs if you would volunteer to take over providing the required support.

premultiply, 2021-02-01:
Die Cloud-API fällt wohl erwartungsgemäß regelmäig aus. Wer hätte das nur geahnt...?
https://easee.upvoty.com/b/ideas/local-simple-api

Aber es sieht zumindest auf den ersten flüchtigen Blick so aus als könnte man auch lokal etwas steuern:
https://github.com/andersalavik/easee-ha/

closed 2021-03-10 by andig

jojoman3, 2021-03-30:
Hi,

I would have a quick questions here. I see that the feature request for EASEE support is closed. On the other hand I see that there is a feature/easee brach available in the project which already contains quite some code and was even changed until a month ago. How much is there already implemented and how much would be left here? I am not too deep into developing, but does it make sense pull this and make it work for my own usage?