evcc-io/evcc #519: Ladung startet nicht automatisch

schenlap, 2020-12-11:
**Describe the bug**
charging with target soc does not start

**To Reproduce**
Steps to reproduce the behavior:
1.  mode = now
2. Target soc = 95% (set with last disconnect)
3. connect car
4. charging does not start
5. clock target soc = 100%
6. -> charging stars

**Expected behavior**
after 3. charging start

**EVCC details:**

```
{"type":"parameters","list":[{"vehicleState":1,"evseState":true,"maxCurrent":16,"actualCurrent":0,"actualPower":0,"duration":0,"alwaysActive":true,"lastActionUser":"","lastActionUID":"","en
ergy":0,"mileage":0,"meterReading":14.4,"currentP1":0,"currentP2":0,"currentP3":0,"useMeter":true}]}                                                                                         
[lp-1  ] TRACE 2020/12/11 17:32:26 charge currents: [0 0 0]A                                                                                                                                 
[cache ] TRACE 2020/12/11 17:32:26 chargeCurrents: [0 0 0]
[cache ] TRACE 2020/12/11 17:32:26 remoteDisabled:
[site  ] DEBUG 2020/12/11 17:32:33 ----
[site  ] DEBUG 2020/12/11 17:32:33 pv power: 0W
[site  ] DEBUG 2020/12/11 17:32:33 grid power: 408W
[site  ] DEBUG 2020/12/11 17:32:33 site power: 408W
[cache ] TRACE 2020/12/11 17:32:33 pvPower: 0
[cache ] TRACE 2020/12/11 17:32:33 gridPower: 408
[cache ] TRACE 2020/12/11 17:32:33 mode: now
[evse  ] TRACE 2020/12/11 17:32:33 GET http://10.0.0.208/getParameters
200 OK
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":0,"actualPower":0,"duration":0,"alwaysActive":true,"lastActionUser":"","lastActionUID":"","e
nergy":0,"mileage":0,"meterReading":14.4,"currentP1":0,"currentP2":0,"currentP3":0,"useMeter":true}]}
[lp-1  ] DEBUG 2020/12/11 17:32:33 charge power: 0W
[cache ] TRACE 2020/12/11 17:32:33 chargePower: 0
[cache ] TRACE 2020/12/11 17:32:33 chargeCurrent: 16
[evse  ] TRACE 2020/12/11 17:32:33 GET http://10.0.0.208/getParameters
200 OK
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":0,"actualPower":0,"duration":0,"alwaysActive":true,"lastActionUser":"","lastActionUID":"","e
nergy":0,"mileage":0,"meterReading":14.4,"currentP1":0,"currentP2":0,"currentP3":0,"useMeter":true}]}
[evse  ] TRACE 2020/12/11 17:32:33 GET http://10.0.0.208/getParameters
200 OK
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":0,"actualPower":0,"duration":0,"alwaysActive":true,"lastActionUser":"","lastActionUID":"","e
nergy":0,"mileage":0,"meterReading":14.4,"currentP1":0,"currentP2":0,"currentP3":0,"useMeter":true}]}
[cache ] TRACE 2020/12/11 17:32:33 chargedEnergy: -389.9999999999988
[cache ] TRACE 2020/12/11 17:32:33 chargeDuration: 0s
[evse  ] TRACE 2020/12/11 17:32:33 GET http://10.0.0.208/getParameters
200 OK
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":0,"actualPower":0,"duration":0,"alwaysActive":true,"lastActionUser":"","lastActionUID":"","e
nergy":0,"mileage":0,"meterReading":14.4,"currentP1":0,"currentP2":0,"currentP3":0,"useMeter":true}]}
[lp-1  ] DEBUG 2020/12/11 17:32:33 charger status: B
[evse  ] TRACE 2020/12/11 17:32:33 GET http://10.0.0.208/getParameters
200 OK
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":0,"actualPower":0,"duration":0,"alwaysActive":true,"lastActionUser":"","lastActionUID":"","e
nergy":0,"mileage":0,"meterReading":14.4,"currentP1":0,"currentP2":0,"currentP3":0,"useMeter":true}]}
[lp-1  ] DEBUG 2020/12/11 17:32:33 charge start energy: 14kWh
[lp-1  ] INFO 2020/12/11 17:32:33 car connected
[cache ] TRACE 2020/12/11 17:32:33 chargedEnergy: 0
[cache ] TRACE 2020/12/11 17:32:33 connectedDuration: 0
[cache ] TRACE 2020/12/11 17:32:33 chargeCurrent: 16
cache ] TRACE 2020/12/11 17:32:33 connected: true                                                                                                                                           
[cache ] TRACE 2020/12/11 17:32:33 charging: false                                                                                                                                           
[evse  ] TRACE 2020/12/11 17:32:33 GET http://10.0.0.208/getParameters                                                                                                                       
200 OK                                                                                                                                                                                       
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":0,"actualPower":0,"duration":0,"alwaysActive":true,"lastActionUser":"","lastActionUID":"","e
nergy":0,"mileage":0,"meterReading":14.4,"currentP1":0,"currentP2":0,"currentP3":0,"useMeter":true}]}                                                                                        
[evse  ] TRACE 2020/12/11 17:32:33 GET http://10.0.0.208/getParameters                                                                                                                       
200 OK                                                                                                                                                                                       
{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":0,"actualPower":0,"duration":0,"alwaysActive":true,"lastActionUser":"","lastActionUID":"","e
[... 231 lines cut ...]
[cache ] TRACE 2020/12/11 17:35:13 chargeCurrent: 16
[cache ] TRACE 2020/12/11 17:35:13 connected: true
[cache ] TRACE 2020/12/11 17:35:13 charging: true
[audi  ] TRACE 2020/12/11 17:35:14 GET https://msg.volkswagen.de/fs-car/bs/batterycharge/v1/Audi/DE/vehicles/WAUZ...../charger
200 OK
{"charger":{"settings":{"maxChargeCurrent":{"content":32,"timestamp":"2019-04-13T16:16:27Z"}},"status":{"chargingStatusData":{"chargingMode":{"content":"invalid","timestamp":"2020-12-11T12:
14:16Z"},"chargingReason":{"content":"invalid","timestamp":"2020-12-11T12:14:16Z"},"externalPowerSupplyState":{"content":"unavailable","timestamp":"2020-12-11T12:14:16Z"},"energyFlow":{"con
tent":"off","timestamp":"2020-12-11T12:14:16Z"},"chargingState":{"content":"off","timestamp":"2020-12-11T12:14:16Z"}},"cruisingRangeStatusData":{"engineTypeFirstEngine":{"content":"typeIsEl
ectric","timestamp":"2020-12-11T16:31:09Z"},"primaryEngineRange":{"content":2,"timestamp":"2020-12-11T16:31:09Z"}},"ledStatusData":{"ledColor":{"content":"none","timestamp":"2020-12-11T12:1
4:16Z"},"ledState":{"content":"off","timestamp":"2020-12-11T12:14:16Z"}},"batteryStatusData":{"stateOfCharge":{"content":8,"timestamp":"2020-12-11T16:31:09Z"},"remainingChargingTime":{"cont
ent":5,"timestamp":"2020-12-11T16:31:09Z"},"remainingChargingTimeTargetSOC":{"content":"unsupported","timestamp":"2020-12-11T16:31:09Z"}},"plugStatusData":{"plugState":{"content":"disconnec
ted","timestamp":"2020-12-11T12:14:16Z"},"lockState":{"content":"unlocked","timestamp":"2020-12-11T12:14:16Z"}}}}}
[lp-1  ] DEBUG 2020/12/11 17:35:14 vehicle soc: 8%
[lp-1  ] DEBUG 2020/12/11 17:35:14 vehicle range: 2km
[cache ] TRACE 2020/12/11 17:35:14 socCharge: 8
[cache ] TRACE 2020/12/11 17:35:14 chargeEstimate: -1ns
[cache ] TRACE 2020/12/11 17:35:14 chargeRemainingEnergy: 7155.555555555556
[cache ] TRACE 2020/12/11 17:35:14 range: 2
[evse  ] TRACE 2020/12/11 17:35:14 GET http://10.0.0.208/getParameters
```

Da ich gestern in der Früh einen leeren Akku hatte schaue ich da jetzt ganz genau. In diesem Fall ist die evse nicht schuld da es ja startet wenn ich auf 100% Target soc clicke. Davor stand in der GUI nur Verbinden (oder connected) und der ganz balken ist gelb (oder orange).

Master:
6724f04e0421e753ec97d1e05a4e64ecdb123615

andig, 2020-12-11:
Zeigtst Du bitte noch Deine Config? Wenn sensitiv gerne über Slack.

schenlap, 2020-12-11:
Ich habe dafür schon ein script dass sensible Informationen löscht :-)

```
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: trace
#log: error
levels:
   core: trace
   influx: error
   lp-1: trace
#   #server: trace
   modbus: error
#   #modbus: trace
   ws: error
   #   wifi: trace
   audi: trace

# influx database (v1)
influx:
   url: http://10.0.0.180:8086
   database: evcc
#  user:
#  password:

# meters
meters:
- name: vzpv # pv Ertrag
  type: default
  power:
    type: http
    uri: http://volkszaehler/api/data/aaaaaaaa-bbbb-cccc-dddd-eeeeeeeeeeee.json?from=now
    jq: .data.tuples[0][1] # parse response json
- name: pvse
  type: default
  power:
    type: calc
    add:
     - type: modbus
       uri: 10.0.0.207:502
       model: sunspec
       value: Power
       id: 1
       scale: 1
     - type: modbus
       uri: 10.0.0.207:502
       model: sunspec
       value: Power
       id: 2
       scale: 1
- name: pvse1 # pv Ertrag from solaredge via modbus
  type: default
  power:
    type: modbus
    uri: 10.0.0.207:502
    id: 1
    model: sunspec
    value: Power
- name: pvse2 # pv Ertrag from solaredge via modbus
  type: default
  power:
    type: modbus
    uri: 10.0.0.207:502
    id: 2
    model: sunspec
    value: Power
- name: vzwb # Wallbox
  type: default
  power:
    type: http
    uri: http://volkszaehler/api/data/aaaaaaaa-bbbb-cccc-dddd-eeeeeeeeeeee.json?from=now
    jq: .data.tuples[0][1] # parse response json
- name: vzws # combined with websockets
  type: default
  power:
    type: calc
    add:
      - type: ws
        uri: ws://10.0.0.180:8082/socket #Bezug
        jq: .data | select(.uuid=="aaaaaaaa-bbbb-cccc-dddd-eeeeeeeeeeee") .tuples[0][1] # import
        timeout: 30s
        scale: 1
      - type: ws
        uri: ws://10.0.0.180:8082/socket #Einspeisung
        jq: .data | select(.uuid=="aaaaaaaa-bbbb-cccc-dddd-eeeeeeeeeeee") .tuples[0][1] # export
        timeout: 30s
        scale: -1

chargers:
- name: evsewifi
  type: evsewifi # Charger with Phoenix Contact controller
  uri: http://10.0.0.208 # SimpleEVSE-Wifi address
  meter:
     energy: true
     currents: true
     power: true

vehicles:
- name: audi
  type: audi
  title: A3 e-tron
  capacity: 7.5 # kWh
  user: geheim@geheim.com
  password: pw
  vin: WAUZ
  cache: 30m
- name: configurable
  type: default
  title: Default
  capacity: 8 # kWh
  charge:
    type: script # use script
    cmd: cat /tmp/soc #/bin/sh -c "echo 50.1" # actual command
    timeout: 3s # kill script after 3 seconds
  cache: 1m

site:
  title: Home
  meters:
     grid: vzws
     pv: pvse
     #pv:
     #- pvse1
     #- pvse2

loadpoints:
- title: main # name for logging
  #vehicle: configurable
  vehicle: audi
  mode: pv
  soc:
    min: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
    estimate: true
    target: 100
    levels: # target soc
    - 30
    - 50
    - 95
    - 100
    poll:
      mode: charging
      interval: 180m # intervall when not charging
  onDisconnect:
    mode: now
    targetSoC: 95
  charger: evsewifi # charger
  #meters:
  #summer
  #enable: # pv mode enable behavior
  #  delay: 2m # threshold must be exceeded for this long, 2m
  #  threshold: -1200 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  #disable: # pv mode disable behavior
  #  delay: 2m # threshold must be exceeded for this long, 2m
  #  threshold: 500 # maximum import power (W # MIN = 1380-enable(600) = 780)
  #winter
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long, 2m
    threshold: -400 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 2m # threshold must be exceeded for this long, 2m
    threshold: 1100 # maximum import power (W # MIN = 1380-enable(600) = 780)
  guardduration: 20s # switch charger contactor not more often than this (default 10m)
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 1 # ev phases
  #maxgridpower: 4000 # max import power in now mode
```

schenlap, 2020-12-11:
Interessant ist auch die Visualisierung von gestern.

[image]

Das Fahrzeug ist am 9.12 um ca. 17:00 angesteckt worden , hat aber nicht geladen. Erst als ich es am nächsten tag um 6:30 gemerkt habe und händich den TargetSoc von 95% auf 100% geändert habe, hat die Ladung gestartet. Ich musste dann aber kurz daruf weg, daher nur die kurze Ladung. Meine Vermutung ist dass aus irgendeinen Grund der Soc jetzt nicht mehr abgeholt wird und daher der Vergleich mit dem TargetSoc nicht stattfindet.

closed 2020-12-12 by andig