evcc-io/evcc #526: ABL eMH1 Support (EVCC2)

danielkestler, 2020-12-16:
Hallo @andig, sehr cooles Projekt hier, das vorab! 

Ich würde hier gern nochmal das Thema ABL eMH1 aufmachen, diese Wallbox wird aktuell leider noch nicht unterstützt. Prinzipiell hat diese eine RS485-Schnittstelle und wohl (?) einen EVCC2-Ladecontroller. Bei der Schnittstellenspezifikation bin ich mir unschlüssig, es gibt einige im Netz zu finden, die Frage ist aber, ob diese für die aktuellen Modelle (2019, 2020) gültig ist, da sich hier wohl der Ladecontroller geändert hat.

Bestimmt hätten mehrere Leute Interesse daran, da es sich um eine sehr beliebte und günstige Wallbox handelt. Gerne unterstütze ich auch, wo ich kann, wobei ich mehr Ahnung von Programmierung als von alten seriellen Schnittstellen habe ;-) 

Viele Grüße
Daniel

andig, 2020-12-16:
Siehe https://github.com/andig/evcc/issues/506:

Zum jetzigen Zeitpunkt ist die Unterstützung weiterer Wallboxen keine Priorität. Falls Hersteller EVCC unterstützen und  Implementierung und v.a. Support finanziell fördern wollen sind sie herzlichen willkommen. 

Ebenso akzeptiere ich gerne PRs um weitere Geräte zu unterstützen wenn der Support des neuen Codes ebenfalls gewährleistet ist.

Bzgl ABL selbst: die Box spricht anscheinend RS485, aber kein Modbus...

danielkestler, 2020-12-16:
Verstehe, sehr schade, aber verstehe ich natürlich. Ich sehe mal, wie ich da weiter komme. Genau, Modbus ist es wohl nicht ...

andig, 2021-01-14:
@danielkestler gabs denn von ABL eine Antwort?

danielkestler, 2021-01-15:
@andig Ja, es gibt "noch (?) keine offizielle Kommunikationsmöglichkeit". Klang aber ehrlich gesagt auch nicht so, als käme da noch etwas.

andig, 2021-01-15:
Dann machen wir hier zu :(

naltatis, 2021-07-04:
Hat hier jemand schon neue Erkenntnisse? Ich habe gerade im Datenblatt folgendes gesehen:

**Kommunikation EVSE: RS485**

https://www.ablmobility.de/global/downloads/datenblaetter/emh1/eMH1_DE/ABL_1W1101_DE.pdf?m=1623325779&

andig, 2021-07-04:
RS485 ist die Physik, fehlt das Protokoll. Von Modbus steht da nix.

premultiply, 2021-07-04:
Guck auch mal hier: https://www.goingelectric.de/forum/viewtopic.php?f=34&t=38749

Ist nicht üblich aber auch kein Hexenwerk.

andig, 2021-07-04:
Es bleibt halt wie immer die Frage: wozu? Bringt uns als Projekt nicht voran...

premultiply, 2021-07-04:
Das sehe ich anders. Jede unterstützte Hardware bringt das Projekt voran.
Insbesondere wenn sie weit verbreitet ist, realtiv günstig oder Protokolle nutzt die nicht so gängig sind. Siehe Wallbe Eco 2.0s.
Schwierig wird es immer dann wenn Nutzer erkennen dass ausgerechnet ihre Wallbox dann doch nicht läuft.
Auf die Hersteller kannst du lange warten. Da wird nichts passieren.

naltatis, 2021-07-04:
Würde ich genau so sehen. Was die Liste der unterstützten Hardware angeht würde ich sagen „mehr ist mehr“.

Diese Box konkret scheint Testsieger beim ADAC Test gewesen zu sein. Das war für meinen Onkel ausschlaggebend die als WBs zur neu geplanten PV Anlage dazuzubestellen. Mein Elektriker hatte nur die damals auch angeboten. Die scheint auf jeden Fall verbreitet zu sein.

premultiply, 2021-08-05:
Hier noch ein paar wichtige Hinweise:
https://www.goingelectric.de/forum/viewtopic.php?p=1431718#p1431718
https://www.goingelectric.de/forum/viewtopic.php?p=1543242#p1543242
https://www.goingelectric.de/forum/viewtopic.php?p=1550459#p1550459

WhyWorrY-PV, 2021-08-05:
Ich unterstütze gerne das Projekt und denke das würde für viele ABL eMH1 User eine tolle Lösung sein. Gerne kann ich versuchen mehr Unterstützer zu organisieren, wenn das hilft?

premultiply, 2021-08-05:
Gerne. Es würden dann auch Leute benötigt die ihre Box mal zu Testzwecken freigeben können.

andig, 2021-08-05:
Reopening due to recent activity.

Ich bräuchte für eine Umsetzung:
- Testbox
- Tester+Support damit sich der Aufwand lohnt
- Bestätigung, dass hier Modbus ASCII gesprochen wird
- Dokumentation der relevanten Befehle für Enable(on/off)/Enabled/Status(A..C)/Stromvorgabe(A oder mA?)
- Eine grobe Idee wie die Schnittstelle zur Box aussieht (nur für Bastler oder Stecker außen dran?)

ThKls, 2021-08-06:
> Reopening due to recent activity.
>
> Ich bräuchte für eine Umsetzung:
>
> * Testbox
> * Tester+Support damit sich der Aufwand lohnt
> * Bestätigung, dass hier Modbus ASCII gesprochen wird
> * Dokumentation der [...]

Punkt 3:
Ja, es wird Modbus ASCII gesprochen, 38400, 8,E1

Punkt 4:
Hier ist eine Beschreibung der zu verwendenden Befehle:
https://www.goingelectric.de/forum/viewtopic.php?p=1550459#p1550459

Punkt 5:
Die Schnittstelle ist auf Klemmen in der Box. Ein USB/ RS485 Adapter funktioniert bei mir. Einige Hinweise zum Anschluss sind hier:
https://www.goingelectric.de/forum/viewtopic.php?p=1542681#p1542681

Komme dieses Wochenende nicht zu mehr …

andig, 2021-08-06:
@ThKls kannst Du bei den Befehlen payload und Protokoll differenzieren? Mich interessiert nur der Payload, mit Modbus Ascii hab ich mich noch nicht auseinander gesetzt.

andig, 2021-08-06:
Nvm. Natürlich gibts das- ändere ich asap…

andig, 2021-08-07:
@ThKls ich schaue mir das gerade an:
- sperren/entsperren heisst Stromfreigabe ein/ausschalten oder Stecker locken?
- Stromvorgabe braucht 4 Byte, payload am Ende?
- es gibt 2 Register für Status- wo ist der Unterschied?

klaus1956, 2021-08-08:
Hallo
Ich gehöre zu den Leuten, die sich da was mit Home Assistant und Node Red "zusammenstümpern" - nutze aber die selbe Beschreibung und kann daher vielleicht etwas beantworten:
- Innerhalb meine Box (mit SENEC gebrandet, aber ohne Technikpaket gekauft) sind Anschlüsse für Kabel in der Box (Foto suche ich raus)
[image]
- Wegen der geforderten Steuerbarkeit sollten die Kabel in der Box eigentlich angeschlossen sein. Bei mir wurde ein CAT7-Kabel mit RJ45 Stecker ins Haus geführt - man muß dann nur noch die richtigen 2 Kabel identifizieren. Den 120 Ohm Abschlußwiderstand habe ich am RS485-Umsetzer angeschlossen. Damit funktioniert es. Mein Umsetzer ist ein **Protoss PW21** mit Wifi den ich im MQTT Modus betreibe. TCP wäre aber auch möglich.
- Der Steuerbefehl für die Leistung sieht dann bei mir im MQTT so aus:  ":01100014000102006474\r\n" wobei die "0064" am Ende die Leistung auf 6A setzen und die "74" das Kontrollbyte ist. 
- Sperren/Entsperren sperrt die Box. Das heißt das beide Kontrollleuchten (Blau und Grün) konstannt leuchten, die Box also gesperrt ist und nicht lädt.

premultiply, 2021-08-08:
Und welche Anschlüsse davon sind es nun genau wo der RS485-Bus anliegt? Die RJ45-Buchse oder die Klemmen daneben?
Hast du den PW21 direkt in der Box unterbringen können?

klaus1956, 2021-08-08:
Der RS485-Bus liegt an den kleine Drähten oben links wobei Orange und Orange/Weiß die Busleitungen sind.
Meine EInbausituation in der Box sieht so aus:
[image]
Da würde ein PW21 noch passen - ich habe aber die Box ohne eingebauten FI (der hängt bei mir im Keller). Wie die Platzsitation in der Box bei den Versionen mit FI sind weiß ich nicht.

premultiply, 2021-08-08:
Wichtiger Hinweis: Derzeit wird die Box wohl auch ohne interne Strommessung ausgeliefert (hier oben links im Bild zu sehen mit Wandlern) https://community.symcon.de/t/modul-abl-wallboxen/121906/69
Die zugehörigen Register liefern dann offensichtlich nur Nullwerte.

klaus1956, 2021-08-08:
Interessanter Hinweis.
Aktuell arbeite ich im Wesentlichen so, dass ich die Befehle zur Steuerung nur "über den Zaun schmeiße" und die Auswirkungen an den Shelly-Messwerten bewerte. 
Das in der Diskussion bei Going-Elektric angesprochen Problem mit verschluckten Befehlen wegen schlafender WB hatte ich anfangs auch und hatte die Befehle deshalb immer mit 1s Abstand 2x geschickt. Inzwischen habe ich aber eine Statusabfrage für Kontrolleuchten in meinem Dashboard, die alle 10 sec ausgeführt wird. Seitdem gibt es das Problem mit verschlafenen Befehlen nicht mehr. Der Standby-Stromverbrauch der Box hat sich dadurch nicht merklich erhöht.

andig, 2021-08-08:
Ich stehe mit dem Modbus ASCII noch auf Kriegsfuss. Gäbe es jemanden, der mir (cpuidle@gmx.de) einen eMH auf einen Port legen könnte oder per SSH Zugang geben und mit testen? Bitte bei Slack melden.

andig, 2021-08-08:
@klaus1956 ich hab jetzt folgenden Stand:

	// :01 10 0005 0001 02 E0E0 19
	wb.Enable(false)

	// :01 10 0005 0001 02 A1A1 97
	wb.Enable(true)

	// :01 10 0014 0001 02 0064 66
	wb.MaxCurrent(6)

	// :01 10 0014 0001 02 010B BE
	wb.MaxCurrent(16)

Leider bekomme ich bei ASCII immer noch einen Knoten im Kopf. Irgendwie erscheint mir die `02` oben überflüssig zu sein im Protokoll weil über die Anzahl der Register schon definiert, aber was solls.

Was jetzt- falls das so stimmt- noch fehlen würde ist die Abfrage des Status (frei/gesperrt)- kann ich das Register 5 auch auslesen? Oder brauche ich dafür das Status Kommando?

Wie gehen wir mit dem Zähler um: optional oder immer drin? Lohnt auslesen?

andig, 2021-08-08:
@klaus1956 und wofür brauche ich

    (Laden unterbrechen)
    :0110001400010203E8

wenn ich enable/disable über Register 5 mache?

klaus1956, 2021-08-08:
Ich fahre momentan zweigleisig:
- :0110001400010203E8 nutze ich um das Laden temporär auf Null zu fahren, wenn ich später wieder Hochregeln will
- Enable/Disable nutze ich z.B. zum Umschalten der Phasen oder um die Box im Urlaub zu sperren
Kann man bestimmt auch anders machen

premultiply, 2021-08-08:
Zähler optional fände ich angemessen sonst kommt da ja nichts wenn nicht verbaut.
In diesem Zusammenhang auch interessant: https://community.symcon.de/t/modul-abl-wallboxen/121906/93

> Laut Doku bedeutet Hex 03E8: „Phase current meter not available or state A“
> 
>     03030A2EC2910B0 3E8 03E8 03E8 A3

Bei den Statuswerten scheint der Ausführlichere (0x002E) zudem mehr Daten auf einen Schlag (Zähler?) zu liefern während der einfache Ladestatus (0x0004) offenbar wirklich nur explizit diesen liefert.

andig, 2021-08-08:
...und wie *lese* ich ob die Box freigeschaltet ist? Im Prinzip könnte ich die 14 verwenden und auf die 5 auch völlig verzichten.

premultiply, 2021-08-08:
14? Du meinst 46 (0x2E), oder?

andig, 2021-08-08:
> 14? Du meinst 46 (0x2E), oder?

Entweder 14 für Strom oder 5 für enable/disable oder dann doch 2E. Ich weiss es nicht. Es muss halt nur konsistent sein...

klaus1956, 2021-08-08:
Zum Status sende ich alle 10 sec ":0103002E0005C9\r\n"
Die Antwort beginnt dann mit >01030A2E gefolgt vom Status A1, B1, B2 ....

andig, 2021-08-08:
Es geht nicht um den Status, sondern um die Ladefreigabe. Die möchte evcc auch lesen. Kannst du einfach mal schauen, ob du Register 5 auch lesen kannst? Lässt sich das auswerten?

premultiply, 2021-08-08:
Wenn ich es richtig deute kommt der Enabled=false-Status eventuell über einen "E0"-Ladestatus.

klaus1956, 2021-08-08:
genau

andig, 2021-08-08:
Baue ich ein 👍🏻

premultiply, 2021-08-08:
An diesen beiden Symcon-Modulen kann man sich ggf. auch ein wenig orientieren:
https://github.com/cvvmedia/symcon.cvvmedia.ablchargepoint/blob/main/ABL%20ChargePoint%20Splitter/module.php
https://github.com/cvvmedia/symcon.cvvmedia.ablchargepoint/blob/main/ABL%20eMH/module.php

andig, 2021-08-08:
> Gäbe es jemanden, der mir ([cpuidle@gmx.de](mailto:cpuidle@gmx.de)) einen eMH auf einen Port legen könnte oder per SSH Zugang geben und mit testen? Bitte bei Slack melden.

oder per email

premultiply, 2021-08-08:
@klaus1956 Könntest du @andig damit aushelfen? Einfach eine temporäre Portweiterleitung des Routers auf den TCP-Port des PW21 wäre sehr hilfreich.

klaus1956, 2021-08-08:
Da ich mich damit nicht auskenne hatte ich @andig schon deswegen angeschrieben - müsste dafür aber erst mal TCP einrichten, da ich mit MQTT arbeite

klaus1956, 2021-08-08:
> 
> Leider bekomme ich bei ASCII immer noch einen Knoten im Kopf. Irgendwie erscheint mir die `02` oben überflüssig zu sein im Protokoll weil über die Anzahl der Register schon definiert, aber was solls.
> 

Habe die Codes leider auch abgekupfert  (und die 02 immer mit drin)

Hier mal meine Befehlsliste -

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 14">
<link id=Main-File rel=Main-File
href="file:///C:\Users\Admin\AppData\Local\Temp\msohtmlclip1\01\clip.htm">
<link rel=File-List
href="file:///C:\Users\Admin\AppData\Local\Temp\msohtmlclip1\01\clip_filelist.xml">

<style>

</style>
</head>

<body link=blue vlink=purple>

Ladeleistung   setzen 0A | :011000140001020010C8\r\n
Ladeleistung   setzen 1A | :011000140001020015C3\r\n
Ladeleistung   setzen 2A | :011000140001020020B8\r\n
Ladeleistung   setzen 3A | :011000140001020031A7\r\n
Ladeleistung   setzen 4A | :01100014000102004296\r\n
Ladeleistung   setzen 5A | :01100014000102005385\r\n
Ladeleistung   setzen 6A | :01100014000102006474\r\n
Ladeleistung   setzen 7A | :01100014000102007563\r\n
Ladeleistung   setzen 8A | :01100014000102008652\r\n
Ladeleistung   setzen 9A | :01100014000102009642\r\n
Ladeleistung   setzen 10A | :0110001400010200A731\r\n
Ladeleistung   setzen 11A | :0110001400010200B820\r\n
Ladeleistung   setzen 12A | :0110001400010200C810\r\n
Ladeleistung   setzen 13A | :0110001400010200D9FF\r\n
Ladeleistung   setzen 14A | :0110001400010200EAEE\r\n
Ladeleistung   setzen 15A | :0110001400010200FADE\r\n
Ladeleistung   setzen 16A | :0110001400010200FFD9\r\n
Ladepunkt   Reset | :011000050001025A5A33\r\n
Abfrage   Firmware | :010300010002F9\r\n
Status Abfrage | :0103002E0005C9\r\n
WB Sperren | :01100005000102E0E027\r\n
WB Entsperren | :01100005000102A1A1A5\r\n
</body>

</html>

andig, 2021-08-08:
> Da ich mich damit nicht auskenne hatte ich @andig schon deswegen angeschrieben - müsste dafür aber erst mal TCP einrichten, da ich mit MQTT arbeite

Hi Klaus, ist leider nix angekommen. TCP wäre super, auf MQTT funktioniert das hier mit der Modbus lib eh nicht.

klaus1956, 2021-08-08:
Habe die Mail jetzt noch mal an cpuidle@gmx.de gesendet

andig, 2021-08-08:
PR aktualisiert. Konnte noch nicht mit Klaus testen, sollte aber gehen (tm). Schmankerl: die Box kann auch granular den Duty Cycle regeln, nicht nur ganze Ampere- hab ich umgesetzt. Enable/Disable hab ich über Strom gelöst, die Box muss also einmalig per Reg. 5 enable freigeschaltet werden, das macht evcc nicht. Was jetzt noch fehlt wären die Zähler. Ströme scheint es zu geben, bei der Leistung hab ichs nicht ganz verstanden...

premultiply, 2021-08-08:
So wie ich es verstehe sind die in 0x002e gelieferten Ströme in der Einheit 1/10 Ampere. Eine Spannungsmessung scheint es (dokumentiert) nicht zu geben. Daher würde ich die dann jeweils einfach erstmal stumpf mit 230 V multiplizieren.

WhyWorrY-PV, 2021-08-08:
Wow, hier geht was. Gute Arbeit zusammen. Wenn ihr mir Anweisungen gibt, wie ich genau testen soll, bitte um Info. Ist aber eher User Acceptance Testing :-/
Schönen Sonntagabend dann noch.

premultiply, 2021-08-08:
Die Box über einen RS485-TCP-Schnitstellenkonverter auf einem externen Port temporär freizugeben würde schon mal sehr helfen.

ThKls, 2021-08-08:
@andig 
Bin wieder im Lande. Habe die Fortschritte bewundert.
Andere haben Dir schon die fehlenden Infos gegeben. Wenn Du noch etwas brauchst, einfach fragen.
Zur Frage enable/ disable: das schaltet nur die Ladebereitschaft der Box ein und aus. Der Stecker am Auto bleibt verriegelt.
Wie muss ich die ABL Box im yaml konfigurieren? Habe ich nicht gefunden.

andig, 2021-08-08:
> Wie muss ich die ABL Box im yaml konfigurieren? Habe ich nicht gefunden.

    sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c3IiOiJkZW1vIiwic3BlIjp0cnVlLCJleHAiOjE2MzA0MTE0NjAsImlhdCI6MTYyNzk5MjI2MCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6ImRlbW8ifQ.NGMM7TsfOrUaFGRjbGnVbi7oTlKc6ApYunwmNsdN_OI

    chargers:
    - name: abl
      type: abl
      uri: <ip:port>
      id: 1

...jedenfalls wenn sie über TCP angeschlossen ist. 

Was wir noch nicht haben ist 
- [ ] Leistungsmessung (Ströme x3?). 
- [ ] Umgang mit Sleep Mode.
- [ ] ggf Zähler konfiguriertbar. Erstmal tun wir so als wäre der vorhanden

Ansonsten fehlt mir immer noch ein Testzugangs- mit Klaus hat es bisher nicht geklappt :/

premultiply, 2021-08-08:
Ja, ein Testzugang zu einer Box wäre echt hilfreich...

andig, 2021-08-09:
Zum testen bitte einfach mal 

    evcc charger

ausführen, falls etwas nicht stimmt zusätzlich `--log trace` und Ergebnis hier posten. Jetzt liegt der Ball bei Euch für die Tests.

WhyWorrY-PV, 2021-08-09:
Wenn jemand Zeit und Lust hat mir Schritt für Schritt zu erklären, welchen Konverter ich brauche (Link), wie ich den anschließe und dann evcc charger zum Laufen bringe, dann teste ich gerne. Verstehe aber wenn es zu viel Aufwand ist...

premultiply, 2021-08-09:
Willst du per USB, Ethernet oder WiFi anschließen?

premultiply, 2021-08-09:
Hier sowas für USB: https://www.reichelt.de/raspberry-pi-usb-rs485-schnittstelle-ch340c-rpi-usb-rs485-p242783.html oder https://www.amazon.de/RS485-Konverter-Converter-Adapter-Unterst%C2%A8%C2%B9tzung/dp/B06Y1JTGZX
und sowas für Ethernet oder WiFi: https://de.aliexpress.com/i/4001153040889.html

WhyWorrY-PV, 2021-08-10:
OK, um sicher zu gehen, das ich den Ablauf richtig verstanden habe:
1. Wifi Adapter in ABL Box mit RS485 verkabeln und irgendwo auch Stromanschluss organisieren (hoffe Klaus kann mir da helfen)
2. Verbindung checken im WLAN Netzwerk
3. evcc charger ausführen und Verbindung zu ABL eMH1 checken
4. evcc an sonnigem Tag ausführen, Auto anschließen und protokollieren was passiert

Ich befürchte der Teil mit "evcc ausführen" ist mir nicht ganz klar. Wo liegt die Datei? Muss ich evcc aus Github starten?
Danke für die klasse Unterstützung bisher.

klaus1956, 2021-08-10:
Hallo
Ich habe nicht wirklich eine breite Erfahrung, als ich aber eine Lösung gesucht habe bin ich auf den **Protoss PW21** gestoßen - charm von dem Ding ist Hutschienenbefestgung und 220V Spannungsversorgeung. Es gibt auch Niederspannung- und LAN-Versionen (PW11) - Vorsicht bei Versionsauswahl und beim Anschließen an eine Absicherung der Versorgungspannung denken.
Viele andere Versionen brauchen einen Netzteil UND eine Einsteckmöglichkeit in der Nähe des Adapters.
Ideal ist wahrscheinlich wenn man den PW21 mit in der WB unterbringen kann.

Schönen Gruß
Klaus

klaus1956, 2021-08-10:
Ich hänge noch mal meine Anschlußsituation im Schaltschrank an:
[image]

premultiply, 2021-08-10:
Es gibt von dem Protoss-Teil mehrere Versionen und jeweils mit und ohne integriertem Netzteil (empfohlen):
PE11 (nur Ethernet)
PW11 (nur WiFi)
PW21 (Ethernet+WiFi)

Natürlich sollte man wo immer möglich über Ethernet statt WiFi gehen. Mit dem PW21 ist man aber ohnehin flexibel.

Einen PW11-H hätte ich noch hier und würde den an Bedürftige abgeben.

Der Montageort des Umsetzers ist egal und sollte nach den besten örtlichen Gegebenheiten gewählt werden. RS485 ist sehr robust. Wenn man es über eine verdrillte Doppelader führt und korrekt an beiden Seiten terminiert sind selbst über 100m Kabelweg kein Problem. Eine Schirmung ist auch nicht unbedingt notwendig. Falls vorhanden wird der Schirm auch immer nur an einer Seite angeschlossen!
Wenn man ohnehin von der Wallbox irgendwie eine Doppelader liegen (z. B. Telefonkabel oder Cat-Kabel) hat kann man darüber auch z. B. bis zu einem Raspberry gehen auf dem dann EVCC läuft und dann per billigem RS485-USB-Adapter direkt rein. Oder man spart sich den Kabelweg und baut z. B. einen Raspberry ZeroW (samt Netzteil und USB-Adapter) direkt in die Wallbox ein. Und wenn man schon da ist kann man dann die Absicherung des Wallbox-Kontrollers auf L1 gleich mitnutzen.

Wichtig ist eigentlich erstmal nur überhaupt mal eine Verbindung von einem Rechnersystem zur Wallbox. Wenn ihr dann @andig einen Zugang einrichtet kann er dann selbst mal schauen ob alles schon vom Code her passt und ggf. direkt nachbessern. Anschließend wird der Code dann übernommen und kann im nächsten Release direkt genutzt werden.

WhyWorrY-PV, 2021-08-11:
Guten Morgen zusammen,
danke für die Infos, schätze das sehr und lerne enorm dabei. Die Option mit Raspberry ZeroW und USB Adapter finde ich nach drüber-schlafen am "sinnvollsten" für mich. Ich besorge mir die Teile und melde mich sobald es ans Einbauen geht. Werde bestimmt wieder Fragen haben. Wegen Urlaub, wird das aber vermutlich Mitte September :-/

[49 more replies]

closed 2021-10-31 by andig