evcc-io/evcc #528: OpenWB -Anbindung funktioniert nicht

junibart, 2020-12-17:
Hallo, 

ich teste gerade evcc v 0.37 in Verbindung mit meiner OpenWB. In einer früheren Version des Wikis lautete die Beispielkonfiguration für die OpenWB wie folgt: 

```
chargers:
- name: openwb
  type: openwb
  broker: 192.168.1**.70 # openWB IP
  id: 1 # loadpoint id
```
evcc startet, allerdings werden im 10-Sekundentakt Fehlermeldungen ausgegeben:
```
[lp-1  ] DEBUG 2020/12/17 15:13:55 charge power: 5432W
[lp-1  ] ERROR 2020/12/17 15:13:55 charger error: openWB/lp/1/boolChargeStat outdated: 24s
[lp-1  ] DEBUG 2020/12/17 15:14:05 charge power: 5391W
[lp-1  ] ERROR 2020/12/17 15:14:05 charger error: openWB/lp/1/boolChargeStat outdated: 34s

```
Jetzt sah ich im Wiki, dass die Konfiguration einer openWB anders erfolgt: 

```
chargers:
- name: openwb
  type: openwb
  status:
    # with openWB, charging status (A..F) this is split between "plugged" and "charging"
    # the openwb type combines both into status (charging=C, plugged=B, otherwise=A)
    type: openwb
    plugged:
      type: mqtt
      topic: openWB/lp/1/boolPlugStat
    charging:
      type: mqtt
      topic: openWB/lp/1/boolChargeStat
  enabled:
    type: mqtt
    topic: openWB/lp/1/ChargePointEnabled
    timeout: 30s
  enable:
    type: mqtt
    topic: openWB/set/lp1/ChargePointEnabled
    payload: ${enable:%d}
  maxcurrent:
    type: mqtt
    topic: openWB/set/lp1/DirectChargeAmps

```

Trage ich diese Konfig in die evcc.yaml ein, erhalte ich nach dem Start von evcc die Fehlermeldung 

```
[main  ] FATAL 2020/12/17 15:30:31 cannot create charger 'openwb': cannot create type 'openwb': 1 error(s) decoding:
* '' has invalid keys: enable, enabled, maxcurrent, status
```
Ich vermisse in dieser Beispielkonfiguration auch die Festlegung der IP des Brokers bzw. der openWB. 
Könnte bitte jemand eine gültige Konfiguration für die Ansteuerung einer OpenWB im Wiki hinterlegen?

Danke!

andig, 2020-12-17:
Also.

```
- name: openwb
  type: openwb
```

wäre der "native" OpenWB Typ und sollte funzen, tut es bei Dir aber nicht. Das wäre ein Bug. Remotezugang würde mir helfen (ssh oder read-only Cloud Broker? cpuidle@gmx.de)

Deine zweite Config ist der "manuelle" Weg. Da stimmen Einrückung nicht und den Typ muss eben `generic` sein, in den Du dann die einzelnen Blöcke rein konfigurierst. Woher hast Du diese falsche Vorlage?

junibart, 2020-12-17:
> Woher hast Du diese falsche Vorlage?

hier aus dem Wiki,  [https://github.com/andig/evcc/wiki/2.-wallbox#OpenWB](url)

junibart, 2020-12-17:
> ...und sollte funzen, tut es bei Dir aber nicht bzw es kommen einfach keine Daten. Das wäre ein Bug. 
> Ob _wirklich_ keine Daten kommen könntest Du mit `mosquitto_sub` mal prüfen.?

Ich schau mir das mal an. Obwohl ich meine, dass da was geht, denn mit dem Aufrufparameter "charger" kommen ja Informationen:

````
pi@EVCC:~ $ ./evcc/evcc charger 
Power:         5371W 
Enabled:       true 
Charge status: C
`````

junibart, 2020-12-17:
So, ich habe die MQTT-Topics der OpenWB mal mit dem MQTT-Snooper beobachtet. 
Es scheint so zu sein, dass das Topic boolChargeStat nicht regelmäßig gesendet wird, sondern nur bei Wechsel des Zustandes, z.B. beim Abstecken. 
Sieht man in folgendem Log-Auszug ganz gut, nach dem Abstecken (Ladeleistung ->0) wird das Topic einmal übertragen, danach beginnt die "outdated"-Zeit wieder bei Null. 

```
[lp-1  ] DEBUG 2020/12/17 18:42:15 charge power: 2509W
[lp-1  ] ERROR 2020/12/17 18:42:15 charger error: openWB/lp/1/boolChargeStat outdated: 18m44s
[lp-1  ] DEBUG 2020/12/17 18:42:25 charge power: 2410W
[lp-1  ] ERROR 2020/12/17 18:42:25 charger error: openWB/lp/1/boolChargeStat outdated: 18m54s
[lp-1  ] DEBUG 2020/12/17 18:42:35 charge power: 2410W
[lp-1  ] ERROR 2020/12/17 18:42:35 charger error: openWB/lp/1/boolChargeStat outdated: 19m4s
[lp-1  ] DEBUG 2020/12/17 18:42:45 charge power: 0W
[lp-1  ] ERROR 2020/12/17 18:42:45 charger error: openWB/lp/1/boolChargeStat outdated: 19m14s
[lp-1  ] DEBUG 2020/12/17 18:42:55 charge power: 0W
[lp-1  ] ERROR 2020/12/17 18:42:55 charger error: openWB/lp/1/boolChargeStat outdated: 5s
[lp-1  ] DEBUG 2020/12/17 18:43:05 charge power: 0W
[lp-1  ] ERROR 2020/12/17 18:43:06 charger error: openWB/lp/1/boolChargeStat outdated: 16s
[lp-1  ] DEBUG 2020/12/17 18:43:15 charge power: 0W
[lp-1  ] ERROR 2020/12/17 18:43:15 charger error: openWB/lp/1/boolChargeStat outdated: 25s

```
Eine Frage, eigentlich bin ich immer noch beim Testen der SoC-Abfrage beim neuen Nissan Leaf, die aus meiner Sicht immer noch hakelig ist... ist der hier angesprochene Fehler kritisch für die Ausführung von evcc? Bevor ich Fehler bei der SoC-Abfrage melde, möchte ich sicher sein, dass alles andere läuft. 

Danke für die engagierte Entwicklungsarbeit hier. Einfach klasse...

andig, 2020-12-17:
Sollte kein Problem sein, aber ich frage mal @snaptec

snaptec, 2020-12-17:
Plug/chargestat wird nur bei Änderung published, ist aber retained.

andig, 2020-12-18:
> Plug/chargestat wird nur bei Änderung published, ist aber retained.

Ok. Bisher erwarte ich auf allen Topics regelmäßige Aktualisierung als "healthy" Erkennung. Ich ändere das für das Topic.

andig, 2020-12-18:
@junibart willst Du #529 mal ausprobieren?

andig, 2020-12-18:
@snaptec bei welchen Topics kann ich mich denn darauf verlassen, dass sie wirklich aktualisiert werden?

    updating charge meter: openWB/lp/1/W outdated: 10s

snaptec, 2020-12-18:
Alle sind retained und werden nur bei Änderung aktualisiert.

junibart, 2020-12-18:
Hab #529 mal gezogen und evcc neu gebaut.... Ohne Gewähr, dass das alles geklappt hat. Bin Laie...
```
pi@EVCC:~/evcc $ make build
Version: 0.37 2020-12-18_12:16:45
go build -v -tags=release -ldflags '-X "github.com/andig/evcc/server.Version=0.37" -X "github.com/andig/evcc/server.Commit=9347cfa"'
```
Die Fehlermeldung kommt nach wie vor: 
```
[lp-1  ] DEBUG 2020/12/18 12:17:11 charge power: 0W
[lp-1  ] ERROR 2020/12/18 12:17:11 charger error: openWB/lp/1/boolChargeStat outdated: 7m5s
[lp-1  ] DEBUG 2020/12/18 12:17:21 charge power: 0W
[lp-1  ] ERROR 2020/12/18 12:17:21 charger error: openWB/lp/1/boolChargeStat outdated: 7m15s
[lp-1  ] DEBUG 2020/12/18 12:17:31 charge power: 0W
[lp-1  ] ERROR 2020/12/18 12:17:31 charger error: openWB/lp/1/boolChargeStat outdated: 7m25s

```

andig, 2020-12-18:
Ich hab den PR aktualisiert.

andig, 2020-12-18:
Ich habs einfach mal gemerged- sag Bescheid falls es noch klemmen sollte!

closed 2020-12-18 by andig

junibart, 2020-12-19:
Hallo, 
sieht alles gut aus. Fehlermeldungen sind weg.

beatboxking, 2021-10-01:
@junibart Hallo junibart, ich bekomme leider auch folgende Fehlermeldungen im Abstand von 10s:
charger: openWB/lp/1/boolChargeStat outdated: 1h8m50s
charge meter: openWB/lp/1/APhase1 outdated: 2m50s
charger: openWB/lp/1/boolChargeStat outdated: 1h8m30s

Allerdings ist es mir bisher noch gar nicht gelungen, die openWB über EVCC zu steuern. Kannst Du mir verraten, welche Einstellungen Du an der openWB vornehmen musstest? Ich habe eine OpenWB series 2 mit 1p3p Umschaltung...

Danke!