evcc-io/evcc #534: Invalid Web App Manifest / Add to home screen not working

rivengh, 2020-12-19:
**Describe the bug**
Add to home screen feature not working correctly with evcc.

**To Reproduce**
Steps to reproduce the behavior:
Open evcc with mobile browser and select "Add to home screen".
A shortcut is created on the home screen, but without an icon and with an invalid URL (e.g.": https://evcc/dist/#/).

**Expected behavior**
A shortcut is created on the home screen, with evcc icon and correct start URL.

**EVCC details:**
evcc 0.37

rivengh, 2020-12-19:
[image]

andig, 2020-12-19:
/cc @naltatis I have the same problem. Looks like another bug in parcel.js. Any idea how to fix this?

naltatis, 2020-12-19:
I already looked into fixing this in parcel, but since this is not our only issue with relative paths it might be easier to switch from parcel to vue-cli (Webpack). I already have a version of this working in a local branch.

mark-sch, 2020-12-21:
Encountered other side effects (necessary workarounds) with parcel also: e.g. adding a simple, relative "a href" seems to be impossible. Needed this ugly workaround, a link like "/broker/login" is impossible:

`<div class="navbar-nav">
        <router-link class="nav-item nav-link pb-1" to="/">Laden</router-link>
        <router-link class="nav-item nav-link pb-1" to="/config">Konfiguration</router-link>
        <script>document.write('<a class="nav-item nav-link pb-1" href="' + document.location.origin + '/broker">Sunny5-Broker</a>')</script>
        <script>document.write('<a class="nav-item nav-link pb-1" href="' + document.location.origin + '/broker/login">Login</a>')</script>
      </div>`

So I would appreciate to drop the parcel usage also.

PanNwt, 2020-12-25:
For me it looks like the relative path inside the webmanifest with only one folder up is the problem. In the release 31/32(?) the deployment of the assets were changed and now the manifest file locates in a different two level deep folder (source and published). Adding another ../ should solve this issue.

A current workaround is to have two container with evcc. One with version 0.30 for creating the homelink and one with the latest version for working with.

andig, 2021-01-02:
It's tool-generated. Problem is that the bundling tool does not work with relative paths.

closed 2021-01-02 by andig