evcc-io/evcc #539: Lademodus wechselt auf "PV"

junibart, 2020-12-22:
Ich möchte EVCC ausschließlich im "Now"-Mode einsetzen, die PV-Modi brauche ich nicht. 
Im "loadpoints"-Abschnitt der evcc.yaml habe ich unter der vehicle-Definition `mode:now` eingetragen. 

Nach dem Start von evcc wird dieser mode auch bestätigt. Dann wird kurz darauf auf PV umgeschaltet- mit der Konsequenz, dass nicht geladen wird (weil es bei mir keinen PV-Strom gibt).  Siehe Protokoll-Auszug. 
Erst nach Aufruf der Webseite und manuellem Umschalten des modes auf "Now" beginnt der Ladevorgang. 
Ich würde erwarten, dass der initial gesetzte mode beibehalten wird, bis er geändert wird. 

Protokoll-Auszug: 
```
[lp-1  ] WARN 2020/12/22 15:15:58 poll interval '15m0s' is lower than 1h0m0s and may deplete your b$
[lp-1  ] INFO 2020/12/22 15:15:59 loadpoint 1:
[lp-1  ] INFO 2020/12/22 15:15:59   mode:      now
[lp-1  ] INFO 2020/12/22 15:15:59   charger:   power — energy — currents — timer —
[lp-1  ] INFO 2020/12/22 15:15:59   meters:    charge ✓
[lp-1  ] INFO 2020/12/22 15:15:59     charge:  power ✓ energy ✓ currents —
[lp-1  ] INFO 2020/12/22 15:15:59   vehicles:  ✓
[lp-1  ] INFO 2020/12/22 15:15:59     car 0:   estimate ✓ status ✓ climate —
[lp-1  ] DEBUG 2020/12/22 15:15:59 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:15:59 charger status: A
[lp-1  ] INFO 2020/12/22 15:15:59 car disconnected
[lp-1  ] INFO 2020/12/22 15:15:59 set charge mode: pv
[lp-1  ] INFO 2020/12/22 15:15:59 set target soc: 100
[lp-1  ] DEBUG 2020/12/22 15:16:00 vehicle soc: 46%
[lp-1  ] DEBUG 2020/12/22 15:16:00 vehicle range: 123km
[lp-1  ] DEBUG 2020/12/22 15:16:01 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:16:01 charger status: A
[lp-1  ] DEBUG 2020/12/22 15:16:29 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:16:29 charger status: A
[lp-1  ] DEBUG 2020/12/22 15:16:59 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:16:59 charger status: A
[lp-1  ] DEBUG 2020/12/22 15:17:29 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:17:29 charger status: A
[lp-1  ] DEBUG 2020/12/22 15:17:59 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:17:59 charger status: A
[lp-1  ] DEBUG 2020/12/22 15:18:29 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:18:29 charger status: B
[lp-1  ] DEBUG 2020/12/22 15:18:29 charge start energy: 657kWh
[lp-1  ] INFO 2020/12/22 15:18:29 car connected
[lp-1  ] DEBUG 2020/12/22 15:18:29 vehicle soc: 46%
[lp-1  ] DEBUG 2020/12/22 15:18:29 vehicle range: 123km
[lp-1  ] DEBUG 2020/12/22 15:18:29 max charge current: 0A = 0A + -0A (0W @ 3p)
[lp-1  ] DEBUG 2020/12/22 15:18:29 target charge current: 0A
[lp-1  ] DEBUG 2020/12/22 15:18:59 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:18:59 charger status: B
[lp-1  ] DEBUG 2020/12/22 15:18:59 max charge current: 0A = 0A + -0A (0W @ 3p)
[lp-1  ] DEBUG 2020/12/22 15:18:59 target charge current: 0A
[lp-1  ] DEBUG 2020/12/22 15:19:29 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:19:29 charger status: B
[lp-1  ] DEBUG 2020/12/22 15:19:29 max charge current: 0A = 0A + -0A (0W @ 3p)
[lp-1  ] DEBUG 2020/12/22 15:19:29 target charge current: 0A
[lp-1  ] DEBUG 2020/12/22 15:19:59 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:19:59 charger status: B
[lp-1  ] DEBUG 2020/12/22 15:19:59 max charge current: 0A = 0A + -0A (0W @ 3p)
[lp-1  ] DEBUG 2020/12/22 15:19:59 target charge current: 0A
[lp-1  ] INFO 2020/12/22 15:20:20 set charge mode: now
[lp-1  ] DEBUG 2020/12/22 15:20:21 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 15:20:22 charger status: B
[lp-1  ] DEBUG 2020/12/22 15:20:22 charger enable
[lp-1  ] DEBUG 2020/12/22 15:20:29 charge power: 4220W
[lp-1  ] DEBUG 2020/12/22 15:20:29 charger status: C
[lp-1  ] DEBUG 2020/12/22 15:20:29 charge start energy: 657kWh
[lp-1  ] INFO 2020/12/22 15:20:29 start charging ->
[lp-1  ] DEBUG 2020/12/22 15:20:29 vehicle soc: 46%
[lp-1  ] DEBUG 2020/12/22 15:20:29 vehicle range: 123km
[lp-1  ] DEBUG 2020/12/22 15:20:59 charge power: 4197W
```

Auszug aus evcc.yaml:
```
# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Carport # display name for UI
  charger: simpleevse #openwb # charger
  meters:
    charge: charge # charge meter
  vehicle: leaf
  mode: now
  soc:
    poll:
      mode: charging
      interval: 15m
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    estimate: true # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 40
    - 60
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge to 100%
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to en$
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 20 # maximum charge current (default 16A)
```

andig, 2020-12-23:
Ich sehe:

    [lp-1  ] INFO 2020/12/22 15:15:59 car disconnected
    [lp-1  ] INFO 2020/12/22 15:15:59 set charge mode: pv

und:

    onDisconnect: # set defaults when vehicle disconnects
        mode: pv # switch back to pv mode
        targetSoC: 100 # charge to 100%

sieht für mich logisch aus?

junibart, 2020-12-25:
> Ich sehe:
> 
> ```
> [lp-1  ] INFO 2020/12/22 15:15:59 car disconnected
> [lp-1  ] INFO 2020/12/22 15:15:59 set charge mode: pv
> ```
> und:
> ```
> onDisconnect: # set defaults when vehicle disconnects
>     mode: pv # switch back to pv mode
>     targetSoC: 100 # charge to 100%
> ```
> sieht für mich logisch aus?

Jein... den Config-Abschnitt `onDisconnect: # set defaults when vehicle disconnects` hatte ich als die Aktion nach dem Abziehen des Ladekabels gesehen. Da muss ich tatsächlich auch "Now" eintragen. 
Offenbar wird aber auch der Umstand, dass EVCC gestartet wird, ohne dass ein Ladekabel steckt (bei mir der Normalfall) als Auslöser dieses Ereignisses gewertet. Das hatte ich so nicht erwartet. 
Zur Klarstellung könnte man den Kommentar in der evcc.(dist).yaml von "when vehicle disconnects" auf "when vehicle is disconnected" stellen... 
Ansonsten kann das issue geschlossen werden.

andig, 2020-12-25:
> Offenbar wird aber auch der Umstand, dass EVCC gestartet wird, ohne dass ein Ladekabel steckt (bei mir der Normalfall) als Auslöser dieses Ereignisses gewertet. Das hatte ich so nicht erwartet.

Das passiert einmalig damit evcc beim Start den aktuellen Stand erkennen kann. Im laufenden Betrieb sollte evcc später eigentlich nicht ständig neu gestartet werden.

closed 2020-12-25 by andig