evcc-io/evcc #540: SoC-Abfrage, Interpolation, Restzeitberechnung fehleranfällig

junibart, 2020-12-22:
Hallo, 

um die etwas spezielle SoC-Abfrage beim Nissan Leaf zu testen, habe ich einen Ladevorgang einmal genau beobachtet und bin auf einige Ungereimtheiten gestoßen. 

1. Mit der Konfiguration "poll: mode: charging" würde ich erwarten, dass die SoC-Abfrage erst bei Anstecken des Fahrzeugs beginnt. Tatsächlich wird das Fahrzeug bereits einmal bei Programmstart abgefragt. Ist das so beabsichtigt? Das sollte m.E. nur bei "poll: mode: always" passieren. 

2. Die Nissan-API liefert bekanntermaßen nicht gleich einen aktuellen SoC-Wert zurück, sondern unter Umständen einen veralteten Wert aus dem Cache. So auch hier: Die erste Abfrage (8:19:20 Uhr) lieferte einen SoC von 33% zurück, tatsächlich lag der SoC jedoch bei 37%. Dieser Wert traf erst später ein, um (8:24:50 Uhr). Diese SoC-Differenz innerhalb weniger Minuten führte dann dazu, dass in der Folgezeit die Interpolation der Zwischenwerte im 30-Sekunden-Takt um ein Prozent bis auf 58% kletterte, bis um 8:35:50 Uhr mit 38% ein neuer Wert von der Nissan-API eintraf. 
Das Spiel wiederholte sich dann bis 08:41:20 noch einmal (Anstieg bis 47%, dann Korrektur auf 40%). Ab dann funktionierte die Interpolation des SoC. 

3) Mit diesem Effekt ging offenbar auch eine grob falsche Berechnung der Rest-Ladezeit einher, die Beträge waren nur halb so hoch, wie es aus zu ladender Strommenge und aktueller Ladeleistung zu erwarten war. 

Irgendwie müsste der SoC-Wert noch plausibilisiert werden, die Verwendung veralteter Werte bringt nichts bzw. nur Ärger. 

 
Protokoll des Ladevorganges:
```
[lp-1  ] WARN 2020/12/22 08:19:18 poll interval '15m0s' is lower than 1h0m0s and may deplete your b$
[lp-1  ] INFO 2020/12/22 08:19:18 loadpoint 1:
[lp-1  ] INFO 2020/12/22 08:19:18   mode:      now
[lp-1  ] INFO 2020/12/22 08:19:18   charger:   power — energy — currents — timer —
[lp-1  ] INFO 2020/12/22 08:19:18   meters:    charge ✓
[lp-1  ] INFO 2020/12/22 08:19:18     charge:  power ✓ energy ✓ currents —
[lp-1  ] INFO 2020/12/22 08:19:18   vehicles:  ✓
[lp-1  ] INFO 2020/12/22 08:19:18     car 0:   estimate ✓ status ✓ climate —
[lp-1  ] DEBUG 2020/12/22 08:19:18 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:19:19 charger status: A
[lp-1  ] INFO 2020/12/22 08:19:19 car disconnected
[lp-1  ] INFO 2020/12/22 08:19:19 set charge mode: pv
[lp-1  ] INFO 2020/12/22 08:19:19 set target soc: 100
[lp-1  ] DEBUG 2020/12/22 08:19:20 vehicle soc: 33%
[lp-1  ] DEBUG 2020/12/22 08:19:20 vehicle range: 82km
[lp-1  ] DEBUG 2020/12/22 08:19:20 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:19:20 charger status: A
[lp-1  ] INFO 2020/12/22 08:19:39 set charge mode: now
[lp-1  ] DEBUG 2020/12/22 08:19:39 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:19:40 charger status: A
[lp-1  ] INFO 2020/12/22 08:19:40 set target soc: 60
[lp-1  ] DEBUG 2020/12/22 08:19:40 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:19:40 charger status: A
[lp-1  ] DEBUG 2020/12/22 08:19:49 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:19:49 charger status: A
[lp-1  ] DEBUG 2020/12/22 08:20:19 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:20:19 charger status: A
[lp-1  ] DEBUG 2020/12/22 08:20:49 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:20:49 charger status: A
[lp-1  ] DEBUG 2020/12/22 08:21:19 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:21:19 charger status: B
[lp-1  ] DEBUG 2020/12/22 08:21:19 charge start energy: 648kWh
[lp-1  ] INFO 2020/12/22 08:21:19 car connected
[lp-1  ] DEBUG 2020/12/22 08:21:19 vehicle soc: 33%
[lp-1  ] DEBUG 2020/12/22 08:21:19 vehicle range: 82km
[lp-1  ] DEBUG 2020/12/22 08:21:20 charger enable
[lp-1  ] DEBUG 2020/12/22 08:21:49 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:21:50 charger status: B
[lp-1  ] WARN 2020/12/22 08:21:50 charger out of sync
[lp-1  ] DEBUG 2020/12/22 08:22:19 charge power: 0W
[lp-1  ] DEBUG 2020/12/22 08:22:19 charger status: B
[lp-1  ] WARN 2020/12/22 08:22:19 charger out of sync
[lp-1  ] DEBUG 2020/12/22 08:22:49 charge power: 4286W
[lp-1  ] DEBUG 2020/12/22 08:22:49 charger status: C
[lp-1  ] DEBUG 2020/12/22 08:22:49 charge start energy: 648kWh
[lp-1  ] INFO 2020/12/22 08:22:49 start charging ->
[lp-1  ] DEBUG 2020/12/22 08:22:49 vehicle soc: 33%
[lp-1  ] DEBUG 2020/12/22 08:22:49 vehicle range: 82km
[lp-1  ] DEBUG 2020/12/22 08:23:18 charge power: 4302W
[lp-1  ] DEBUG 2020/12/22 08:23:19 charger status: C
[lp-1  ] DEBUG 2020/12/22 08:23:19 vehicle soc: 33%
[lp-1  ] DEBUG 2020/12/22 08:23:19 vehicle range: 82km
[lp-1  ] DEBUG 2020/12/22 08:23:49 charge power: 4261W
[lp-1  ] DEBUG 2020/12/22 08:23:49 charger status: C
[... 173 lines cut ...]
[lp-1  ] DEBUG 2020/12/22 08:44:49 vehicle soc: 41%
[lp-1  ] DEBUG 2020/12/22 08:44:49 vehicle range: 98km
[lp-1  ] DEBUG 2020/12/22 08:45:19 charge power: 4267W
[lp-1  ] DEBUG 2020/12/22 08:45:19 charger status: C
[lp-1  ] DEBUG 2020/12/22 08:45:19 vehicle soc: 41%
[lp-1  ] DEBUG 2020/12/22 08:45:19 vehicle range: 98km
[lp-1  ] DEBUG 2020/12/22 08:45:48 charge power: 4294W
[lp-1  ] DEBUG 2020/12/22 08:45:49 charger status: C
[lp-1  ] DEBUG 2020/12/22 08:45:49 vehicle soc: 41%
[lp-1  ] DEBUG 2020/12/22 08:45:49 vehicle range: 98km
[lp-1  ] DEBUG 2020/12/22 08:46:18 charge power: 4278W
[lp-1  ] DEBUG 2020/12/22 08:46:19 charger status: C
[lp-1  ] DEBUG 2020/12/22 08:46:19 vehicle soc: 41%
[lp-1  ] DEBUG 2020/12/22 08:46:19 vehicle range: 98km
[lp-1  ] DEBUG 2020/12/22 08:46:48 charge power: 4312W
[lp-1  ] DEBUG 2020/12/22 08:46:49 charger status: C
[lp-1  ] DEBUG 2020/12/22 08:46:50 vehicle soc: 41%
[lp-1  ] DEBUG 2020/12/22 08:46:50 vehicle range: 102km
[lp-1  ] DEBUG 2020/12/22 08:47:19 charge power: 4321W
[lp-1  ] DEBUG 2020/12/22 08:47:20 charger status: C
[lp-1  ] DEBUG 2020/12/22 08:47:20 vehicle soc: 41%
[lp-1  ] DEBUG 2020/12/22 08:47:20 vehicle range: 102km
[lp-1  ] INFO 2020/12/22 08:47:29 set target soc: 60
[lp-1  ] DEBUG 2020/12/22 08:47:29 charge power: 4312W
```

andig, 2020-12-23:
> Mit der Konfiguration "poll: mode: charging" würde ich erwarten, dass die SoC-Abfrage erst bei Anstecken des Fahrzeugs beginnt. Tatsächlich wird das Fahrzeug bereits einmal bei Programmstart (nicht angeschlossen) abgefragt. Ist das so beabsichtigt? Das sollte m.E. nur bei "poll: mode: always" passieren.

Korrekt. Da stimmt etwas nicht. Schaue ich mir an.

> Die Nissan-API liefert bekanntermaßen nicht gleich einen aktuellen SoC-Wert zurück

Aktuell schubse ich das Update erst an und hole dann den Wert. Mein Eindruck bisher war, dass auch der aktuelle Wert kommt. Anscheinend aber nicht?

> Mit diesem Effekt ging offenbar auch eine grob falsche Berechnung der Rest-Ladezeit einher, die Beträge waren nur halb so hoch, wie es aus zu ladender Strommenge und aktueller Ladeleistung zu erwarten war.

Es fehlt die Config im Ticket. Hast Du `estimate: true` gesetzt? Falls ja bitte mal aus machen. Dann liefert Dir das Nissan API selbst die Restladezeit, vielleicht hilft das ja?

andig, 2020-12-23:
> Mit der Konfiguration "poll: mode: charging" würde ich erwarten, dass die SoC-Abfrage erst bei Anstecken des Fahrzeugs beginnt. Tatsächlich wird das Fahrzeug bereits einmal bei Programmstart (nicht [...]

> Korrekt. Da stimmt etwas nicht. Schaue ich mir an.

Fixed

andig, 2020-12-28:
@junibart Problem erledigt? Dann könnten wir hier zumachen?

junibart, 2020-12-29:
> Es fehlt die Config im Ticket. Hast Du `estimate: true` gesetzt? Falls ja bitte mal aus machen. Dann liefert Dir das Nissan API selbst die Restladezeit, vielleicht hilft das ja?

wenn ich estimate abschalte, dann tritt der im ersten Post beschriebene Fehler (falsche SoC-Prozentwerte) nicht mehr auf. Alllerdings sehe ich dann keine Prognose über die verbleibende Ladezeit mehr- die Nissan-API liefert sie nicht oder sie wird nicht dargestellt... auf jeden Fall fehlt sie. 

Auf die Restzeitanzeige würde ich ungern verzichten, zumal sie sich aus den verbleibenden SoC-Prozentpunkten, der Akkukapazität und der aktuellen Ladestromstärke recht einfach berechnen lässt. Kannst Du das nicht aufteilen- "estimate" nur für SoC-Zwischenwerte, die Restzeit aber immer berechnen?

Auch die SoC-Estimation würde ich nicht einfach abschalten, sondern robuster gestalten, ggf. auch anhand der geladenen Energiemenge plausibilisieren. Wenn man die SoC-Abfrage bei der Hersteller-API z.B. nur alle halbe Stunde machen will (z.B. weil eine häufigere Abfrage als Missbrauch gesehen wird), werden im worst case dieser Zeit an einer 22KW-Wallbox 11KWh in den  Akku gepumpt, das sind bei einem 50KWh-Akku 20%SoC... da wird das Erreichen eines Ziel-SoC dann zum Glücksspiel.

Leider konnte ich den ursprünglichen Fehler nicht mehr reproduzieren- die SoC-Abfrage lieferte danach immer zuverlässig den richtigen Wert beim Ladebeginn.  

Ein Hinweis noch:  

Im Log wird angezeigt: 
`[lp-1  ] INFO 2020/12/22 08:19:18     car 0:   estimate ✓ status ✓ climate —`

Der Haken bei estimate bleibt auch stehen, wenn der Parameter `estimate `ausgeschaltet ist.

premultiply, 2020-12-29:
Der SoC-Estimator kann stark abweichende Batteriekapazitäten benutzen wenn
1) die gemeldeten SoC-Sprünge sehr klein sind. Das tritt auch bei häufen Unterbrechungen auf. Hier ist der Fehlerstreuung dann durch die Ganzzahlauflösuung des SoC und das unpräzise Timing (Jitter) der Update recht groß.
2) die Fahrzeug-API verzögert SoC-Werte meldet (z. B. bei Ladeunterbrechungen) die nicht mehr zum aktuellen Ladezustand passen. Hier findet momentan kein Abgleich statt und so werden mitunter Samples verarbeitet die eigentlich nicht verarbeitet werden dürften. Das ist eine bekannte unzulänglichkeit die ich aber selbst nicht beheben kann, da mir die notwendigen Schnittstellen bzw. Daten im SoC-Estimator und die go-Kenntnisse dazu fehlen dies entsprechend zu erweitern. Woran das liegt und was zu tun ist mir klar aber wie genau das im Code und im Klassenmodell von EVCC umsetzbar ist bin ich überfragt.

Wichtig: Alle diese derzeitigen Ungenauigkeiten betreffen jeweils immer nur ein SoC-Updateintervall (also zwischen zwei SoC-Samples). Läuft der Ladevorgang kontinuierlich und es kommt z. B. alle 30 Minuten ein Update sollte das sehr präzise funktionieren.

andig, 2020-12-29:
> Wichtig: Alle diese derzeitigen Ungenauigkeiten betreffen jeweils immer nur ein SoC-Updateintervall (also zwischen zwei SoC-Samples). Läuft der Ladevorgang kontinuierlich und es kommt z. B. alle 30 Minuten ein Update sollte das sehr präzise funktionieren.

Ich denke auch- diese Ungenauigkeit muss akzeptiert werden.

andig, 2020-12-29:
> Allerdings sehe ich dann keine Prognose über die verbleibende Ladezeit mehr- die Nissan-API liefert sie nicht oder sie wird nicht dargestellt... auf jeden Fall fehlt sie.

> [lp-1 ] INFO 2020/12/22 08:19:18 car 0: estimate ✓ status ✓ climate —

> Der Haken bei estimate bleibt auch stehen, wenn der Parameter estimate ausgeschaltet ist.

Genau der Haken bedeutet, dass Nissan die Restzeit liefert. Die Prognose müsste deshalb eigentlich immer kommen. Wenn sie das nicht tut bräuchten wir nochmal ein Logfile (`trace`) in dem das ersichtlich wird.

> Auch die SoC-Estimation würde ich nicht einfach abschalten, sondern robuster gestalten, ggf. auch anhand der geladenen Energiemenge plausibilisieren oder erstrangig aus der geladenen Energie ermitteln. Eventuell ist dafür die Eingabe einer Wandlereffizienz zusätzlich zur nutzbaren Akkukapazität hilfreich.

Schauen wir mal wie weit die Änderung vom @premultiply da hilft.

junibart, 2020-12-30:
> Genau der Haken bedeutet, dass Nissan die Restzeit liefert. Die Prognose müsste deshalb eigentlich immer kommen. Wenn sie das nicht tut bräuchten wir nochmal ein Logfile (`trace`) in dem das ersichtlich wird.

Gerne. Du bekommst ein Trace-Log eines Ladevorgangs (57%-80% SoC) per Mail. Weiss leider nicht, wo im Trace überall sensible Daten (VIN, Zugangsdaten etc.) drinstecken...

Estimation war ausgeschaltet. Interessant war, dass laut Log sehr wohl eine Rest-Ladezeit berechnet wird, die wird jedoch im Webinterface nicht angezeigt.

andig, 2020-12-30:
@junibart mir fehlt die Zeit alles selbst zu analysieren. Im Zweifel müsstest Du sensible Daten raus löschen und das Log hier posten damit @premultiply sich das auch anschauen kann.

andig, 2021-01-03:
@junibart das Nissan API scheint zickig zu sein:

```json
{
	"data": {
		"type": "Car",
		"id": "xx",
		"attributes": {
			"timeRequiredToFullNormal": 360,
			"rangeHvacOff": 57,
			"timeRequiredToFullSlow": 780,
			"chargeStatus": 0,
			"timeRequiredToFullFast": 180,
			"batteryCapacity": 40000,
			"batteryLevel": 22,
			"plugStatus": 0,
			"rangeHvacOn": 52,
			"batteryBarLevel": 54,
			"lastUpdateTime": "2020-12-26T15:16:01Z"
		}
	}
}
```

Hier fehlt (zumindest zeitweise) der `chargingRemainingTime` Eintrag- deshalb wird vmtl bei Dir auch nichts angezeigt.

Ich mache das Ticket mit dieser Erläuterung mal zu, gerne neues Ticket wenn notwendig.

andig, 2021-01-03:
PS.: da gibt es auch diverse andere Möglichkeiten, die sehen aber eher nach Standardwerten aus. Davon würde ich egtl. nichts verwenden wollen...

closed 2021-01-03 by andig