evcc-io/evcc #549: Einbindung Kona/Bluelink scheint nicht zu klappen

michisa86888, 2020-12-26:
Hallo zusammen,
ich bekomme die Abfrage des SoC von meinem Kona nicht zur Anzeige.
Meine aktuelle evcc.yaml:
`uri: 192.168.178.27:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: error
levels:
  core: debug
  lp-1: trace
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
meters:
- name: grid
  type: default
  power: # power reading
    type: mqtt # use mqtt plugin
    topic: fronius/verbrauch # mqtt topic
    timeout: 20s # don't use older values
- name: pv
  type: default
  power: # power reading
    type: mqtt # use mqtt plugin
    topic: fronius/leistung # mqtt topic
    timeout: 20s # don't use older values
    
  
 # charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:

- name: keba
  type: keba
  uri: 192.168.178.210:7090

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
- name: hyundai
  type: hyundai
  title: Kona # display name for UI
  capacity: 39 # kWh
  user: ******@**.com # user
  password: ***************** # password
  cache: 15m # cache API response

# site describes the EVU connection, PV and home battery
site:
  title: Home Ladestation # display name for UI
  meters:
    grid: grid # grid meter
    pv: pv # pv meter
    battery:  # battery meter
  prioritySoC: 60 # give home battery priority up to this soc (0 to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Garage # display name for UI
  charger: keba # charger
  vehicle: hyundai
  # vehicles: # use if multiple vehicles allowed to charge on this loadpoint
  # - ID.3
  # - e-Up
  mode: pv
  soc: 
    # polling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll: 
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: always
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 1m
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    estimate: false # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 30
    - 50
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge to 100%
  phases: 1 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

# mqtt message broker
mqtt:
   broker: localhost:1883
   topic: # root topic for publishing, set empty to disable
   user:
   password: 

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>`
Ich bin da leider echt kein Profi drin. Kann es sein das am loadpoint was falsch ist?
Bekomme auch im Log keine Fehlermeldung zum Fahrzeug:
`pi@raspberrypi:~ $ ./evcc
[lp-1  ] WARN 2020/12/26 17:01:10 poll mode '{always 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] ERROR 2020/12/26 17:01:11 charger error: recv timeout
[lp-1  ] INFO 2020/12/26 17:01:11 loadpoint 1:
[lp-1  ] INFO 2020/12/26 17:01:11   mode:      pv
[lp-1  ] INFO 2020/12/26 17:01:11   charger:   power ? energy ? currents ? timer —
[lp-1  ] INFO 2020/12/26 17:01:11   meters:    charge ?
[lp-1  ] INFO 2020/12/26 17:01:11     charge:  power ? energy ? currents ?
[lp-1  ] INFO 2020/12/26 17:01:11   vehicles:  ?
[lp-1  ] INFO 2020/12/26 17:01:11     car 0:   estimate — status — climate —
[lp-1  ] ERROR 2020/12/26 17:01:31 updating charge meter: recv timeout
[lp-1  ] ERROR 2020/12/26 17:01:32 charge rater error: recv timeout
[lp-1  ] ERROR 2020/12/26 17:01:33 charger error: recv timeout
[lp-1  ] ERROR 2020/12/26 17:01:36 updating charge meter: recv timeout
[lp-1  ] ERROR 2020/12/26 17:01:37 charge rater error: recv timeout
[lp-1  ] ERROR 2020/12/26 17:01:38 charger error: recv timeout`

michisa86888, 2020-12-26:
Mein Charger (Keba P30c) ist noch nicht installiert deshalb noch die Fehlermeldung im Log.
Hat jemand zufällig eine Beispiel evcc.yaml mit erfolgreicher Einbindung eines Konas?

andig, 2020-12-26:
Mach bitte ein "richtiges" Ticket auf mit Deiner Konfiguration. Zum Test des Autos einfach 

    evcc vehicle

starten, bei Bedarf mit ` --log trace`.

andig, 2020-12-28:
@michisa86888 gibts das Problem noch? Sonst könnten wir hier zumachen.

closed 2021-01-03 by andig