evcc-io/evcc #551: VW vehicle integration not working (with e-Golf)

henning89, 2020-12-27:
**Describe the bug**
I get the error message "502 Bad Gateway" when opening the EVCC user interface. It does not happen if I skip the vehicle part in the evcc.yaml, so I figured that there must be an issue with the vw vehicle integration

**Expected behavior**
EVCC user interface loading and showing the current battery status of my e-Golf.

**EVCC details:**
Here is an extract of the vehicle configuration in my evcc.yaml which apparently causes the problem:
`# vehicle definition
- type: vw
  title: Golf # display name for UI
  capacity: 36 # kWh
  user: mail@xyz.de # e-mail address which is registered on Volkswagen CarNet
  password: xyz # Password for Volkswagen CarNet
  vin: WVWZZZ... # I put it the full VIN of my e-Golf
  cache: 5m # cache API response`

andig, 2020-12-27:
Kannst Du mir Deine Logindaten unter cpuidle@gmx.de zum testen zur Verfügung stellen?

andig, 2020-12-28:
@henning89 Deine Daten funktionieren in kurzem Test. Bitte Konfiguration überprüfen und Ticket mit den notwendigen Daten vervollständigen.

henning89, 2020-12-28:
Ich nutze EVCC as Hassio Addon in Home Assistant. Ohne die Vehicle Konfiguration läuft es wie gesagt einwandfrei. Sobald ich diese in der evcc.yaml hinzufüge, kommt im Log dieser Eintrag:

20-12-28 11:25:57 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with f8b6a559_evcc!
20-12-28 11:25:57 INFO (SyncWorker_7) [supervisor.docker.interface] Cleaning addon_f8b6a559_evcc application
20-12-28 11:25:59 INFO (SyncWorker_7) [supervisor.docker.addon] Starting Docker add-on andig/evcc with version latest 

Kann ich noch irgendwie anders ein Logfile erstellen, aus dem mehr hervorgeht?

Anbei noch meine gesamte evcc.yaml, vielleicht liegt der Fehler irgendwo anders?

```
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: Photovoltaikanlage
  type: default
  power:
    type: modbus
    uri: 192.168.178.61:502
    id: 1 # ModBus slave id
    register: # manual register configuration
      address: 40067 # (40068 - 1) "Photovoltaikleistung in Watt"
      type: holding
      decode: int32s
    scale: -1 # reverse sign
- name: Hausbatterie
  type: default
  power:
    type: modbus
    uri: 192.168.178.61:502
    id: 1 # ModBus slave id
    register: # manual register configuration
      address: 40069
      type: holding
      decode: int32s
    scale: -1 # reverse direction
  soc:
    type: modbus
    uri: 192.168.178.61:502
    id: 1 # ModBus slave id
    register: # manual register configuration
      address: 40082
      type: holding
      decode: uint16
- name: Stromnetz
  type: default
  power:
    type: modbus
    uri: 192.168.178.61:502
    id: 1 # ModBus slave id
    register: # manual register configuration
      address: 40073 # "Netzbezugleistung in Watt"
      type: holding
      decode: int32s

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: go-e
  type: go-e
  uri: http://192.168.178.67

# vehicle definition
- type: vw
  title: Golf # display name for UI
  capacity: 36 # kWh
  user: henning@xyz.de # user
  password: xyz # password
  vin: WVWZZZ... # optional
  cache: 5m # cache API response

# add site heresite describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    pv: Photovoltaikanlage # pv meter
    battery: Hausbatterie # battery meter
    grid: Stromnetz
  prioritySoC: 60 # give home battery priority up to this soc (0 to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Carport # display name for UI
  charger: go-e # charger
  vehicle: Golf
  mode: now
  soc:
    min: 50
    target: 80
    alwaysUpdate: true # set true to update vehicle soc even when disconnected
    estimate: true # set true to interpolate between api updates
    levels:
    - 30
    - 50
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: now # switch back to pv mode
    targetSoC: 80 # charge to 80%
  enable: # pv mode enable behavior
    delay: 2m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 500 # maximum import power (W)
    ```

andig, 2020-12-28:
Du könntest es einfach probehealber mal ausserhalb von Hassio versuchen, auch Docker ist keine Pflicht.

andig, 2020-12-28:
Ah. Dein Auto ist schon richtig, aber Du hast es unter chargers eingetragen. Das ist natürlich falsch, da fehlt ein vehicles:!

closed 2020-12-28 by andig