evcc-io/evcc #557: Fronius Symo inkl. Smart Meter

loibi, 2020-12-31:
**Describe the bug**
Fronius Symo inkl. Smart Meter als Meter bringt fehler

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.38 (95cfe45)

```

Power:  read failed: modbus: response crc '2998' does not match expected '41017'
Energy: read failed: EOF

# meter definitions
meters:
- name: grid
  type: modbus
  model: sdm # SDM630
  uri: 192.168.1.6:502
  rtu: true # rs485 device connected using ethernet adapter
  id: 2
  power: Power # default value, optionally override
  energy: Sum # default value, optionally override

andig, 2021-01-02:
Wie ist denn der Zähler angeschlossen? Was hat der Fronius damit zu run? Gibts ein Logfile? Sicher dass das `rtu` in Deinem Fall richtig ist?

closed 2021-01-03 by loibi

premultiply, 2021-01-25:
Probier mal sowas:
```
meters:
- name: grid # smart meter via inverter
  type: modbus
  model: 213
  uri: 10.122.60.57:502
  id: 200
- name: pv
  type: modbus
  model: sunspec
  uri: 10.122.60.57:502
  id: 1
  scale: -1
```