evcc-io/evcc #568: evcc charges in PV mode after each container restart for 5 minutes regardless of PV production

goebelmeier, 2021-01-03:
**Describe the bug**
evcc charges in PV mode after each container restart for 5 minutes regardless of PV production

**To Reproduce**
Steps to reproduce the behavior:
1. Stop all containers
2. Start a new evcc container or let watchtower spawn a new one after updating
3. Have a look out of your window into the darkness
4. Get telegram message "Started charging in "pv" mode"
5. Wait 5 minutes
6. Get telegram message about finished charging because site power is over disable threshold

**Expected behavior**
evcc should not even start charging in pv mode if site power exceeds production.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
[13:02:01] openhabian@openHABianPi:~$ docker-compose exec evcc evcc -v
evcc version 0.38 (a0df9b4)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
config
======

grid: e3dc-grid
---------------
Power: -4W

pv: e3dc-pv
-----------
Power: -452W

battery: e3dc-battery
---------------------
Power: -24W
SoC:   11%

loadpoint 1
===========

charger: wallbe
---------------
Power:          0W
Energy:         7614.5kWh
Current L1..L3: 0.0314A 0A 0A
Enabled:        false
Duration:       1h5m58s
Charge status:  A
Diagnostic dump:
Firmware:	SL-01.03.08

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
[13:04:13] openhabian@openHABianPi:~$ cat evcc.yaml
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  vw: trace
  id: trace

# meter definitions
[... 592 lines cut ...]
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:15 max charge current: 0A = 0A + -1.1A (249W @ 1p)
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:15 target charge current: 0A
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:15 climater active: false, target temp: 22.5°C, outside temp: 3.5°C
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:24 charge power: 0W
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:24 charger status: B
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:24 max charge current: 0A = 0A + -1.1A (250W @ 1p)
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:24 target charge current: 0A
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:24 climater active: false, target temp: 22.5°C, outside temp: 3.5°C
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:34 charge power: 0W
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:34 charger status: B
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:35 max charge current: 0A = 0A + -1.1A (248W @ 1p)
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:35 target charge current: 0A
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:35 climater active: false, target temp: 22.5°C, outside temp: 3.5°C
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:44 charge power: 0W
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:44 charger status: B
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:44 max charge current: 0A = 0A + -1.1A (258W @ 1p)
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:44 target charge current: 0A
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:44 climater active: false, target temp: 22.5°C, outside temp: 3.5°C
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:54 charge power: 0W
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:54 charger status: B
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:55 max charge current: 0A = 0A + -1.1A (254W @ 1p)
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:55 target charge current: 0A
evcc_1           | [lp-1  ] DEBUG 2021/01/02 23:24:55 climater active: false, target temp: 22.5°C, outside temp: 3.5°C

</pre></details>

goebelmeier, 2021-01-04:
```
[main  ] INFO 2021/01/04 14:35:49 evcc 0.38 (cc969c3)
[main  ] INFO 2021/01/04 14:35:49 using config file /etc/evcc.yaml
[main  ] INFO 2021/01/04 14:35:49 listening at 0.0.0.0:7070
[mqtt  ] INFO 2021/01/04 14:35:49 connecting evcc-1914682306 at 192.168.4.4:1883
[mqtt  ] DEBUG 2021/01/04 14:35:49 192.168.4.4:1883 connected
[id    ] DEBUG 2021/01/04 14:35:53 found vehicle: WVWZZZAAZLD913543
[cache ] TRACE 2021/01/04 14:35:53 title: Ladepunkt
[cache ] TRACE 2021/01/04 14:35:53 minCurrent: 6
[cache ] TRACE 2021/01/04 14:35:53 maxCurrent: 16
[cache ] TRACE 2021/01/04 14:35:53 phases: 1
[cache ] TRACE 2021/01/04 14:35:53 activePhases: 1
[cache ] TRACE 2021/01/04 14:35:53 soc: true
[cache ] TRACE 2021/01/04 14:35:53 mode: pv
[cache ] TRACE 2021/01/04 14:35:53 targetSoC: 100
[cache ] TRACE 2021/01/04 14:35:53 minSoC: 33
[cache ] TRACE 2021/01/04 14:35:53 socLevels: [30 50 80 100]
[cache ] TRACE 2021/01/04 14:35:53 socTitle: VW e-Up
[cache ] TRACE 2021/01/04 14:35:53 socCapacity: 32
[wallbe] TRACE 2021/01/04 14:35:53 modbus: send 00 01 00 00 00 06 ff 01 01 90 00 01
[wallbe] TRACE 2021/01/04 14:35:53 modbus: recv 00 01 00 00 00 04 ff 01 01 00
[site  ] INFO 2021/01/04 14:35:53 site config:
[site  ] INFO 2021/01/04 14:35:53   meters:    grid ✓ pv ✓ battery ✓
[site  ] INFO 2021/01/04 14:35:53     grid:    power ✓ energy — currents —
[cache ] TRACE 2021/01/04 14:35:53 title: Home
[cache ] TRACE 2021/01/04 14:35:53 gridConfigured: true
[site  ] INFO 2021/01/04 14:35:53     pv:      power ✓ energy — currents —
[cache ] TRACE 2021/01/04 14:35:53 pvConfigured: true
[cache ] TRACE 2021/01/04 14:35:53 batteryConfigured: true
[site  ] INFO 2021/01/04 14:35:53     battery: power ✓ energy — currents — soc ✓
[lp-1  ] INFO 2021/01/04 14:35:53 loadpoint 1:
[lp-1  ] INFO 2021/01/04 14:35:53   mode:      pv
[lp-1  ] INFO 2021/01/04 14:35:53   charger:   power ✓ energy ✓ currents ✓ timer ✓
[lp-1  ] INFO 2021/01/04 14:35:53   meters:    charge ✓
[lp-1  ] INFO 2021/01/04 14:35:53     charge:  power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2021/01/04 14:35:53   vehicles:  ✓
[lp-1  ] INFO 2021/01/04 14:35:53     car 0:   finish ✓ status ✓ climate ✓
[cache ] TRACE 2021/01/04 14:35:53 chargeConfigured: true
[site  ] DEBUG 2021/01/04 14:35:53 ----
[modbus] TRACE 2021/01/04 14:35:53 modbus: send 00 01 00 00 00 06 01 03 9c 83 00 02
[modbus] TRACE 2021/01/04 14:35:53 modbus: recv 00 01 00 00 00 07 01 03 04 01 1f 00 00
[site  ] DEBUG 2021/01/04 14:35:53 pv power: -287W
[cache ] TRACE 2021/01/04 14:35:53 pvPower: -287
[influx] TRACE 2021/01/04 14:35:53 write pvPower=-287 (map[])
[modbus] TRACE 2021/01/04 14:35:53 modbus: send 00 02 00 00 00 06 01 03 9c 89 00 02
[modbus] TRACE 2021/01/04 14:35:53 modbus: recv 00 02 00 00 00 07 01 03 04 01 5e 00 00
[site  ] DEBUG 2021/01/04 14:35:53 grid power: 350W
[modbus] TRACE 2021/01/04 14:35:53 modbus: send 00 03 00 00 00 06 01 03 9c 85 00 02
[cache ] TRACE 2021/01/04 14:35:53 gridPower: 350
[influx] TRACE 2021/01/04 14:35:53 write gridPower=350 (map[])
[modbus] TRACE 2021/01/04 14:35:53 modbus: recv 00 03 00 00 00 07 01 03 04 00 00 00 00
[site  ] DEBUG 2021/01/04 14:35:53 battery power: -0W
[modbus] TRACE 2021/01/04 14:35:53 modbus: send 00 04 00 00 00 06 01 03 9c 92 00 01
[influx] TRACE 2021/01/04 14:35:53 write batteryPower=-0 (map[])
[cache ] TRACE 2021/01/04 14:35:53 batteryPower: -0
[modbus] TRACE 2021/01/04 14:35:53 modbus: recv 00 04 00 00 00 05 01 03 02 00 02
[site  ] DEBUG 2021/01/04 14:35:53 battery soc: 2%
[site  ] DEBUG 2021/01/04 14:35:53 site power: 350W
[wallbe] TRACE 2021/01/04 14:35:53 modbus: send 00 02 00 00 00 06 ff 04 00 78 00 02
[influx] TRACE 2021/01/04 14:35:53 write batterySoC=2 (map[])
[cache ] TRACE 2021/01/04 14:35:53 batterySoC: 2
[cache ] TRACE 2021/01/04 14:35:53 mode: pv
[wallbe] TRACE 2021/01/04 14:35:53 modbus: recv 00 02 00 00 00 07 ff 04 04 00 00 00 00
[lp-1  ] DEBUG 2021/01/04 14:35:53 charge power: 0W
[cache ] TRACE 2021/01/04 14:35:53 chargePower: 0
[influx] TRACE 2021/01/04 14:35:53 write chargePower=0 (map[loadpoint:Ladepunkt])
[wallbe] TRACE 2021/01/04 14:35:53 modbus: send 00 03 00 00 00 06 ff 04 00 66 00 02
[cache ] TRACE 2021/01/04 14:35:53 chargeCurrent: 0
[influx] TRACE 2021/01/04 14:35:53 write chargeCurrent=0 (map[loadpoint:Ladepunkt])
[wallbe] TRACE 2021/01/04 14:35:53 modbus: recv 00 03 00 00 00 07 ff 04 04 11 d6 00 00
[... 752 lines cut ...]
[cache ] TRACE 2021/01/04 14:37:33 chargeRemainingEnergy: 23111.111111111113
[cache ] TRACE 2021/01/04 14:37:33 range: 61
[influx] TRACE 2021/01/04 14:37:33 write chargeRemainingEnergy=23111.111111111113 (map[loadpoint:Ladepunkt])
[wallbe] TRACE 2021/01/04 14:37:33 modbus: recv 00 57 00 00 00 04 ff 01 01 01
[wallbe] TRACE 2021/01/04 14:37:33 modbus: send 00 58 00 00 00 06 ff 04 00 72 00 02
[wallbe] TRACE 2021/01/04 14:37:33 modbus: recv 00 58 00 00 00 07 ff 04 04 41 80 a8 dd
[wallbe] TRACE 2021/01/04 14:37:33 modbus: send 00 59 00 00 00 06 ff 04 00 74 00 02
[wallbe] TRACE 2021/01/04 14:37:33 modbus: recv 00 59 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/01/04 14:37:33 modbus: send 00 5a 00 00 00 06 ff 04 00 76 00 02
[wallbe] TRACE 2021/01/04 14:37:33 modbus: recv 00 5a 00 00 00 07 ff 04 04 00 00 00 00
[lp-1  ] TRACE 2021/01/04 14:37:33 charge currents: [16.1 0 0]A
[lp-1  ] DEBUG 2021/01/04 14:37:33 detected phases: 1p [16.1 0 0]A
[influx] TRACE 2021/01/04 14:37:33 write chargeCurrents=[16.08245277404785 0 0] (map[loadpoint:Ladepunkt])
[... cut ...]
[lp-1  ] DEBUG 2021/01/04 14:37:33 pv max charge current: 0A
[lp-1  ] DEBUG 2021/01/04 14:37:33 climater active: false, target temp: 22.5°C, outside temp: 22.5°C
[lp-1  ] DEBUG 2021/01/04 14:37:33 charger disable - contactor delay 3m20s
[cache ] TRACE 2021/01/04 14:37:33 chargeCurrents: [16.08245277404785 0 0]
[cache ] TRACE 2021/01/04 14:37:33 activePhases: 1
[cache ] TRACE 2021/01/04 14:37:33 timerSet: false
[cache ] TRACE 2021/01/04 14:37:33 climater: off
[cache ] TRACE 2021/01/04 14:37:33 remoteDisabled:
```

goebelmeier, 2021-01-04:
`containrrr/watchtower` getriggerter Restart:
```
[14:47:18] openhabian@openHABianPi:~$ docker logs 2c19b4653226
[main  ] INFO 2021/01/04 14:44:20 evcc 0.38 (5cadb98)
[main  ] INFO 2021/01/04 14:44:20 using config file /etc/evcc.yaml
[main  ] INFO 2021/01/04 14:44:20 listening at 0.0.0.0:7070
[mqtt  ] INFO 2021/01/04 14:44:20 connecting evcc-2066905290 at 192.168.4.4:1883
[mqtt  ] DEBUG 2021/01/04 14:44:20 192.168.4.4:1883 connected
[id    ] DEBUG 2021/01/04 14:44:24 found vehicle: WVWZZZABCDE123456
[cache ] TRACE 2021/01/04 14:44:24 title: Ladepunkt
[cache ] TRACE 2021/01/04 14:44:24 minCurrent: 6
[cache ] TRACE 2021/01/04 14:44:24 maxCurrent: 16
[cache ] TRACE 2021/01/04 14:44:24 phases: 1
[cache ] TRACE 2021/01/04 14:44:24 activePhases: 1
[cache ] TRACE 2021/01/04 14:44:24 soc: true
[cache ] TRACE 2021/01/04 14:44:24 mode: pv
[cache ] TRACE 2021/01/04 14:44:24 targetSoC: 100
[cache ] TRACE 2021/01/04 14:44:24 minSoC: 33
[cache ] TRACE 2021/01/04 14:44:24 socLevels: [30 50 80 100]
[cache ] TRACE 2021/01/04 14:44:24 socTitle: VW e-Up
[cache ] TRACE 2021/01/04 14:44:24 socCapacity: 32
[wallbe] TRACE 2021/01/04 14:44:24 modbus: send 00 01 00 00 00 06 ff 01 01 90 00 01
[wallbe] TRACE 2021/01/04 14:44:24 modbus: recv 00 01 00 00 00 04 ff 01 01 00
[site  ] INFO 2021/01/04 14:44:24 site config:
[site  ] INFO 2021/01/04 14:44:24   meters:    grid ✓ pv ✓ battery ✓
[site  ] INFO 2021/01/04 14:44:24     grid:    power ✓ energy — currents —
[site  ] INFO 2021/01/04 14:44:24     pv:      power ✓ energy — currents —
[cache ] TRACE 2021/01/04 14:44:24 title: Home
[cache ] TRACE 2021/01/04 14:44:24 gridConfigured: true
[cache ] TRACE 2021/01/04 14:44:24 pvConfigured: true
[cache ] TRACE 2021/01/04 14:44:24 batteryConfigured: true
[site  ] INFO 2021/01/04 14:44:24     battery: power ✓ energy — currents — soc ✓
[lp-1  ] INFO 2021/01/04 14:44:24 loadpoint 1:
[lp-1  ] INFO 2021/01/04 14:44:24   mode:      pv
[lp-1  ] INFO 2021/01/04 14:44:24   charger:   power ✓ energy ✓ currents ✓ timer ✓
[lp-1  ] INFO 2021/01/04 14:44:24   meters:    charge ✓
[lp-1  ] INFO 2021/01/04 14:44:24     charge:  power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2021/01/04 14:44:24   vehicles:  ✓
[lp-1  ] INFO 2021/01/04 14:44:24     car 0:   finish ✓ status ✓ climate ✓
[cache ] TRACE 2021/01/04 14:44:24 chargeConfigured: true
[site  ] DEBUG 2021/01/04 14:44:24 ----
[modbus] TRACE 2021/01/04 14:44:24 modbus: send 00 01 00 00 00 06 01 03 9c 83 00 02
[modbus] TRACE 2021/01/04 14:44:24 modbus: recv 00 01 00 00 00 07 01 03 04 00 e3 00 00
[site  ] DEBUG 2021/01/04 14:44:24 pv power: -227W
[cache ] TRACE 2021/01/04 14:44:24 pvPower: -227
[influx] TRACE 2021/01/04 14:44:24 write pvPower=-227 (map[])
[modbus] TRACE 2021/01/04 14:44:24 modbus: send 00 02 00 00 00 06 01 03 9c 89 00 02
[modbus] TRACE 2021/01/04 14:44:24 modbus: recv 00 02 00 00 00 07 01 03 04 00 77 00 00
[site  ] DEBUG 2021/01/04 14:44:24 grid power: 119W
[modbus] TRACE 2021/01/04 14:44:24 modbus: send 00 03 00 00 00 06 01 03 9c 85 00 02
[cache ] TRACE 2021/01/04 14:44:24 gridPower: 119
[influx] TRACE 2021/01/04 14:44:24 write gridPower=119 (map[])
[modbus] TRACE 2021/01/04 14:44:24 modbus: recv 00 03 00 00 00 07 01 03 04 00 00 00 00
[site  ] DEBUG 2021/01/04 14:44:24 battery power: -0W
[modbus] TRACE 2021/01/04 14:44:24 modbus: send 00 04 00 00 00 06 01 03 9c 92 00 01
[cache ] TRACE 2021/01/04 14:44:24 batteryPower: -0
[influx] TRACE 2021/01/04 14:44:24 write batteryPower=-0 (map[])
[modbus] TRACE 2021/01/04 14:44:24 modbus: recv 00 04 00 00 00 05 01 03 02 00 02
[site  ] DEBUG 2021/01/04 14:44:24 battery soc: 2%
[site  ] DEBUG 2021/01/04 14:44:24 site power: 119W
[wallbe] TRACE 2021/01/04 14:44:24 modbus: send 00 02 00 00 00 06 ff 04 00 78 00 02
[cache ] TRACE 2021/01/04 14:44:24 batterySoC: 2
[cache ] TRACE 2021/01/04 14:44:24 mode: pv
[influx] TRACE 2021/01/04 14:44:24 write batterySoC=2 (map[])
[wallbe] TRACE 2021/01/04 14:44:24 modbus: recv 00 02 00 00 00 07 ff 04 04 00 00 00 00
[lp-1  ] DEBUG 2021/01/04 14:44:24 charge power: 0W
[cache ] TRACE 2021/01/04 14:44:24 chargePower: 0
[wallbe] TRACE 2021/01/04 14:44:24 modbus: send 00 03 00 00 00 06 ff 04 00 66 00 02
[influx] TRACE 2021/01/04 14:44:24 write chargePower=0 (map[loadpoint:Ladepunkt])
[cache ] TRACE 2021/01/04 14:44:24 chargeCurrent: 0
[... 1347 lines cut ...]
[cache ] TRACE 2021/01/04 14:47:24 chargeRemainingEnergy: 22755.555555555555
[cache ] TRACE 2021/01/04 14:47:24 range: 62
[influx] TRACE 2021/01/04 14:47:24 write chargeRemainingEnergy=22755.555555555555 (map[loadpoint:Ladepunkt])
[wallbe] TRACE 2021/01/04 14:47:24 modbus: recv 00 97 00 00 00 04 ff 01 01 01
[wallbe] TRACE 2021/01/04 14:47:24 modbus: send 00 98 00 00 00 06 ff 04 00 72 00 02
[wallbe] TRACE 2021/01/04 14:47:24 modbus: recv 00 98 00 00 00 07 ff 04 04 41 80 d5 52
[wallbe] TRACE 2021/01/04 14:47:24 modbus: send 00 99 00 00 00 06 ff 04 00 74 00 02
[wallbe] TRACE 2021/01/04 14:47:24 modbus: recv 00 99 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/01/04 14:47:24 modbus: send 00 9a 00 00 00 06 ff 04 00 76 00 02
[wallbe] TRACE 2021/01/04 14:47:24 modbus: recv 00 9a 00 00 00 07 ff 04 04 00 00 00 00
[lp-1  ] TRACE 2021/01/04 14:47:24 charge currents: [16.1 0 0]A
[lp-1  ] DEBUG 2021/01/04 14:47:24 detected phases: 1p [16.1 0 0]A
[lp-1  ] DEBUG 2021/01/04 14:47:24 max charge current: 0A = 16A + -16A (3745W @ 1p)
[cache ] TRACE 2021/01/04 14:47:24 chargeCurrents: [16.10416030883789 0 0]
[lp-1  ] DEBUG 2021/01/04 14:47:24 site power 3745W >= disable threshold 200W
[... cut ...]
[lp-1  ] DEBUG 2021/01/04 14:47:24 pv disable timer elapsed
[lp-1  ] DEBUG 2021/01/04 14:47:24 pv max charge current: 0A
[cache ] TRACE 2021/01/04 14:47:24 timerSet: false
[lp-1  ] DEBUG 2021/01/04 14:47:24 climater active: false, target temp: 22.5°C, outside temp: 22.5°C
[influx] TRACE 2021/01/04 14:47:24 write chargeCurrents=[16.10416030883789 0 0] (map[loadpoint:Ladepunkt])
[lp-1  ] DEBUG 2021/01/04 14:47:24 charger disable - contactor delay 2m0s
[cache ] TRACE 2021/01/04 14:47:24 climater: off
[cache ] TRACE 2021/01/04 14:47:24 remoteDisabled:
```

andig, 2021-01-04:
Closed in 39b6228

goebelmeier, 2021-01-04:
Machst du noch mal auf? 39b6228 hilft leider nicht.
```
[18:37:58] openhabian@openHABianPi:~$ docker logs c208a9759285
[main  ] INFO 2021/01/04 18:37:04 evcc 0.39 (39b6228)
[main  ] INFO 2021/01/04 18:37:04 using config file /etc/evcc.yaml
[main  ] INFO 2021/01/04 18:37:04 listening at 0.0.0.0:7070
[mqtt  ] INFO 2021/01/04 18:37:04 connecting evcc-53114687 at 192.168.4.4:1883
[mqtt  ] DEBUG 2021/01/04 18:37:04 192.168.4.4:1883 connected
[id    ] DEBUG 2021/01/04 18:37:07 found vehicle: ...
[cache ] TRACE 2021/01/04 18:37:08 title: Ladepunkt
[cache ] TRACE 2021/01/04 18:37:08 minCurrent: 6
[cache ] TRACE 2021/01/04 18:37:08 maxCurrent: 16
[cache ] TRACE 2021/01/04 18:37:08 phases: 1
[cache ] TRACE 2021/01/04 18:37:08 activePhases: 1
[cache ] TRACE 2021/01/04 18:37:08 soc: true
[cache ] TRACE 2021/01/04 18:37:08 mode: pv
[cache ] TRACE 2021/01/04 18:37:08 targetSoC: 100
[cache ] TRACE 2021/01/04 18:37:08 minSoC: 33
[cache ] TRACE 2021/01/04 18:37:08 socLevels: [30 50 80 100]
[cache ] TRACE 2021/01/04 18:37:08 socTitle: VW e-Up
[cache ] TRACE 2021/01/04 18:37:08 socCapacity: 32
[wallbe] TRACE 2021/01/04 18:37:08 modbus: send 00 01 00 00 00 06 ff 01 01 90 00 01
[wallbe] TRACE 2021/01/04 18:37:08 modbus: recv 00 01 00 00 00 04 ff 01 01 00
[site  ] INFO 2021/01/04 18:37:08 site config:
[site  ] INFO 2021/01/04 18:37:08   meters:    grid ✓ pv ✓ battery ✓
[site  ] INFO 2021/01/04 18:37:08     grid:    power ✓ energy — currents —
[cache ] TRACE 2021/01/04 18:37:08 title: Home
[cache ] TRACE 2021/01/04 18:37:08 gridConfigured: true
[site  ] INFO 2021/01/04 18:37:08     pv:      power ✓ energy — currents —
[cache ] TRACE 2021/01/04 18:37:08 pvConfigured: true
[site  ] INFO 2021/01/04 18:37:08     battery: power ✓ energy — currents — soc ✓
[lp-1  ] INFO 2021/01/04 18:37:08 loadpoint 1:
[lp-1  ] INFO 2021/01/04 18:37:08   mode:      pv
[lp-1  ] INFO 2021/01/04 18:37:08   charger:   power ✓ energy ✓ currents ✓ timer ✓
[lp-1  ] INFO 2021/01/04 18:37:08   meters:    charge ✓
[lp-1  ] INFO 2021/01/04 18:37:08     charge:  power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2021/01/04 18:37:08   vehicles:  ✓
[lp-1  ] INFO 2021/01/04 18:37:08     car 0:   finish ✓ status ✓ climate ✓
[site  ] DEBUG 2021/01/04 18:37:08 ----
[cache ] TRACE 2021/01/04 18:37:08 batteryConfigured: true
[cache ] TRACE 2021/01/04 18:37:08 chargeConfigured: true
[modbus] TRACE 2021/01/04 18:37:08 modbus: send 00 01 00 00 00 06 01 03 9c 83 00 02
[modbus] TRACE 2021/01/04 18:37:08 modbus: recv 00 01 00 00 00 07 01 03 04 00 00 00 00
[site  ] DEBUG 2021/01/04 18:37:08 pv power: -0W
[modbus] TRACE 2021/01/04 18:37:08 modbus: send 00 02 00 00 00 06 01 03 9c 89 00 02
[cache ] TRACE 2021/01/04 18:37:08 pvPower: -0
[influx] TRACE 2021/01/04 18:37:08 write pvPower=-0 (map[])
[modbus] TRACE 2021/01/04 18:37:08 modbus: recv 00 02 00 00 00 07 01 03 04 01 36 00 00
[site  ] DEBUG 2021/01/04 18:37:08 grid power: 310W
[modbus] TRACE 2021/01/04 18:37:08 modbus: send 00 03 00 00 00 06 01 03 9c 85 00 02
[cache ] TRACE 2021/01/04 18:37:08 gridPower: 310
[influx] TRACE 2021/01/04 18:37:08 write gridPower=310 (map[])
[modbus] TRACE 2021/01/04 18:37:08 modbus: recv 00 03 00 00 00 07 01 03 04 00 00 00 00
[site  ] DEBUG 2021/01/04 18:37:08 battery power: -0W
[modbus] TRACE 2021/01/04 18:37:08 modbus: send 00 04 00 00 00 06 01 03 9c 92 00 01
[cache ] TRACE 2021/01/04 18:37:08 batteryPower: -0
[influx] TRACE 2021/01/04 18:37:08 write batteryPower=-0 (map[])
[modbus] TRACE 2021/01/04 18:37:08 modbus: recv 00 04 00 00 00 05 01 03 02 00 01
[site  ] DEBUG 2021/01/04 18:37:08 battery soc: 1%
[site  ] DEBUG 2021/01/04 18:37:08 site power: 310W
[wallbe] TRACE 2021/01/04 18:37:08 modbus: send 00 02 00 00 00 06 ff 04 00 78 00 02
[cache ] TRACE 2021/01/04 18:37:08 batterySoC: 1
[cache ] TRACE 2021/01/04 18:37:08 mode: pv
[influx] TRACE 2021/01/04 18:37:08 write batterySoC=1 (map[])
[wallbe] TRACE 2021/01/04 18:37:08 modbus: recv 00 02 00 00 00 07 ff 04 04 00 00 00 00
[lp-1  ] DEBUG 2021/01/04 18:37:08 charge power: 0W
[wallbe] TRACE 2021/01/04 18:37:08 modbus: send 00 03 00 00 00 06 ff 04 00 66 00 02
[cache ] TRACE 2021/01/04 18:37:08 chargePower: 0
[influx] TRACE 2021/01/04 18:37:08 write chargePower=0 (map[loadpoint:Ladepunkt])
[cache ] TRACE 2021/01/04 18:37:08 chargeCurrent: 0
[... 383 lines cut ...]
[wallbe] TRACE 2021/01/04 18:37:58 modbus: send 00 2f 00 00 00 06 ff 01 01 90 00 01
[cache ] TRACE 2021/01/04 18:37:58 chargeRemainingEnergy: 22044.444444444445
[cache ] TRACE 2021/01/04 18:37:58 range: 64
[wallbe] TRACE 2021/01/04 18:37:58 modbus: recv 00 2f 00 00 00 04 ff 01 01 01
[wallbe] TRACE 2021/01/04 18:37:58 modbus: send 00 30 00 00 00 06 ff 04 00 72 00 02
[wallbe] TRACE 2021/01/04 18:37:58 modbus: recv 00 30 00 00 00 07 ff 04 04 41 80 91 53
[wallbe] TRACE 2021/01/04 18:37:58 modbus: send 00 31 00 00 00 06 ff 04 00 74 00 02
[wallbe] TRACE 2021/01/04 18:37:58 modbus: recv 00 31 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/01/04 18:37:58 modbus: send 00 32 00 00 00 06 ff 04 00 76 00 02
[wallbe] TRACE 2021/01/04 18:37:58 modbus: recv 00 32 00 00 00 07 ff 04 04 00 00 00 00
[lp-1  ] TRACE 2021/01/04 18:37:58 charge currents: [16.1 0 0]A
[cache ] TRACE 2021/01/04 18:37:58 chargeCurrents: [16.070959091186523 0 0]
[influx] TRACE 2021/01/04 18:37:58 write chargeCurrents=[16.070959091186523 0 0] (map[loadpoint:Ladepunkt])
[lp-1  ] DEBUG 2021/01/04 18:37:58 detected phases: 1p [16.1 0 0]A
[lp-1  ] DEBUG 2021/01/04 18:37:58 max charge current: 0A = 16A + -17A (3893W @ 1p)
[... cut ...]
[lp-1  ] DEBUG 2021/01/04 18:37:58 pv disable timer elapsed
[lp-1  ] DEBUG 2021/01/04 18:37:58 pv max charge current: 0A
[lp-1  ] DEBUG 2021/01/04 18:37:58 climater active: false, target temp: 22.5°C, outside temp: 22.5°C
[lp-1  ] DEBUG 2021/01/04 18:37:58 charger disable - contactor delay 4m10s
[cache ] TRACE 2021/01/04 18:37:58 activePhases: 1
[cache ] TRACE 2021/01/04 18:37:58 timerSet: false
[cache ] TRACE 2021/01/04 18:37:58 climater: off
[cache ] TRACE 2021/01/04 18:37:58 remoteDisabled:
````

andig, 2021-01-31:
@goebelmeier siehe https://github.com/andig/evcc/issues/606#issuecomment-770388972 - evtl gelöst?

goebelmeier, 2021-01-31:
Kann ich bestätigen! Closed in #608

closed 2021-01-31 by goebelmeier