evcc-io/evcc #576: Neue Version verfügbar! Installiert: 0.39. Verfügbar

xerion3800, 2021-01-04:
Obwohl die 0.39 installiert ist (Docker File) verschwindet der Update-Header nicht:
/ # evcc -v
evcc version 0.39 (39b6228)

[image]

andig, 2021-01-04:
/cc @naltatis hast Du eine Idee?

andig, 2021-01-05:
@xerion3800 könntest Du mal bitte den Quelltext der Webseite anzeigen? Ganz oben im "body" müsste ein kleines Javascript sein, das würde ich gerne sehen.

rivengh, 2021-01-05:
Gleiches Problem taucht bei mir auch auf. Hier der Quelltext mit dem Javascript:

[image]

xerion3800, 2021-01-05:
@andig habe gerade die Seite erneut aufgerufen. Es wird oben ganz kurz angezeigt und dann wird es ausgeblendet. Nach einem Refresh kommt e kurz wieder und geht sofort wieder weg.

<script>window.evcc = {
        version: "0.39",
        configured: "1",
      };</script>

andig, 2021-01-05:
Sollte mit https://github.com/andig/evcc/commit/a78f92e7cc69acd337d75bd2c1202bee538faef4 behoben sein.

closed 2021-01-05 by andig

xerion3800, 2021-01-06:
Ist mit a78f92e behoben