evcc-io/evcc #577: Bei V0.39 fehlen die Modi Min+PV & Nur PV

userjp123, 2021-01-05:
**Describe the bug**
Bei V0.39 fehlen in der UI die Modi *Min+PV* & *Nur PV* 

**To Reproduce**
evcc.yaml:
site:
  title: Home # display name for UI
  meters:
    grid: grid # grid meter

**Expected behavior**
Modi Min+PV & Nur PV in UI sichtbar

**EVCC details:**
Show output of <code>evcc -v</code>:
```
V0.39

```
Ich übermittle die erzeugte PV-Leistung bewusst **nicht** an evcc, denn die ist ja zur PV-Regelung nicht nötig.
Deshalb fehlt bei mir im Config-File der Eintrag  eines PV-Meters in  **site**.
**Ohne** diesen Eintrag kommt es zum beschriebenen Bug.
Bei Version 0.38 funktionierte es noch.

andig, 2021-01-05:
Mhhm. Die Änderung war tatsächlich absichtlich, aber ich sehe das Problem. Denke drüber nach und melde mich wieder- kannst Du so lange auf 0.38 bleiben?

userjp123, 2021-01-06:
> Mhhm. Die Änderung war tatsächlich absichtlich, aber ich sehe das Problem. Denke drüber nach und melde mich wieder- kannst Du so lange auf 0.38 bleiben?

Kein Problem, hat Zeit bis zur nächsten Version. Danke !

andig, 2021-04-30:
@userjp123 könntest Du den PR testen? Sorry, dass es soooo lange gedauert hat...

userjp123, 2021-04-30:
...sieht gut aus. 
Habe jetzt nicht alles getestet, aber Buttons sind auch ohne PV-Meter da.

Vielen Dank !

closed 2021-05-01 by andig