evcc-io/evcc #584: Broken Carwings access breaks execution

maku1604, 2021-01-10:
Currently carwings server access seems to be broken (Returns Error 404 on launch)
Unfortunately this stops evcc from working and issue a restart loop of the container

Steps to reproduce the behavior:
Launch evcc with carwings vehicle configuration

Testenvironment:
Docker container (tag latest)
* SMA Sunny Boy PV
* SMA Energy Manager 1.0 
* Keba P30 c-series 
* 2018 Nissan Leaf ZE1 40kwh

Docker log: 
`[main  ] INFO 2021/01/10 15:54:00 evcc 0.39 (b78e30a)
[main  ] INFO 2021/01/10 15:54:00 using config file /etc/evcc.yaml
[main  ] INFO 2021/01/10 15:54:00 listening at 0.0.0.0:7070
[mqtt  ] INFO 2021/01/10 15:54:00 connecting evcc-834113411 at localhost:1883
[mqtt  ] DEBUG 2021/01/10 15:54:00 localhost:1883 connected
[main  ] FATAL 2021/01/10 15:54:05 cannot create vehicle 'Leaf': cannot create type 'carwings': received status code 404 (INVALID PARAMS)
`
evcc.dist.yaml:
`uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: shm
  type: sma
  serial: xxxxxxx
- name: sunnyboy
  type: modbus
  uri: 192.168.2.37:502 # 
  id: 126 # ModBus slave id
  model: sma-sunspec
  power: Power # default value, optionally override
  energy: Sum # energy value (Zählerstand)
# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: keba
  type: keba # charger
  uri: 192.168.2.19:7090 # ModBus address

hems: 
  type: sma

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
#- name: E-Auto
#  type: default
#  title: E-Auto # display name for UI
#  capacity: 200 # kWh
#  charge:  
#    type: script # use script plugin
#    cmd: /bin/sh -c "echo 70" # actual command
#    timeout: 3s # kill script after 3 seconds
#  cache: 5m # cache duration

- name: Leaf
  type: carwings
  title: Leaf
  capacity: 40 # kWh
  user: xxxxx
  password: xxxxx
  cache: 30m

# site describes the EVU connection, PV and home battery
site:
  title: Zuhause # display name for UI
  meters:
    grid: shm # grid meter
    pv: sunnyboy # pv meter
#    battery: battery # battery meter
  prioritySoC: 0 # give home battery priority up to this soc (0 to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Garage # display name for UI
  charger: keba # charger
  meters:
#    charge: keba # charge meter
#  vehicle: Leaf
  vehicles: # use if multiple vehicles allowed to charge on this loadpoint
#  - E-Auto
  - Leaf
  mode: pv
  soc:
    min: 50 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 80 # always charge to 100%
    estimate: true # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 70
    - 80
    - 85
    - 90
    - 100
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 80 # charge to 100%
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: -1380 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 20 # maximum charge current (default 16A)

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>`

Expected Behaviour:
1.Evcc can still control charging based on PV powerr an mode but shows error about unavailable SOC from carwings.
2. carwings works as before.

andig, 2021-01-10:
> I am using a generic vehicle config now to be able to charge, but it would be great if a broken car manufacturer backend would not stop evcc from working but fallback to some generic behaviour instead.

Understood. Currently, some devices are checked during creation, some at a later stage. At least for the vehicles, we probably shouldn't have any of these fail during startup?

andig, 2021-01-10:
Could you test the PR? Does not fix it for other cars though.

maku1604, 2021-01-10:
I will try to test the PR this evening. 
Regarding the general question about failing the Startup, i personaly would prefer a design where only issues with the host itself would fail the startup of the ui. All modules that need to be reached over network or that are added via configuration should only cause error messages and red crosses in the configuration page if evcc can still have basic functionality without those modules. E.g charging to SOC could still work with normal and min mode when grid meter values are not available. Having a ui to display the issues ist always more user friendly then just failing to launch, espacially for a dockerized web app. But I have no understanding of the effort needed to change it and i understand this might be much to much to ask for.

maku1604, 2021-01-10:
I have tested the PR and it is working fine. Evcc is launching and an error message is shown with the error response 
from carwings. Thank you! 
One litte thing I noticed was: I have a minimum charge level of 50% defined in the config and without the SOC available, evcc immediately started charging with max speed independent from configured default mode PV. 
I was a bit surprised by this, but I think evcc just uses 0% as SOC and behaves accordingly now.
I'm not sure if this is a good approach or if it should stick to the selected mode if no SOC is available.

andig, 2021-01-10:
> I have a minimum charge level of 50% defined in the config and without the SOC available, evcc immediately started charging with max speed independent from configured default mode PV.

Feel free to open new ticket. The problem is- if we dont do this- the car will be empty next morning. Either that or we charge it.

Closing here, happy to discuss in separate issue.

closed 2021-01-10 by andig

maku1604, 2021-01-10:
It's fine. The car is not detected as connected, therefore adding a generic vehicle with default soc 70 prevents the charging to min SOC if required.