evcc-io/evcc #586: Add API support for PSA vehicles

Alex-ander-s, 2021-01-11:
**API documentation**
https://developer.groupe-psa.io/webapi/b2c/overview/about/

**API specifications**
https://developer.groupe-psa.io/webapi/b2c/api-reference/specification/

**Scope of this API**
Supported vehicles by this API (100% electric):
- DS 3 Crossback e-tense
- Opel Corsa-e
- Peugeot e-2008
- Peugeot e-208

Supported vehicles by this API (Plug-in Hybrid):
- DS 7 Crossback e-tense
- Peugot 3008 Hybrid & Hybrid4
- Peugot 508 Hybrid
- Peugot 508 SW Hybrid
- Opel Grandland X Hybrid4
- Citroén C5 Aircross Hybrid4

**Goal**
- PSA vehicles are configurable within evcc.yaml
- PSA Vehicle SOC information could be fetched from PSA API and is usable within evcc

andig, 2021-01-11:
It would be nice to add this. However, I need live credentials for testing in order to implement this. Please get in contact @cpuidle@gmx.de if you're interested in helping.

andig, 2021-02-21:
According to https://github.com/ioBroker/AdapterRequests/issues/194 there seems to be support via bluetooth-connected phones only which is not suitable. Closing as wontfix.

andig, 2021-02-22:
Need to reopen. There is https://developer.groupe-psa.com/inc/node/1896 which looks good. Would still need credentials for testing.

/cc @Alex-ander-s

alexan-d, 2021-02-22:
Hallo andig,
ich könnte Unterstützung anbieten, habe eine Mail gesendet.
Grüße, Alexander

andig, 2021-02-23:
Ist leider nix angekommen. Benötigt werden User/Passwort sowie ClientID/Secret, siehe https://github.com/flobz/psa_car_controller.

andig, 2021-02-23:
Ping @alexan-d willst Du es nochmal versuchen (cpuidle@gmx.de)?

closed 2021-02-28 by andig

mamaretti, 2021-08-09:
Hi andig, thx for your project. EVCC working great.

can you give some informations or template how to add a Peugeot208 or  another PSA car?

I'm using this on iobroker which working great just need username and password.
there is no server "flobz/psa_car_controller" needed 
https://github.com/TA2k/ioBroker.psa

thx

andig, 2021-08-09:
You can check https://github.com/andig/evcc/blob/master/vehicle/psa.go#L50 For the required structure

andig, 2021-08-09:
Ich iobroker works without clientid/secret please open a new issue to investigate

mamaretti, 2021-08-09:
ok, will do

PR #692 (2021-02-28): Add Peugeot/Opel/Citroen api
Fix #586