evcc-io/evcc #590: Add Fiat (FCA) Support for 500e

mfueser, 2021-01-13:
Currently there is no open Implentation of the APIs used for the Fiat 500e. 
There is an App for Android and iOS whichs uses APIs to query SOC and additional vehicle information.

Goal

    FCA vehicles respective Fiat 500e are configurable within evcc.yaml
    FCA Vehicle SOC information can be fetched from FCA APIs and is usable within evcc

mfueser, 2021-01-13:
I have working credentials and will try to reverse engineer some of the API calls and can volunteer on implementation testing with my vehicle.

andig, 2021-01-13:
Sounds good! Please contact me on Slack if you have something to get startted.

closed 2021-03-11 by stale[bot]

andig, 2021-07-13:
@mfueser any success on reverse engineering? Did you see https://github.com/andig/evcc/discussions/1090?

mfueser, 2021-07-13:
Hi, not yet unfortunately. Have to check that using Android oder the Webapp mentioned in #1090.

andig, 2021-07-13:
Seems there's also https://myuconnect.fiat.com which has the relevant data and might be easier to debug.