evcc-io/evcc #591: Charger Support Heidelberg Wallbox Energy Control

leinich, 2021-01-15:
**Is your feature request related to a problem? Please describe.**
I ordered an Heidelberg Wallbox and I would like to see support for this in evcc

**Describe the solution you'd like**
The Wallbox has an RS485 Modbus interface as descibred here, which would allow to add support:
https://cdn.shopify.com/s/files/1/0101/2409/9669/files/heidelberg-energy-control-modbus.pdf?v=1603662569

andig, 2021-01-15:
Sorry for not having a better replay to a good issue: at this time support for additional chargers is not a priority.

I cannot currently spend the time for implementing and supporting an additional charger. If the vendor wants to fund this on a regular basis he is welcome.

I'd also accept PRs if you would volunteer to take over providing the required support.

andig, 2021-01-15:
That being said, it may be possible to use the box utilising the [EVCC plugins](https://github.com/andig/evcc#plugins). See https://github.com/andig/evcc-config#i-charge-cion-modbus for an example.

kahuwi14, 2021-01-17:
Dear Andi
I landed here because of the article on the pv-magazine.
I'm very thankfull for your great work! Otherwise, I would have to develop a software to controle my (hopefully soon arriving) Heidelberg-Wallbox. I tested the OpenWB-Software a bit, but it isn't as helpfull as I expected. With your software, I'll be able to controle my wallbox and expand my selfmade smarthome. With this said, I'd like to help the developing of this plugin as soon as my device arrives. I already startet with some testing with some of the plugins and also thought, that the plugin you mentioned, could be the right starting point.
When I'm able to test my work, I'll post my results in this thread.
Grüße aus der Eifel!

leinich, 2021-01-18:
@kahuwi14 
May I ask how you plan to use the Modbus RTU on Heidelberg with evcc.
I only havon one Network cable in my Garage so far. 
So I'm currently looking into following options:
1) Use the network cable for Modbus with an USB adapter CH340C ~ 5€
2) Use a Modbus to Ethernet Adapter ~ 25€

premultiply, 2021-01-18:
Falls noch jemand braucht: Ich habe noch je ein Protoss PW-11-H (WiFi, interne und externe Antenne) und Protoss PE-11-H (Ethernet) übrig. 1TE, integriertes Netzteil.
Produktseite des Herstellers: http://www.hi-flying.com/pw11 bzw. http://www.hi-flying.com/pe11

premultiply, 2021-01-18:
Ansonsten dürfte der Support für die Box relativ einfach anhander vorhandenen Implementierungen zu ergänzen sein:
Es gibt nur zwei Modbus-Register von Relevanz:
30006 (R) für den Status
40257 (R/W) für den Ladestrom und An/Aus (>0 bzw. =0)

premultiply, 2021-01-18:
Als Vorlage könnte die Implementierung für die SimpleEVSE dienen, die sehr ähnlich funktioniert:
https://github.com/andig/evcc/blob/master/charger/simpleevse.go

premultiply, 2021-01-18:
@leinich @kahuwi14 testet das mal bitte und meldet euch am besten im Slack-Channel.

kahuwi14, 2021-02-12:
So, meine Box kommt am Montag und wenn alles klappt, ist sie nächstes Wochenende angeschlossen und ich kann mit dem Testen beginnen! 👌

andig, 2021-04-21:
ping @kahuwi14 was ist denn beim Test rausgekommen?

steff393, 2021-04-24:
Bin gerade auf EVCC gestoßen, als interessante Alternative zu openWB. Ich hab die Heidelberg jetzt im WLAN (https://github.com/steff393/wbec) und überlege aber noch, was ich bzgl. WebInterface mache. Ich schau mir die Tage mal an, wie das mit EVCC funktioniert.

therobbot, 2021-05-17:
What's the current status on support for the Heidelberg WB? I'd also be very interested in it.

steff393, 2021-05-17:
wbec implements the relevant parts of the go-echarger local API (which is supported by EVCC). I've a local installation of EVCC here and the debug commands from EVCC are returning that everything is fine. Also basic checks work as expected. 
However, I don't have a real car. But since it works with openWB, it should also work with EVCC.

andig, 2021-05-18:
> What's the current status on support for the Heidelberg WB? I'd also be very interested in it.

Unchanged. This thread offers two options:
- use evcc plugins (e.g. mqtt)
- use https://github.com/steff393/wbec (Go-E api)

andig, 2021-08-07:
Refs https://github.com/andig/evcc/discussions/1257

closed 2021-08-08 by andig