evcc-io/evcc #597: Logfile um sicherheitsrelevante Daten bereinigen

Tscherno, 2021-01-20:
Ich bin eben in https://github.com/andig/evcc/issues/488 über ein Logfile mit Bearer-Token gestolpert. Das ist sicherheitstechnisch sensibel. Das Logfile sollte Bearer-Token wie Passwörter berhandeln und nicht loggen.

andig, 2021-01-20:
Es ist halt ein Trace der HTTP Kommunikation. `http: error` löst das Problem. An der Stelle gibts keine Information, was für ein Aufruf das ist. Zum Glück sind Bearer Tokens shortlived...

Tscherno, 2021-01-20:
Das schon, aber es stehen trotzdem diverse persönliche Daten drin, je nach Implementierung...

andig, 2021-01-20:
Ich sehe keinen sinnvollen Lösungsweg. Das Logfile weiss einfach nicht was für Daten das sind. Was schlägst Du vor?

tmeidinger, 2021-01-25:
Steht im Logfile irgendwas, was nicht in der Konfig steht bzw. daraus abgeleitet werden kann?
Hat jemand auf's Logfile Zugriff und nicht auf die Konfig?

andig, 2021-01-25:
Ich glaube es geht eher um das posten von Informationen. Ich sehe da keine gute Lösung. Im neuen Config UI werden die Felder ausgegraut sein. Aus dieser Sicht erstmal wontfix. Gerne wieder aufmachen wenn jemand eine gute Idee hat.

closed 2021-01-25 by andig