evcc-io/evcc #599: Windows Defender meldet Bedrohung durch evcc.exe (V0.41)

JBSullivan, 2021-01-21:
Ich denke Kopfzeile und Screenshot stehen für sich selbst.

[image]

andig, 2021-01-21:
Ich denke nicht. Das Programm wird unter Github durch Go aus git compiliert. Ich denke das ist ein Fehlalarm. Hast Du mal geschaut, ob Deine lokale evcc.exe der aus dem Release entspricht? Falls nein: lokale Infektion?!

andig, 2021-01-21:
Ich hab aber gar kein Problem damit, die exe und Windows Builds zu entfernen anstatt damit Zeit zu vergeuden. Was würdest Du vorschlagen?

Cruiserlive, 2021-01-21:
Servus zusammen,

bin heute neu dazu gekommen :-). Finde das Projekt einfach nur Klasse! Respekt .

Hatte eine ähnliche Viruswarnung, es war allerdings eher bzgl der verdächtigen Ausgangs Kommunikation ohne speziellen Virus. Kann es mir aber gerne morgen Abend nochmal anschauen wie die Meldung im Viruslog genau beschrieben war . 
Schönen Abend
Chris alias cruiser

JBSullivan, 2021-01-22:
Was soll ich vorschlagen? Ich bin Endanwender und kein Programmierer. Ich weiß ja gar nicht was man überhaupt anders machen könnte. Die Meldung poppte gestern direkt nach der Installation des neuesten Windows Defender Update auf. Checksumme passt. Heute Morgen habe ich noch einmal einen Scan laufen lassen und heute gibt es keine "Bedrohung".

Die Datei ist auch nicht in Quarantäne oder so. evcc läuft aktuell auch ganz normal.

[image]

goebelmeier, 2021-01-22:
Also zum Einen hat der Defender nur einen Generic Alarm geworfen:
> Behavior:Win32/DefenseEvasion.I!ml - This generic detection for suspicious behaviors is designed to catch potentially malicious files. If you downloaded a file or received it through email, ensure that it is from a reliable source before opening it.

https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?Name=Behavior:Win32/DefenseEvasion.I!ml&ThreatID=2147738312

Mittlerweile wirft es den Alarm auch nicht mehr und auch VirusTotal findet nichts mehr: https://www.virustotal.com/gui/file/a7ad13b60b511472b1c76bde96e61aa35b28dd42c08b43180293dbd1ac02f256/detection

In meinen Augen False Positive und kann geschlossen werden.

andig, 2021-01-22:
Super, danke!

> Was soll ich vorschlagen? 

Beim nächsten mal bitte erstmal prüfen, ob die lokale Datei *überhaupt* mit der bei GitHub identisch ist!

closed 2021-01-22 by andig