evcc-io/evcc #606: Push messages: Strange format of message variable ${connectedDuration} - e.g. 2562047h47m16.854775807s

thierolm, 2021-01-23:
Value of message variable ${connectedDuration} is printed in a strange format, e.g. 2562047h47m16.854775807s,
while ${chargeDuration} is printed in a nice way, e.g. 6h30m0s.

Steps to reproduce the behavior:
1. Configure push messages using evcc 0.41 (e.g. Telegram)
2. Track Telegram push messages when connecting and disconnecting your vehicle to the charger.

Expected behavior
Readable format corresponding to ${chargeDuration} print output. :-)

Telegram example
[image]

andig, 2021-01-23:
Hättest Du mal ein TRACE Log dazu? Ich vermute das passiert, wenn das Auto zum Start schon connected ist und er die Dauert seit zeit NULL ausrechnet?

thierolm, 2021-01-24:
Weiter unten die relevanten Teile aus dem Trace-Log. 
Es geschieht auch, wenn kein Auto connected bei jedem restart von evcc. Bei mir läuft evcc auf Ubuntu Server 20.04. Die systemd services werden jede Nacht gegen 03:30h restarted ... .
Telegram hat mir jetzt jede Nacht eine Disconnect Meldung geschickt :-) Ich denke mal, weil ${connectedDuration} initial auf dem schrägen Wert und nicht auf 0 sitzt.

[image]

Jetzt der Trace-Log Auszug:
```
thierolm@tserver:~$ sudo evcc --log trace
[main  ] INFO 2021/01/24 10:40:21 evcc 0.41 (c368f7d)
[main  ] INFO 2021/01/24 10:40:21 using config file /etc/evcc.yaml
[main  ] INFO 2021/01/24 10:40:21 listening at 0.0.0.0:7070
...
[lp-1  ] DEBUG 2021/01/24 10:40:25 charger status: A
[lp-1  ] INFO 2021/01/24 10:40:25 car disconnected
[lp-1  ] INFO 2021/01/24 10:40:25 set charge mode: minpv
[lp-1  ] INFO 2021/01/24 10:40:25 set target soc: 60
[cache ] TRACE 2021/01/24 10:40:25 chargedEnergy: 0
[cache ] TRACE 2021/01/24 10:40:25 connectedDuration: 2562047h47m16.854775807s
[cache ] TRACE 2021/01/24 10:40:25 chargeCurrent: 0
[cache ] TRACE 2021/01/24 10:40:25 connected: false
[cache ] TRACE 2021/01/24 10:40:25 charging: false
[cache ] TRACE 2021/01/24 10:40:25 enabled: false
[push  ] TRACE 2021/01/24 10:40:25 telegram: sending to 351210227
...
```

andig, 2021-01-24:
Das ist mit PR oder ohne?

thierolm, 2021-01-24:
Hilf mir kurz, was meinst du mit "PR"? :-)

andig, 2021-01-24:
In der #608 ist ein Pull Request (=PR) mit der Fehlerbehebung. Kannst Du den compilieren/ testen? Falls nein würde ich ihn einbauen und du bekommst es mit dem nächsten Release zum Test.

thierolm, 2021-01-24:
Ah, ok. Ich kann leider noch nicht kompilieren und testen. Ich habe auf meinem Ubuntu Server 20.04 am vergangenen WE die Entwicklungsumgebung noch nichts ans Laufen bekommen ... (es haben während des makes irgendwelche GO module gefehlt ...).
Ich kann auf auf das nächste Release warten. 

Werde aber auch an meiner Entwicklungsumgebung arbeiten, wenn ich Zeit finde :-) (=> 3 Kids, + Frau, + Haus, + Job ...)

andig, 2021-01-24:
Oj ja... bzgl make: siehe Readme und aktuelles go. Ich merge es einfach mal und wenns wieder auftaucht bitte neues Ticket!

closed 2021-01-24 by andig

thierolm, 2021-01-31:
Format ist jetzt mit Version 0.42 lesbar. 👍 
Auch der nächtliche Ladestart beim Restart des systemd evcc services wird nicht mehr ausgelöst.
Super!

[image]

andig, 2021-01-31:
Ich frage nach, danke für den Tipp! Mir schein aber, wir sollten die Dauer wenigstens noch auf ganze Sekunden runden?

thierolm, 2021-01-31:
ja, gerundet auf ganze Sekunden wäre perfekt!

Noch ein schneller Hinweis:
Ich habe für die Fronius Solar API V1 meter config templates (pv + grid) zusammengestellt und einen PR erstellt
Schau mal bei Gelegenheit drauf (es hängen auch die Fronius pdf Dokus dran): 
[https://github.com/andig/evcc-config/pull/14](https://github.com/andig/evcc-config/pull/14)

andig, 2021-01-31:
> ja, gerundet auf ganze Sekunden wäre perfekt!

Siehe https://github.com/andig/evcc/commit/4a434767a4aa96c26778c230deddd8909ad031f5