evcc-io/evcc #607: Add FRITZ!DECT 200/210 support (e.g. to charge E-Motorcyles or E-Bikes)

thierolm, 2021-01-23:
Hallo Andi,
ich sehe in deinen GIT Projekten, dass du auch tiefe AVM Fritz-Kenntnisse besitzt.
Neben dem/der ZOE und dem Audi Q5 eTron habe ich auch ein E-Motorad ([Zero S](https://www.zeromotorcycles.com/de-de/model/zero-s)), die in Garage an einer [FRITZ!DECT 200](https://avm.de/produkte/fritzdect/fritzdect-200/) hängt, welche ich aktuell mittels [FHEM](https://fhem.de/), bei Überschreitung von einer PV Leistung von 1400 W an- bzw. wieder ausschalte. 

Wäre für die FRITZ!DECT 200/210 ein charger Modul verfügbar, welches über einen Threshold-Parameter konfiguriert werden könnte, könnte evcc die Beladung aller meiner _Vehicle_ orchestrieren. :-)

Über den "Universal Fritz Charger" könnten dann auch E-Bikes, E-Motorräder, E-Werkzeuge etc. mit eingebunden werden.

andig, 2021-01-23:
Steht auf der Todoliste, kommt aber nicht kurzfristig. Ideen zur Konfiguration wären willkommen (gerne auch bei openWB spicken): welche Annahmen treffen wir hier für Leistung und Laufzeit?

thierolm, 2021-01-24:
Anbei ein paar Ideen zum zukünfigen FritzDect Charger:

Grundsätzlich müssen wir dabei beachten, dass der Charger ja im "Vehicle" (bspw. meine Zero S) eingebaut ist, und wir mit der FritzDect nur Schalten und den Strom messen können.

Konfiguratiosparameter
(Die müssen m.E. dem charger oder dem Vehicle Modul zugeordnet werden 
- Gespickte Parameter 
-- EinschaltUrl `FritzDectCharger.ChargeStartUrl` (gem. Fritz API)
-- AusschaltUrl  `FritzDectCharger.ChargeStop)` (gem. Fritz API)
-- Einschaltschwelle `Vehicle.ChargeStartThreshold` - Würde im `vehicle` Abschnitt für meine Zero S auf 1400W sitzen (siehe Screenshot)
-- Einschaltverzögerung `FritzDectCharger.ChargeStartDelay` - Dauer (Minuten) für die die Einschaltschwelle vom PV Überschuss überschritten sein muss
-- Einschaltdauer `FritzDectCharger.ChargeDurationMin`
-- Ausschaltschwelle `FritzDectCharger.ChargeStopThreshold`
-- Ausschaltverzögerung `FritzDectCharger.ChargeStopDelay`
- Von mir ergänzte Parameter, da wir ja ein Charge-Modul simulieren wollen ...
-- Standbyschwelle `Vehicle.ChargeStandbyThreshold` - Würde im `vehicle` Abschnitt für meine Zero S auf 10W sitzen (siehe Screenshot) 
-- StandbyAusschaltdauer `Vehicle.StandbyDurationMin` - Erst nach Ablauf der hier angegebenen Minuten schaltet der FritzDect Charger die Steckdose aus.

Weitere Ideen:
- Nur ein Vehicle pro FritzDectCharger    
Ein FritzDect Charger kann m.E. nur fest mit einem `vehicle` verbunden werden, da durch den verbauten Loader die individuellen Lade-Parameter des `vehicles` mit dem charger zusammenspielen müssen.
- Simulation der Charger Connect / Disconnect Erkennung    
Das FritzDectCharger Modul könnte beim Start von evcc die Steckdose initial ausschalten und dann in einem regelmäßigen Intervall anschalten und prüfen, wie hoch der Stromverbrauch ist.    
Liegt er bei 0 ist das vehicle nicht angeschlossen (disconnected),     
ist er größer 0 und kleiner `Vehicle.ChargeStandbyThreshold` oder größer Vehicle.ChargeStartThreshold` ist das vehicle angeschlossen (connected).
- Laufzeit
Hier ist vielleicht eine Worst Case Betrachtung sinnvoll. Wir gehen immer von 100% `capacity` aus. 

Werde noch weiter drüber nachdenken und ggf. ergänzen.

Screenshot 1: FritzDect Steckdose an, Zero S angeschlossen und lädt, nach 8 Minuten schaltet der eingebaute Charger ab.
[image]

Screenshot 2: FritzDect Steckdose an, Zero S angeschlossen, Stecker nach 5 Minuten gezogen
[image]

andig, 2021-01-24:
Wozu Standby?

premultiply, 2021-01-25:
Link #231

andig, 2021-02-05:
Im Prinzip wäre die FritzDect ja ein ganz normaler Charger:
- Status() (ChargeStatus, error)
- Enabled() (bool, error)
- Enable(enable bool) error
- MaxCurrent(current int64) error

Einzig die Stromregelung fehlt. Selbst den Status (connected/charging) könntest Du simulieren.

Wäre vielleicht mal einen Versucht wert, das mit dem `default` Charger zu prototypen und zu schauen wie weit Du kommst. Steuerung kannst Du extern machen (mit `script` Plugin o.ä.).

Mein Vorschlag: einfach mal probieren wie weit das trägt und dann schauen wir wie wir die Lücken patchen können?

thierolm, 2021-02-05:
OK, ich versuche mich mal daran. 
Kämpfe noch etwas mit git in Kombination mit VS Code und meiner Ubuntu Kiste ... :-)
Finde das Thema super spannend auch dadurch tiefer in go einzusteigen. :-)

thierolm, 2021-02-14:
Hi Andi,
bin gerade am testen einer ersten Version des "fbdect" chargers.
Parallel dazu lege ich auch ein "nonapi" vehicle an, über welches ich meine Zero S konfiguriere, das aber für alle möglichen zu ladenden Fahrzeugen die über eine FrirtzDect Steckdose mit einem Ladegerät (eingebaut oder extern) verwendet werden kann.

Beide Module lassen sich mit make bauen. Ein Test ergibt folgendes Ergebnis:
a) charger/fbdect.go (Zero ist an der FritzDect 200 angeschlossen und im Standby
``thierolm@tserver:~/evcc/git/evcc$ ./evcc -c ./evcc_fbdect.yaml -l debug charger
[main  ] INFO 2021/02/14 14:11:08 evcc 0.43 (9120014)
[main  ] INFO 2021/02/14 14:11:08 using config file ./evcc_fbdect.yaml
Power:          3W
Current L1..L3: 0.0133A 0A 0A
Status:         B
Enabled:        true
Charged:        0.4kWh
Charge status:  B
``
b) vehicle/nonapi.go (SOC wird auf 50% gesetzt)
``
thierolm@tserver:~/evcc/git/evcc$ ./evcc -c ./evcc_fbdect.yaml -l debug vehicle
[main  ] INFO 2021/02/14 14:06:52 evcc 0.43 (9120014)
[main  ] INFO 2021/02/14 14:06:52 using config file ./evcc_fbdect.yaml
SoC:      unexpected status: 400
Capacity: 10kWh
State:    unexpected status: 400
``
Kannst du mir einen Tip geben, warum/wann der "unexpected status" kommt?

closed 2021-03-08 by andig