evcc-io/evcc #617: mit SOC abfrage und erneuten anstecken kein laden.

martinez81, 2021-01-26:
Wenn ich mein Auto mit Zielladen auf 80% SOC über den EVCC lade., das Auto danach benutze und es wieder anstecke, z.B. mit einen SOC von 30% wird nicht wieder geladen. Im EVCC erscheint die Meldung "Verbunden", es passiert aber nichts. Auch nach "Stop" und wieder auf "Sofort" wird nicht geladen. Erst wenn ich das Zielladen auf 100% stelle fängt EVCC an zu laden. Danach wird auch der SOC vom Auto wieder neu abgefragt und an die Stelle geschrieben. Danach kann man das Zielladen wieder auf 80% stellen.
Meiner Meinung wird nicht nach erneuten verbinden der SOC vom Auto abgefragt und entsprechend entschieden ob geladen werden muss.
[image]

andig, 2021-01-26:
> Meiner Meinung wird nicht nach erneuten verbinden der SOC vom Auto abgefragt und entsprechend entschieden ob geladen werden muss.

Das sollte so sein. Bitte vollständiges Issue mit Config und Logfile.

martinez81, 2021-01-26:
Mache ich gern.
[evcc.yaml.zip](https://github.com/andig/evcc/files/5873785/evcc.yaml.zip)

Der SOC vom Nissan LEAF wurde auch nicht aktualisiert. Da kam immer eine Fehlermeldung.
[image]

Beim Versuch ein Logfile zu erzeugen habe ich den EVCC beendet und mit "sudo systemctl start evcc.service" wieder gestartet. 
Danach wurde auch der SOC vom LEAF wieder eingelesen.

Kannst du mir bitte sagen wo ich ein Logfile vorhandenes Logfile finde. Oder wie ich eins anlegen lassen kann?
In der Config habe ich das immer an:

`# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug`

andig, 2021-01-26:
Also verstehe ichs richtig:

- target soc 80%
- auto abziehen
- auto mit 30% wieder anstecken
- laden startet nicht?

Wo im Logfile finde ich die entsprechenden Events (Zeile/Zeit)? Was hat SoC kann nicht gelesen werden damit zu tun? 

Richtig wäre: falls der SoC nicht gelesen werden kann, immer zu laden. Ziel: niemand steht mit leerem Auto da.

martinez81, 2021-01-26:
> Also verstehe ichs richtig:
> 
> * target soc 80%
> * auto abziehen
> * auto mit 30% wieder anstecken
> * laden startet nicht?
Ganz genau, so ist es.

 
> Wo ist das entsprechende Logfile? Wo im Logfile finde ich die entsprechenden Events (Zeile/Zeit)? Was hat SoC kann nicht gelesen werden damit zu tun?
> 
Ich verstehe nicht was du mir damit sagen willst. Ich wollte wissen, ob irgendwo auf dem Raspberry das Logfile geschrieben wird. Wenn ich das wüsste könnte ich das Logfile hier posten.

> Richtig wäre: falls der SoC nicht gelesen werden kann, immer zu laden. Ziel: niemand steht mit leerem Auto da.

Genau, das wäre schon mal der richtige Weg, für den Fall dass kein SOC neu eingelesen werden kann.
Ich weiß ja, dass die SOC abfrage beim Leaf ziemlich "kacke" ist, aber beim Model 3 funktioniert das sehr gut. Auch das Model 3 hat kein SOC angezeigt. Es stand nur "verbunden" da.
Nachdem ich EVCC neu gestartet habe hat wieder alles funktioniert.
Aktuell sieht man das, bei SOC werden angezeigt.
[image]

Zur Erklärung warum das Model 3 nicht lädt. Beim Model 3 habe ich intern ein Ladeziel von 80% eingestellt. Deshalb lädt er jetzt nicht weiter, EVCC Ladeziel steht auf 100%
WARUM?
Bei diesen Temperaturen wird der nutzbare SOC im Model 3 stückweise immer mal wieder um 1% reduziert (Akku Schutz). Wenn das passiert wird EVCC das 1% immer wieder nachladen. Ich möchte das aber nicht. Das Auto würde alle paar Stunden zum Laden geweckt und ein paar Minuten nachgeladen. Die Kapazität ist ja nicht weg, sondern wird nur nicht mit so kalten Akku genutzt.
Man könnte das ändern in dem man nicht den nutzbaren SOC an EVCC übergibt sondern den tatsächlichen SOC. Ob das besser oder schlechter ist könnte man diskutieren.

andig, 2021-01-26:
Mir fehlt einfach ein aussagekräftiges Logfile um das nachvollziehen zu können.

martinez81, 2021-01-26:
> Mir fehlt einfach ein aussagekräftiges Logfile um das nachvollziehen zu können.

Ja, das verstehe ich.
Kannst du mir bitte mitteilen wie ich ein Logfile finde oder ein neues erzeuge. Ich kann ja den Fehler bestimmt reproduzieren.

andig, 2021-01-26:
Log wird einfach auf Konsole geschrieben, das sollte egtl zu sehen sein. Wie startest Du es denn?

martinez81, 2021-01-26:
Ich starte bzw. lassen EVCC automatisch über einen systemctl auf einen Raspberry starten. Deshalb sehe ich da nichts.
Mit Absicht habe ich das Issuetemplate nicht ignoriert. Der Code in den Issuetemplate ist im Editor für mich mit den ganzen Zeichen nicht so leicht zu lesen. Ich sehe es mir jetzt nochmal genau an und versuche hier alles zu beantworten.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.41 (c368f7d)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
Unsupported Config Type ""

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: sma-grid
  type: sma
  serial: 300613xxxx
- name: sma-pv
  type: modbus
  model: sma-sunspec
  uri: 192.168.0.165:502
  id: 126

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: GO-eCharger1
  type: go-e # GO-eCharger
  uri: http://192.168.0.192
- name: GO-eCharger2
  type: go-e
  uri: http://192.168.0.193

# vehicle definitions

vehicles:
 - name: tesla
   type: tesla
   title: Model 3
   capacity: 79
   clientid: xxx
   clientsecret: xxx
   user: xxx # your email address used on Tesla website
   password: xxx # your password used on Tesla website password
   vin: xxx # the VIN of you car (can be obtained from Tesla website)
   cache: 5m

 - name: nissan
   type: nissan
   title: Leaf # display name for UI
   capacity: 40 # kWh
   user: xxx # user
   password: xxx # password
[... 464 lines cut ...]
[go-e  ] TRACE 2021/01/26 20:43:38 GET http://192.168.0.192/status
200 OK
{"version":"B","tme":"2601212043","rbc":"8","rbt":"294000185","car":"4","amp":"16","err":"0","ast":"0","alw":"0","stp":"0","cbl":"32","pha":"56","tmp":"8","tma":[5.13,4.38,4.50,5.50],"amt":"32","dws":"7568383","dwo":"0","adi":"1","uby":"0","eto":"330","wst":"3","txi":"0","nrg":[221,218,220,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"040.0","sse":"026130","wss":"Runway","wke":"*********","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"129","aho":"3","afi":"7","azo":"0","ama":"32","al1":"6","al2":"8","al3":"10","al4":"12","al5":"16","cid":"255","cch":"16711735","cfi":"65280","lse":"1","ust":"0","wak":"071281.me","r1x":"0","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"0","ecr":"0","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"047A6152","rcr":"","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"","rnm":"","rne":"","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":17,"mce":0,"mcs":"test.mosquitto.org","mcp":1883,"mcu":"","mck":"","mcc":0}
[cache ] TRACE 2021/01/26 20:43:38 chargedEnergy: 21023.286111111112
[cache ] TRACE 2021/01/26 20:43:38 chargeDuration: 20s
[go-e  ] TRACE 2021/01/26 20:43:38 GET http://192.168.0.192/status
200 OK
{"version":"B","tme":"2601212043","rbc":"8","rbt":"294000218","car":"4","amp":"16","err":"0","ast":"0","alw":"0","stp":"0","cbl":"32","pha":"56","tmp":"8","tma":[5.13,4.38,4.50,5.50],"amt":"32","dws":"7568383","dwo":"0","adi":"1","uby":"0","eto":"330","wst":"3","txi":"0","nrg":[221,218,220,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"040.0","sse":"026130","wss":"Runway","wke":"*********","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"129","aho":"3","afi":"7","azo":"0","ama":"32","al1":"6","al2":"8","al3":"10","al4":"12","al5":"16","cid":"255","cch":"16711735","cfi":"65280","lse":"1","ust":"0","wak":"071281.me","r1x":"0","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"0","ecr":"0","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"047A6152","rcr":"","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"","rnm":"","rne":"","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":17,"mce":0,"mcs":"test.mosquitto.org","mcp":1883,"mcu":"","mck":"","mcc":0}
[lp-1  ] DEBUG 2021/01/26 20:43:38 charger status: B
[lp-1  ] INFO 2021/01/26 20:43:38 stop charging <-
[cache ] TRACE 2021/01/26 20:43:38 chargeCurrent: 0
[cache ] TRACE 2021/01/26 20:43:38 connected: true
[cache ] TRACE 2021/01/26 20:43:38 charging: false
[cache ] TRACE 2021/01/26 20:43:38 enabled: false
[go-e  ] TRACE 2021/01/26 20:43:38 GET http://192.168.0.192/status
200 OK
{"version":"B","tme":"2601212043","rbc":"8","rbt":"294000248","car":"4","amp":"16","err":"0","ast":"0","alw":"0","stp":"0","cbl":"32","pha":"56","tmp":"8","tma":[5.13,4.38,4.50,5.50],"amt":"32","dws":"7568383","dwo":"0","adi":"1","uby":"0","eto":"330","wst":"3","txi":"0","nrg":[221,218,220,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"040.0","sse":"026130","wss":"Runway","wke":"*********","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"129","aho":"3","afi":"7","azo":"0","ama":"32","al1":"6","al2":"8","al3":"10","al4":"12","al5":"16","cid":"255","cch":"16711735","cfi":"65280","lse":"1","ust":"0","wak":"071281.me","r1x":"0","dto":"0","nmo":"0","sch":"AAAAAAAA [... cut ...] rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"","rnm":"","rne":"","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":17,"mce":0,"mcs":"test.mosquitto.org","mcp":1883,"mcu":"","mck":"","mcc":0}
[lp-1  ] DEBUG 2021/01/26 20:43:38 targetSoC reached: 82.0 > 80
[cache ] TRACE 2021/01/26 20:43:38 chargeCurrents: [0 0 0]
[cache ] TRACE 2021/01/26 20:43:38 remoteDisabled: 

</pre></details>

andig, 2021-01-26:
Wo im Logfile (=wann) finde ich jetzt das beschriebene Verhalten? Es tut mir wirklich leid, aber mir fehlt einfach die Zeit, Seitenweise Logfiles ohne Kontext zu lesen :(

andig, 2021-01-26:
Schau mal hier: https://github.com/andig/evcc/blob/master/core/soc/soctimer.go. Für den PV oder Awattar Case brauchts sowas Ähnliches.

martinez81, 2021-01-26:
Ah. Nein, das findest du da nicht.
Mir war nicht bewusst, dass du das hier brauchst tut mir leid.

Ich weiß jetzt was ich tun muss und werde morgen, wenn ich mein Auto wieder anstecke tracen.

andig, 2021-01-26:
> Wo ist das entsprechende Logfile? Wo im Logfile finde ich die entsprechenden Events (Zeile/Zeit)? Was hat SoC kann nicht gelesen werden damit zu tun?

Ok, dann neuer Versuch und Logfile. Wenn ich sehe, was passiert kann ich auch rausfinden warum ;)

goebelmeier, 2021-01-27:
@martinez81 wenn du über SystemD startest, kannst du das Log nachträglich sehen über `journalctl -e -u evcc.service`

junibart, 2021-01-27:
Ich kann aus den Beiträgen oben nicht lesen, welches Auto (Leaf oder Model3) von dem Fehler betroffen ist. Wenn es um den Leaf geht, dann habe ich einen Verdacht, mir ist gestern dasselbe passiert. Es ist offenbar immer noch so, dass bei der SoC-Abfrage des Leaf anfangs ein veralteter Wert  zurückgegeben wird, im geschilderten Fall kann es sein, dass bei der ersten Abfrage nach dem Anstecken der SoC-Wert vor der Abfahrt zurückgegeben wird und kein Ladebedarf erkannt wird. Martinez, das kannst Du einfach prüfen, indem Du während der Fahrt bzw. kurz vor dem Anstecken mit der Nissan-App oder MyLeaf mal den SoC checkst. Dann ist der aktuelle Wert im "Cache", und evcc bekommt bei der Abfrage einen realistischen Wert. 

Ich füge hier mal ein Trace-Log des gestrigen Ladeverhaltens an, auf die config.yaml habe ich leider erst heute abend wieder Zugriff. 

**Folgende Ereignisse sind wichtig:** 

Leaf aufgeladen am Mittag bis SoC 87% (nicht im Log enthalten). Abfahrt ca. 15:00

Neustart von evcc v0.41 um 15:56:39 (Zeile 1)

Rückkehr und Anstecken des Leaf um 18:02:29 (Zeile 7112). SoC laut Fahrzeugdisplay: 63%
Start der Aufladung mit 19A Ladestrom. 

Abfrage des SoC bei Nissan um 18:03:00. (Zeile 7162) Sofortige Rückmeldung eines SoC von 87%- das ist unplausibel, denn die Abfrage dauert 30 sek. bis eine Minute. Es ist der Wert vor der Abfahrt. 

Feststellung, dass "targetSoC reached: 87.0 > 80 ... charger disable" (Zeile 7176). Ladung wird beendet. 

Um 18:25:56 setze ich im Webinterface den SoC auf 90. (Zeile 8362). Die Ladung beginnt. 

Um 18:26:30 wird erneut der SoC bei Nissan abgefragt, er liefert den korrekten Wert 63% zurück. (Zeile 8453)

Um 18:57:18 setze ich im Webinterface den SoC auf 80 zurück. (Zeile 10608). 

Um 21:28:30 wird ein SoC von 80% erreicht und die Ladung beendet.  (Zeile 20767)

Es müsste in EVCC sichergestellt werden, dass der SoC-Wert wirklich aktuell ist, z.B. den ersten Wert verwerfen und erst den zweiten Wert nehmen. 

Ich hoffe, das hilft weiter. 

Beste Grüße,

Gunnar
[evcc.log](https://github.com/andig/evcc/files/5878501/evcc.log)

martinez81, 2021-01-27:
Moin Gunnar,
welches Auto bei mir den Fehler hatte, bzw. verursacht hat... Aus dem Bauch heraus würde ich sagen beide. Aber das werde ich genau testen.
Danke!

martinez81, 2021-01-27:
So, heute habe ich es hoffentlich zufriedenstellen und ergebnisorientiert untersucht und notiert.
Da ist 4x mal den EVCC im Testablauf neu gestartet habe hänge ich 4 Logs hier an.
Ich hoffe sehr, dass wir so das Problem finden.
Nur nur der Leaf startet nicht wieder sondern auch das Model 3. Ich gehe also davon aus, das es nicht mit der schlechten API des Nissan Leaf zu tun hat.
Im Testablauf hat aber auch zum Ende hin die SOC Abfrage des Leaf gehangen, was auch blöd ist. Aber es muss meiner Meinung nach, wenn keine neuen Daten vom SOC kommen geladen werden. Ein leerer Akku muss verhindert werden.

Jetzt zu meiner Auswertung. Erst sieht ihr den zeitlichen Ablauf danach kommen die Logfiles.

**Ablauf:**

13:18 Uhr EVCC Neustart mit Trace im Terminal ($ sudo evcc  --log trace)
13:20 Uhr Model 3 mit SOC 77% angeschlossen, Zielladen über EVCC auf 80% - laden beginnt
13:21 Uhr Leaf mit SOC 54% angeschlossen, Zielladen über EVCC auf 80% - laden beginnt
13:31 Uhr Model 3 erreicht SOC 80% EVCC beendet das Laden
14:08 Uhr Model 3 abgesteckt
16:50 Uhr Model 3 mit SOC 61% angeschlossen, Zielladen über EVCC auf 80% - laden beginnt NICHT
16:51 Uhr Leaf mit SOC 81% abgesteckt (kurze Testfahrt mit den Leaf um den SOC unter 80% zu bringen)
16:57 Uhr Leaf mit SOC 76% angeschlossen, Zielladen über EVCC auf 80% - laden beginnt NICHT
17:04 Uhr beide Autos laden nicht, sind aber angeschlossen
17:04 Uhr EVCC Neustart
17:04 Uhr Model 3 bekommt sofort den aktuellen SOC von 61%, EVCC startet das Laden
17:04 Uhr Leaf hat noch einen SOC von 81% (real aber 76%) im EVCC beginnt kurz zu laden und nach ca. 30 sek stoppt EVCC das Laden
17:16 Uhr Model 3 lädt, Leaf hat immer noch im EVCC einen SOC von 81% (real 76%) und wird deshalb nicht geladen.
17:17 Uhr EVCC Neustart
17:18 Uhr Model 3 lädt (SOC aktuell), Leaf hat immer noch im EVCC einen SOC von 81% (real 76%) und wird deshalb nicht geladen.
17:19 Uhr Leaf SOC wird über App abgerufen. App zeigt 81% an (real 76%), reload gesendet.
17:24 Uhr Leaf SOC wird immer noch nicht korrekt in der App angezeigt, App sagt SOC 81%, App mehrmals alle 5 Min zum SOC pushen angeregt.
18:11 Uhr Leaf App, endlich wird jetzt der aktuellen SOC 76% angezeigt.
18:13 Uhr EVCC Leaf SOC immer noch bei 81%
18:26 Uhr Model 3 SOC 80% erreicht EVCC beendet das Laden vom Model 3
18:32 Uhr Leaf abgesteckt und wieder neu angesteckt.
18:45 Uhr Leaf hat im EVCC immernoch das alten SOC von 81%, die Leaf App zeigt 76% an. Der Leaf wird nicht geladen.
18:47 Uhr EVCC stop.
18:47 Uhr EVCC start
18:48 Uhr Leaf bekommt sofort neuen aktuellen SOC von 76%, EVCC beginnt mit Laden
18:49 Uhr Test wird beendet. EVCC Stop.

Ich bekomme hier eine Fehlermeldung beim Einfügen von den ganzen Logfiles, deshalb im nächsten Post per zip File.
**"You can't comment at this time — your comment is too long (maximum is 65536 characters)."**

martinez81, 2021-01-27:
Logfiles:
[13.18 Uhr EVCC Start Terminal-Ausgabe.txt](https://github.com/andig/evcc/files/5882217/13.18.Uhr.EVCC.Start.Terminal-Ausgabe.txt)

[17.04 Uhr EVCC Neustart Terminal-Ausgabe.txt](https://github.com/andig/evcc/files/5882229/17.04.Uhr.EVCC.Neustart.Terminal-Ausgabe.txt)

[17.17 Uhr EVCC Neustart Terminal-Ausgabe.txt](https://github.com/andig/evcc/files/5882230/17.17.Uhr.EVCC.Neustart.Terminal-Ausgabe.txt)

[17.47 Uhr EVCC Neustart Terminal-Ausgabe.txt](https://github.com/andig/evcc/files/5882235/17.47.Uhr.EVCC.Neustart.Terminal-Ausgabe.txt)

[EVCC Terminal-Ausgabe.zip](https://github.com/andig/evcc/files/5882239/EVCC.Terminal-Ausgabe.zip)

andig, 2021-01-28:
Super. Erste Erkenntnis von hinten:

> 18:45 Uhr Leaf hat im EVCC immernoch das alten SOC von 81%, die Leaf App zeigt 76% an. Der Leaf wird nicht geladen.
> 16:50 Uhr Model 3 mit SOC 61% angeschlossen, Zielladen über EVCC auf 80% - laden beginnt NICHT

Da gibts anscheinend einen Fehler mit `mode: charging`, analysieren wir gerade in Slack. Klingt genau so. Aus irgendeinem Grund wird auch beim Reconnect der SoC nicht abgefragt obwohl er sollte.

> 17:19 Uhr Leaf SOC wird über App abgerufen. App zeigt 81% an (real 76%), reload gesendet.

Das müssen wir akzeptieren. Bessere Werte als die App gibts nicht. An der Stelle: keine Chance. Ok?

@martinez81 je nachdem wie fit Du bist: eine Remote Debug Session mit LiveShare uns VSCode wäre der Traum- dann könnte ich direkt bei Dir rein schauen. Falls Du Go hast und das compiliert bekommst?

martinez81, 2021-01-28:
> Da gibts anscheinend einen Fehler mit `mode: charging`, analysieren wir gerade in Slack. Klingt genau so. Aus irgendeinem Grund wird auch beim Reconnect der SoC nicht abgefragt obwohl er sollte. Mit `mode: connected` geht es vmtl, aber das willst Du egtl. ja nicht.
> 
Ja, das will ich vermeiden. Das hält das Auto wach, und kostet beim Model gute 300-400Wh.

> > 17:19 Uhr Leaf SOC wird über App abgerufen. App zeigt 81% an (real 76%), reload gesendet.
> 
> Das müssen wir akzeptieren. Bessere Werte als die App gibts nicht. An der Stelle: keine Chance. Ok?
> 
Richtig, das ist der schlechten API oder Nissan Server geschuldet. Gestern war es echt schlimm. Heute ging es ziemlich schnell, dass er den aktuellen Wert hatte.

> @martinez81 je nachdem wie fit Du bist: eine Remote Debug Session mit LiveShare uns VSCode wäre der Traum- dann könnte ich direkt bei Dir rein schauen. Falls Du Go hast und das compiliert bekommst?

Gerne kann ich helfen, wenn es das Projekt voran bringt.
Go und VSCode habe ich eben auf den Mac geladen.

Welchen zeitlichen Ablauf denkst du an? 40 Std Arbeitswoche, Familie mit kleinen Kinder ist vorhanden ;)
30 Min per Teamviewer ist kein Thema, wenn das aber ein paar Stunden bis spät in die Nacht gehen soll, wird das schwierig...

andig, 2021-01-28:
Bist du bei uns im Slack? Link siehe Wiki? LiveShare Plugin für vscode installiert? Debuggen mit Deiner normalen Config geht? Dann könntest mich mal über LiveShare einladen. Teamviewer müssten wir übers WE verabreden.

andig, 2021-01-28:
@junibart das Leafproblem müssen wir uns nochmal extra anschauen. Hier ist irgendwas anderes faul.

andig, 2021-01-29:
Ich glaube ich habs gefunden. Falls ihr so nicht testen könnt würde ich es mergen.

martinez81, 2021-01-29:
Super!
Wenn du mir ein fertiges File für linux_armv6 sendest, teste ich es gern!

andig, 2021-01-29:
ARMv6: [evcc.gz](https://github.com/andig/evcc/files/5892601/evcc.gz)

martinez81, 2021-01-29:
Ich habe das File eben drauf gespielt. Ich werden berichten.

andig, 2021-01-29:
...uuuuuuuund? Eine Positivmeldung hab ich schon. Falls Du nicht dazu kommst würde ich es sonst einfach durchschieben.

martinez81, 2021-01-29:
Ich bin seit 12 Uhr unterwegs.... 400km
Ich komme erst 19 Uhr wieder zuhause an.
Du kannst es auch durchschieben...

closed 2021-01-29 by andig

junibart, 2021-01-29:
Hi, 

ich kann auch etwas positives berichten. 
Poll mode "charging", mit der Armv6-Datei hier aus dem issue zwei Ladevorgänge gestartet: In beiden Fällen wird wie gewünscht der SoC abgefragt und anschließend geladen. (Nissan Leaf) 
An den Tagen zuvor hatte ich auch die Abfrageprobleme beim Anstecken, leider kein Log geschrieben. 

Ich kann aber auch bestätigen, dass es nach wie vor mit der Aktualität des SoC von Nissan hapert. Es scheint, als wären die Werte immer die von der vorherigen Abfrage. Das führt auch immer dazu, dass der Leaf bei Ziel-SoC 80% bis 82% real geladen wird. 
Hätte ein neues, aktuelles Tracelog von heute, das dieses Verhalten zeigt. 

Wollen wir dazu #488 nochmal aufmachen oder soll es ein neues Issue sein?

andig, 2021-01-30:
> Es scheint, als wären die Werte immer die von der vorherigen Abfrage. Das führt auch immer dazu, dass der Leaf bei Ziel-SoC 80% bis 82% real geladen wird.

Das wär ja erstmal nicht so schlimm. Mach bitte ein neues Ticket. Wir müssten rausfinden:
- wo das heute wirklich ein Problem ist (jenseits von Optik)
- wie der SoC verlässlich ausgelesen werden kann ohne dabei die Regelungslogik aus dem Trab zu bringen

PR #621 (2021-01-29): Fix soc not updated once when car connected
Fix #617