evcc-io/evcc #618: Estimate Time für Hyundai ergänzen

xerion3800, 2021-01-26:
Über das Bluelinky Projekt besteht die Möglichkeit die Erwartete Ladezeit auszulesen.
Zu finden ist es in folgenden Objekt:
- remainTime2
- atc

[image]

Das Value ist in Minuten angegeben.

goebelmeier, 2021-01-26:
In https://github.com/andig/evcc/blob/b85c8b3f05a631128a5015ca8607e3c6bebb366a/vehicle/bluelink/bluelink.go#L77 wird die Response sogar schon abgeholt, allerdings nicht weiterverarbeitet

andig, 2021-01-26:
Ist die immer vorhanden?

xerion3800, 2021-01-26:
Wenn nicht geladen wird kommt "0" zurück.

andig, 2021-01-26:
Test wäre gut, ich hab keine Zugangsdaten

xerion3800, 2021-01-26:
Ich  kann das gerne Testen wenn du das pusht

andig, 2021-01-26:
Mehr pushen geht nicht. Du meinst mergen?

xerion3800, 2021-01-26:
Du bist einfach zu schnell :-) habe gar nicht mitbekommen das du schon gepusht hattest

xerion3800, 2021-01-26:
Ich nutze Docker kann ich das irgendwie damit pullen?

andig, 2021-01-26:
Gar nicht, muss gebaut werden. Ich merge es einfach und Du meldest Dich falls es nicht funktioniert.

closed 2021-01-26 by andig

xerion3800, 2021-01-26:
Okay mache ich. Estimate muss dann in evcc.yaml auf true. Richtig?

goebelmeier, 2021-01-26:
Nur wenn du möchtest, dass evcc zwischen den API Requests noch probiert zu interpolieren. Wenn dir ein Update bei jedem API Request (cache: Configoption im Vehicle) reicht, dann kannst du es auf false lassen

xerion3800, 2021-01-26:
Habe Mal getestet ich könnte auf den ersten Blick nicht die Restzeit sehen.
[image]

xerion3800, 2021-01-26:
> Nur wenn du möchtest, dass evcc zwischen den API Requests noch probiert zu interpolieren. Wenn dir ein Update bei jedem API Request (cache: Configoption im Vehicle) reicht, dann kannst du es auf false lassen

In welchen Abstand wird dann gepollt?

goebelmeier, 2021-01-26:
```
vehicles:
- name: renault
  type: renault
  title: Zoe
  capacity: 60 # kWh
  user: # user
  password: # password
  vin: WREN...
  cache: 5m
```

Das wird in cache: definiert

xerion3800, 2021-01-26:
> ```
> vehicles:
> - name: renault
>   type: renault
>   title: Zoe
>   capacity: 60 # kWh
>   user: # user
>   password: # password
>   vin: WREN...
>   cache: 5m
> ```
> 
> Das wird in cache: definiert
Wir das nicht jetzt darüber definiert?

interval: 60m

Oder ist das mit für den ausgewählten Mode?

andig, 2021-01-26:
Was für ein interval? Bzgl api: wenn du —log trace machst siehst Du was vom Server geliefert wird.

goebelmeier, 2021-01-26:
> > ```
> > vehicles:
> > - name: renault
> >   type: renault
> >   title: Zoe
> >   capacity: 60 # kWh
> >   user: # user
> >   password: # password
> >   vin: WREN...
> >   cache: 5m
> > ```
> > 
> > 
> > Das wird in cache: definiert
> > Wir das nicht jetzt darüber definiert?
> 
> 
> interval: 60m
> 
> 
> Oder ist das mit für den ausgewählten Mode?

interval in der evcc.yaml definiert den Regelzyklus. So oft werden die Zähler ausgelesen und ein neuer Ladestrom definiert. Das sollte eher so bei 10-20 Sekunden liegen.

xerion3800, 2021-01-26:
@andig: Der Wert kommt auf jeden Fall im Log.:
**"atc":{"value":75,"unit":1}}**

Aber in der Web UI kommt die Zeit nicht oder muss ich dafür noch was in der evcc.yaml konfigurieren?

`{"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":93,"batteryPlugin":2,"remainTime2":{"etc1":{"value":55,"unit":1},"etc2":{"value":90,"unit":1},"etc3":{"value":55,"unit":1},"atc":{"value":75,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":270,"unit":1},"totalAvailableRange":{"value":270,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":9,"time":{"time":"1200","timeSection":0}},"end":{"day":9,"time":{"time":"1200","timeSection":0}}},"targetSOClist":[{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":290,"unit":1},"totalAvailableRange":{"value":290,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":290,"unit":1},"totalAvailableRange":{"value":290,"unit":1}},"type":2},"plugType":1}]}},"ign3":true,"hoodOpen":false,"transCond":true,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":83,"batState":0},"time":"20210126201754"},"msgId":"11d03105-e28b-4327-888c-1186858e4b7e"}`

xerion3800, 2021-01-26:
@goebelmeier : Ich glaube wir meinen unterschiedliche Dinge. Du meinst das Intervall oben in der evvc.yaml und ich meine das intervall in diesem CodeSchnipsel.
Es gab ja Probleme mit EV wobei die Hilfsbatterie bei ständigem Pollen entleert wurde. Darauf hin wurde der Poll Mode nagepasst und kann nun ausgewählt werden wann gepollt wird in diesem Zusammenhang wurde "cache: 5" durch intervall im Attribute poll ergänzt
```
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    estimate: false # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 30
    - 50
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge to 100%
  phases: 1 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

```