evcc-io/evcc #624: Tesla API doesn't work any more

naltatis, 2021-01-30:
**Describe the bug**
Tesla has deprecated the authentication endpoint that's indirectly used via evcc.

**To Reproduce**
Configure a vehicle with Tesla credentials and run `evcc vehicle`

**Expected behavior**
I recieve a 400 Bad Request. No vehicle information is shown.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
./evcc --log trace vehicle
[main  ] INFO 2021/01/30 10:50:17 evcc 0.0.1-alpha (HEAD)
[main  ] INFO 2021/01/30 10:50:17 using config file /Users/michael/lab/evcc/evcc.yaml
[main  ] FATAL 2021/01/30 10:50:18 cannot create vehicle 'tesla': cannot create type 'tesla': 400 Bad Request
```

Executing the auth request agains the Tesla api returns 400 Bad Request and this response:
```
{
    "response": "endpoint_deprecated:_please_update_your_app."
}
```

The deprecation has been known for some time but now it seems the be gone.
See: https://tesla-api.timdorr.com/api-basics/authentication

> ⚠ This information is out of date ⚠
> Tesla has deprecated the /oauth/token endpoint in favor of using auth.tesla.com. I'm working on updating the documentation as soon as possible. Feel free to discuss this in issue 215.

More details here: https://github.com/timdorr/tesla-api/issues/215#issuecomment-770081554

martinez81, 2021-01-30:
Der Bug zeigt sich bei mir auch.

andig, 2021-01-30:
Habt ihr MFA irgendwie manuell aktiviert? Anderenfalls scheint die Umstellung von Tesla zu kommen. Zum Testen bräuchte ich Zugangsdaten.

martinez81, 2021-01-30:
> Habt ihr MFA irgendwie manuell aktiviert? 
Was ist MFA?

andig, 2021-01-30:
2FA oder MFA = Multi Factor Authentication 

[image]

Ist die bei Dir an @martinez81?

andig, 2021-01-30:
Halbe Lösung. Mit gültigem Token gehts erstmal weiter. Das muss allerdings anderweitig erzeugt werden. Die Erzeugung könnten wir erstmal mittels kleinem Workaround ebenfalls mit einbauen, z.B.:

    evcc tesla-token

andig, 2021-01-30:
Ist drin. Damit kann das Token erzeugt werden (jedenfalls für 2FA) und dann manuell in die Config übernommen.

martinez81, 2021-01-30:
> 2FA oder MFA = Multi Factor Authentication
> 
> [image]
> 
> Ist die bei Dir an @martinez81?

> 2FA oder MFA = Multi Factor Authentication
> 
> [image]
> 
> Ist die bei Dir an @martinez81?

Nein, ist aus.

andig, 2021-01-31:
> Nein, ist aus.

@martinez81 das ist super. Einen Test *mit* MFA habe ich jetzt, fehlt noch einer *ohne* MFA. Falls Du Deine Logindaten teilen würdest würde mir das sehr helfen (cpuidle@gmx.de). Mangels Auto kann ich sonst nicht testen.

martinez81, 2021-01-31:
Mail ist raus..

naltatis, 2021-01-31:
`evcc tesla-token` funktioniert.
Ich bekomme dann ein `AccessToken`, ein `RefreshToken` und die Angabe `ExpiresIn:300`. Wenn ich das `AccessToken` in meiner Config als `token` hinterlege klappt die API Kommunikation. Allerdings nur für eine kurze Zeit (ich vermute 5min = 300s). Danach bekomme ich `401 Unauthorized`.
Muss das Token regelmäßig mit dem `RefreshToken` refreshed werden um auf die 45 Tage Laufzeit zu kommen?

andig, 2021-02-01:
> evcc tesla-token funktioniert.

Gerade aktualisiert: jetzt werden auch Tokens generiert falls keine MFA  aktiv ist. Prinzipiell hätten wir damit eine Lösung die wir einbauen könnten. Wer ohne MFA arbeitet bekommt damit ein Token.

Was jetzt noch fehlt ist der Token Refresh damit EVCC auch wieder in Ruhe ohne Neustarts laufen kann.

naltatis, 2021-02-01:
Ja, die Lösung funktioniert seit gestern problemlos bei mir. Jetzt wo die Debug Meldung raus ist gibts auch kein Verwirrungspotential mehr.

andig, 2021-02-03:
@all: gerne nochmal testen. Die Lösung ist jetzt rund und sollte auch Token Refresh unterstützen. Alternativ: ich schiebs durch...

naltatis, 2021-02-03:
Das generierte Token funktioniert mit der API. Ob's in ein paar Wochen auch korrekt refresht wird kann ich natürlich erst dann sagen :) Aber ich würde dann einfach ein neues Ticket aufmachen wenns dann hakt.

andig, 2021-02-03:
Dir fehlt noch der Refreshtoken und die Syntax ist anders. Erzeug Dir nochmal einen Neuen, dann siehst Du es.

closed 2021-02-03 by andig