evcc-io/evcc #629: Add Hyundai RemainTime in Web UI

xerion3800, 2021-01-31:
To have the importants Infomations on the Mainpage it would be helpful to have the Rangevalue for Hyundai Vehicles.
In the API Respons is the payloyd "payload.status.evStatus.drvDistance[0].rangeByFuel.totalAvailableRange" for this Informations.

[image]

andig, 2021-01-31:
Wir nutzen bisher nur den EvStatus. Hättest Du mal ein vollständiges JSON oder Zugangsdaten?

xerion3800, 2021-01-31:
Ich habe dir Mal ein Auszug aus unseren letzten logdaten von EVCC rauskopiert. Die Daten werden schon abgefragt.
`{"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"10H","unit":0,"hvacTempType":1},"defrost":true,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":91,"batteryPlugin":2,"remainTime2":{"etc1":{"value":55,"unit":1},"etc2":{"value":115,"unit":1},"etc3":{"value":70,"unit":1},"atc":{"value":75,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":263,"unit":1},"totalAvailableRange":{"value":263,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[5],"time":{"time":"0330","timeSection":1}},"reservChargeSet":true,"reservFatcSet":{"defrost":false,"airTemp":{"value":"10H","unit":0,"hvacTempType":0},"airCtrl":1,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"10H","unit":0,"hvacTempType":0},"airCtrl":1,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":9,"time":{"time":"1200","timeSection":0}},"end":{"day":9,"time":{"time":"1200","timeSection":0}}},"targetSOClist":[{"targetSOClevel":100,"plugType":0},{"targetSOClevel":100,"plugType":1}]}},"ign3":true,"hoodOpen":false,"transCond":true,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":87,"batState":0},"time":"20210129102524"},"msgId":"57ff88c1-b3b8-4aac-8703-6f0ff32b4b56"}`

Du findest die Daten hier: drvDistance":[{"rangeByFuel":{"evModeRange":{"value":263,"unit":1},"totalAvailableRange":{"value":263,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[5],"time":{"time":"0330","timeSection":1}}

Gruß Frank

andig, 2021-01-31:
Kannst Du's compilieren und testen?

xerion3800, 2021-01-31:
Habe ich mich ehrlich gesagt noch nicht mit beschäftigt, werde es mal versuchen ist für die Zukunft mit Sicherheit hilfreich wenn ich auch selber kann.

andig, 2021-01-31:
Das wäre Klasse! Siehe README und dann einfach `make build`.

xerion3800, 2021-02-01:
Sieht gut aus: 
[image]

closed 2021-02-01 by andig