evcc-io/evcc #630: Add "prioritySoC" for MQTT Write settings

xerion3800, 2021-01-31:
For "minSoC" is it allready possible to use MQTT Write settings. So it is possible to use the SmartHome to override the minSoc via MQTT. It would be helpful to have this feature as well for the "prioritySoC" to override the SoC of the Solar Battery.

andig, 2021-02-02:
I'm wondering- why do we really need this? I was assuming that `prioritySoC` is pretty much static and hence doesn't need an API?

xerion3800, 2021-02-02:
Maybe i only misunderstood  `prioritySoC`. When it is winter and the Solar Battery is empty and my `prioritySoC is 60 %` than i understood it, that EVCC will in Mode PV wait so long as the SoC of the Solar is 60%. In this case i will "never" load up my EV via PV  unless the prioritySoC ist more than 60%. Is that right?
Yes i could use as well the minSoc, but with minSoc EVVC use the `maxcurrent `and not what via PC currently available. What could be in Winter less.
In this case it would be nice the reduce the `prioritySoC `without change the `evcc.yaml`. So it would be possible to change it via Smart Home regarding the Solar Forecast the `prioritySoC`.

andig, 2021-02-02:
> Is that right?

Correct.

> In this case it would be nice the reduce the prioritySoC without change the evcc.yaml.

Understand.

Were you able to test the attached PR? For build instructions see README.

xerion3800, 2021-02-02:
I get via MQTT now the prioritySoC but when i use the set `evcc/site/prioritySoC/set` than i didn't got an new prioritySoC subscribe.

`[main  ] INFO 2021/02/02 15:22:03 evcc  ()
[main  ] INFO 2021/02/02 15:22:03 using config file /mnt/c/WLS/evcc.yaml
[main  ] INFO 2021/02/02 15:22:03 listening at 0.0.0.0:7070
[mqtt  ] INFO 2021/02/02 15:22:03 connecting evcc-35746931 at 192.168.178.75:1883
[mqtt  ] DEBUG 2021/02/02 15:22:03 192.168.178.75:1883 connected
[cache ] TRACE 2021/02/02 15:22:03 title: go-eCharger HOME+
[cache ] TRACE 2021/02/02 15:22:03 minCurrent: 6
[cache ] TRACE 2021/02/02 15:22:03 maxCurrent: 16
[cache ] TRACE 2021/02/02 15:22:03 phases: 1
[cache ] TRACE 2021/02/02 15:22:03 activePhases: 1
[cache ] TRACE 2021/02/02 15:22:03 hasVehicle: true
[cache ] TRACE 2021/02/02 15:22:03 mode: pv
[cache ] TRACE 2021/02/02 15:22:03 targetSoC: 100
[cache ] TRACE 2021/02/02 15:22:03 minSoC: 0
[cache ] TRACE 2021/02/02 15:22:03 socLevels: [30 50 80 100]
[cache ] TRACE 2021/02/02 15:22:03 socTitle: Kona
[cache ] TRACE 2021/02/02 15:22:03 socCapacity: 39
[site  ] INFO 2021/02/02 15:22:03 site config:
[site  ] INFO 2021/02/02 15:22:03   meters:    grid ✓ pv ✓ battery ✓
[cache ] TRACE 2021/02/02 15:22:03 title: SENEC.Home
[cache ] TRACE 2021/02/02 15:22:03 gridConfigured: true
[site  ] INFO 2021/02/02 15:22:03     grid:    power ✓ energy — currents —
[site  ] INFO 2021/02/02 15:22:03     pv:      power ✓ energy — currents —
[site  ] INFO 2021/02/02 15:22:03     battery: power ✓ energy — currents — soc ✓
[lp-1  ] INFO 2021/02/02 15:22:03 loadpoint 1:
[lp-1  ] INFO 2021/02/02 15:22:03   mode:      pv
[lp-1  ] INFO 2021/02/02 15:22:03   charger:   power ✓ energy — currents ✓ timer —
[lp-1  ] INFO 2021/02/02 15:22:03   meters:    charge ✓
[lp-1  ] INFO 2021/02/02 15:22:03     charge:  power ✓ energy — currents ✓
[lp-1  ] INFO 2021/02/02 15:22:03   vehicles:  ✓
[lp-1  ] INFO 2021/02/02 15:22:03     car 0:   finish ✓ status — climate —
[cache ] TRACE 2021/02/02 15:22:03 pvConfigured: true
[cache ] TRACE 2021/02/02 15:22:03 batteryConfigured: true
[cache ] TRACE 2021/02/02 15:22:03 prioritySoC: 60
[cache ] TRACE 2021/02/02 15:22:03 chargeConfigured: true
[site  ] DEBUG 2021/02/02 15:22:03 ----
[site  ] DEBUG 2021/02/02 15:22:09 pv power: 150W
[cache ] TRACE 2021/02/02 15:22:09 pvPower: 150
[site  ] DEBUG 2021/02/02 15:22:10 grid power: 462W
[site  ] DEBUG 2021/02/02 15:22:10 battery power: -0W
[cache ] TRACE 2021/02/02 15:22:10 gridPower: 462.24
[site  ] DEBUG 2021/02/02 15:22:10 battery soc: 100%
[site  ] DEBUG 2021/02/02 15:22:10 site power: 462W
[cache ] TRACE 2021/02/02 15:22:10 batteryPower: -0
[cache ] TRACE 2021/02/02 15:22:10 batterySoC: 100
[cache ] TRACE 2021/02/02 15:22:10 mode: pv
[lp-1  ] DEBUG 2021/02/02 15:22:10 charge power: 0W
[cache ] TRACE 2021/02/02 15:22:10 chargePower: 0
[cache ] TRACE 2021/02/02 15:22:10 chargeCurrent: 0
[cache ] TRACE 2021/02/02 15:22:10 chargedEnergy: 5247.669444444444
[cache ] TRACE 2021/02/02 15:22:10 chargeDuration: 0s
[lp-1  ] DEBUG 2021/02/02 15:22:10 charger status: B
[cache ] TRACE 2021/02/02 15:22:10 connectedDuration: 0s
[cache ] TRACE 2021/02/02 15:22:10 chargeCurrent: 0
[cache ] TRACE 2021/02/02 15:22:10 connected: true
[cache ] TRACE 2021/02/02 15:22:10 charging: false
[cache ] TRACE 2021/02/02 15:22:10 enabled: false
[lp-1  ] DEBUG 2021/02/02 15:22:15 vehicle soc: 88%
[cache ] TRACE 2021/02/02 15:22:15 socCharge: 88
[cache ] TRACE 2021/02/02 15:22:15 chargeEstimate: -1ns
[cache ] TRACE 2021/02/02 15:22:15 chargeRemainingEnergy: 5200
[lp-1  ] DEBUG 2021/02/02 15:22:15 max charge current: 0A = 0A + -2A (462W @ 1p)
[lp-1  ] DEBUG 2021/02/02 15:22:15 pv max charge current: 0A
[site  ] DEBUG 2021/02/02 15:22:15 ----
[site  ] DEBUG 2021/02/02 15:22:15 pv power: 150W
[cache ] TRACE 2021/02/02 15:22:15 chargeCurrents: [0 0 0]
[cache ] TRACE 2021/02/02 15:22:15 timerSet: false
[site  ] DEBUG 2021/02/02 15:22:15 grid power: 462W
[... 998 lines cut ...]
[site  ] DEBUG 2021/02/02 15:29:43 ----
[site  ] DEBUG 2021/02/02 15:29:43 pv power: 130W
[site  ] DEBUG 2021/02/02 15:29:43 grid power: 502W
[cache ] TRACE 2021/02/02 15:29:43 pvPower: 130
[site  ] DEBUG 2021/02/02 15:29:43 battery power: -0W
[site  ] DEBUG 2021/02/02 15:29:43 battery soc: 100%
[site  ] DEBUG 2021/02/02 15:29:43 site power: 502W
[cache ] TRACE 2021/02/02 15:29:43 gridPower: 502.19
[cache ] TRACE 2021/02/02 15:29:43 batteryPower: -0
[cache ] TRACE 2021/02/02 15:29:43 batterySoC: 100
[cache ] TRACE 2021/02/02 15:29:43 mode: pv
[lp-1  ] DEBUG 2021/02/02 15:29:43 charge power: 0W
[cache ] TRACE 2021/02/02 15:29:43 chargePower: 0
[cache ] TRACE 2021/02/02 15:29:43 chargeCurrent: 0
[cache ] TRACE 2021/02/02 15:29:43 chargedEnergy: 5247.669444444444
[cache ] TRACE 2021/02/02 15:29:43 chargeDuration: 0s
[lp-1  ] DEBUG 2021/02/02 15:29:44 charger status: B
[cache ] TRACE 2021/02/02 15:29:44 connected: true
[cache ] TRACE 2021/02/02 15:29:44 charging: false
[cache ] TRACE 2021/02/02 15:29:44 enabled: false
[lp-1  ] DEBUG 2021/02/02 15:29:44 max charge current: 0A = 0A + -2.2A (502W @ 1p)
[lp-1  ] DEBUG 2021/02/02 15:29:44 pv max charge current: 0A
[cache ] TRACE 2021/02/02 15:29:44 chargeCurrents: [0 0 0]
[cache ] TRACE 2021/02/02 15:29:44 timerSet: false
[cache ] TRACE 2021/02/02 15:29:44 remoteDisabled:`

xerion3800, 2021-02-02:
Yes perfect is working! Thank you verry much!

PR #633 (2021-02-02): Add prioritySoC api (mqtt only)
Fix #630 ~~and harmonize vehicle API naming~~
closed 2021-02-02 by andig