evcc-io/evcc #631: New battery animation in wrong direction with sonnenBatterie ECO 8

rivengh, 2021-01-31:
**Describe the bug**
The new battery charging/discharging animation is running in the opposite direction as expected with sonnenBatterie ECO 8:
* from left to right, when discharging.
* from right to left, when charging.

**Expected behavior**
Battery charging/discharging animation is running:
* from left to right, when charging
* from right to left, when discharging

**EVCC details:**
```
0.42
```

**Debug Log (Battery Discharging) **
```
[site  ] DEBUG 2021/01/31 19:15:30 ----
[site  ] DEBUG 2021/01/31 19:15:30 pv power: -0W
[site  ] DEBUG 2021/01/31 19:15:30 grid power: 49W
[site  ] DEBUG 2021/01/31 19:15:30 battery power: -312W
[site  ] DEBUG 2021/01/31 19:15:30 battery soc: 78%
[site  ] DEBUG 2021/01/31 19:15:30 site power: -263W
```

**evcc.yaml / battery meter**
```
- name: battery
  type: default
  power: # power reading
    type: mqtt # use mqtt plugin
    topic: iobroker/sonnen/0/status/pacTotal # mqtt topic
    timeout: 30s # don't use older values
    scale: -1 # reverse direction
```

andig, 2021-01-31:
Are you entirely sure the battery config is correct?

andig, 2021-01-31:
Wir haben das mittlerweile aus Slack diskutiert und sind gemeinsam der Meinung, dass die Darstellung korrekt ist. Evtl. war Deine Config bisher (und ggf. auch das Beispiel auf evcc-config) fehlerhaft?

rivengh, 2021-01-31:
Gut möglich. Ich hatte mich mit der Config nach dem Beispiel auf evcc-config gerichtet.
Sollte der battery power Wert dann also positiv beim Entladen, sein?
Dann wäre natürlich "scale: -1" hier nicht korrekt.

andig, 2021-02-01:
Wir betrachten die Batterie als "extern" zu den Verbrauchern: https://github.com/andig/evcc/blob/master/core/site.go#L297

Alles was "raus" fließt ist negativ. Damit ist:
- Batterie laden: negativ
- Batterie entladen: positiv

Falls Du so nett sein willst- wäre Klasse das Beispiel in evcc-soc zu korrigieren.

rivengh, 2021-02-01:
Vielen Dank. Pull request zur Korrektur des Beispiels ist raus.

Schaut jetzt bei mir so im Log aus und die Richtung der Batterie-Animation ist nun auch korrekt.

```
[site  ] DEBUG 2021/02/01 12:57:58 pv power: -632W
[site  ] DEBUG 2021/02/01 12:57:58 grid power: 33W
[site  ] DEBUG 2021/02/01 12:57:58 battery power: 2022W
[site  ] DEBUG 2021/02/01 12:57:58 battery soc: 29%
[site  ] DEBUG 2021/02/01 12:57:58 site power: 2055W
[lp-1  ] DEBUG 2021/02/01 12:57:58 charge power: 0W
[lp-1  ] DEBUG 2021/02/01 12:57:58 charger status: A
```

andig, 2021-02-01:
Super, dann machen wir hier zu.

closed 2021-02-01 by andig

premultiply, 2021-02-03:
PV-Power ist bei dir aber noch falsch. Die müsste positiv sein.

rivengh, 2021-02-03:
@premultiply Stimmt. Vielen Dank für deinen Hinweis.

rivengh, 2021-02-04:
Habe gerade den Hinweis von @premultiply ausprobiert und "scale: -1 # reverse direction" bei meinem PV Meter in der Config entfernt, so dass PV Power nun positiv ist.

Allerdings ist das Sonnen-Symbol bei Erzeugung nun nicht mehr blau sondern grau.
Laut assets/js/components/SiteDetails.vue hätte ich daher eher angenommen, dass hier ein negativer Wert erwartet wird:

```
<div class="col-6 col-md-3 mt-3" v-if="pvConfigured">
	<div class="mb-2 value">
		Erzeugung
		<fa-icon
			icon="sun"
			:class="{
				'text-primary': pvPower < 0,
				'text-muted': pvPower >= 0,
			}"
		></fa-icon>
	</div>
	<h2 class="value">
		{{ fmt(pvPower) }}
		<small class="text-muted">{{ fmtUnit(pvPower) }}W</small>
	</h2>
</div>
```
@andig: Ist nun positiv oder negativ korrekt für die PV Power?

premultiply, 2021-02-04:
Guter Punkt!
Machst du ein neues Issue dafür auf?

rivengh, 2021-02-04:
Alles klar, Danke.
Ja, mache gleich ein neues Issue auf.