evcc-io/evcc #635: Charging always starts, regardless of mode or threshold

martink173, 2021-02-01:
**Describe the bug**
enable/threshold is set to -1000, disable threshold is set to 400, but charging starts regardless of the condition immediately after plugging vehicle in.
In the log files, I see "charger out of sync" after plugging in the vehicle.
I am using NRGKick charger via Bluetooth, and a Zoe. Both seem to be connected, as I see current information.
At the time of collecting the log below, PV power was insufficient, but it started charging immediately.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the vehicle to NRGKick charger, which is being controlled by EVCC.
2. Check the vehicle dashboard and log file

**Expected behavior**
I expect the vehicle to charge in PV mode only when the conditions are met

**EVCC details:**
Show output of <code>evcc -v</code>:
```
Jan 31 13:13:55 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:13:55 charge power: 0W
Jan 31 13:13:57 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:13:57 charger status: A
Jan 31 13:14:05 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:05 charge power: 0W
Jan 31 13:14:07 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:07 charger status: A
Jan 31 13:14:15 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:15 charge power: 0W
Jan 31 13:14:17 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:17 charger status: A
Jan 31 13:14:25 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:25 charge power: 0W
Jan 31 13:14:27 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:27 charger status: B
Jan 31 13:14:30 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:30 charge start energy: 252kWh
Jan 31 13:14:30 charger evcc[2908]: [lp-1  ] INFO 2021/01/31 13:14:30 car connected
Jan 31 13:14:33 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:33 vehicle soc: 78%
Jan 31 13:14:33 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:33 vehicle range: 203km
Jan 31 13:14:35 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:35 max charge current: 0A = 0A + -3.3A (2310W @ 3p)
Jan 31 13:14:35 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:35 pv max charge current: 0A
Jan 31 13:14:37 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:37 charge power: 0W
Jan 31 13:14:42 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:42 charger status: C
Jan 31 13:14:44 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:44 charge start energy: 252kWh
Jan 31 13:14:44 charger evcc[2908]: [lp-1  ] INFO 2021/01/31 13:14:44 start charging ->
Jan 31 13:14:44 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:44 vehicle soc: 78%
Jan 31 13:14:44 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:44 vehicle range: 203km
Jan 31 13:14:46 charger evcc[2908]: [lp-1  ] WARN 2021/01/31 13:14:46 charger out of sync
Jan 31 13:14:54 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:54 detected phases: 3p [6.58 6.56 6.59]A
Jan 31 13:14:54 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:54 max charge current: 0A = 0A + -6.3A (4377W @ 3p)
Jan 31 13:14:54 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:54 pv max charge current: 0A
Jan 31 13:14:56 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:14:56 charge power: 1340W
Jan 31 13:15:00 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:00 charger status: C
Jan 31 13:15:00 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:00 vehicle soc: 78%
Jan 31 13:15:00 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:00 vehicle range: 203km
Jan 31 13:15:02 charger evcc[2908]: [lp-1  ] WARN 2021/01/31 13:15:02 charger out of sync
Jan 31 13:15:11 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:11 detected phases: 3p [6.59 6.58 6.6]A
Jan 31 13:15:11 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:11 max charge current: 0A = 0A + -5.7A (3950W @ 3p)
Jan 31 13:15:11 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:11 pv max charge current: 0A
Jan 31 13:15:13 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:13 charge power: 1340W
Jan 31 13:15:17 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:17 charger status: C
Jan 31 13:15:17 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:17 vehicle soc: 78%
Jan 31 13:15:17 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:17 vehicle range: 203km
Jan 31 13:15:19 charger evcc[2908]: [lp-1  ] WARN 2021/01/31 13:15:19 charger out of sync
Jan 31 13:15:28 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:28 detected phases: 3p [6.59 6.54 6.58]A
Jan 31 13:15:28 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:28 max charge current: 0A = 0A + -5.9A (4072W @ 3p)
Jan 31 13:15:28 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:28 pv max charge current: 0A
Jan 31 13:15:30 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:30 charge power: 1300W
Jan 31 13:15:34 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:34 charger status: C
Jan 31 13:15:34 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:34 vehicle soc: 78%
Jan 31 13:15:34 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:34 vehicle range: 203km
Jan 31 13:15:36 charger evcc[2908]: [lp-1  ] WARN 2021/01/31 13:15:36 charger out of sync
Jan 31 13:15:44 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:44 detected phases: 3p [6.59 6.57 6.58]A
Jan 31 13:15:44 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:44 max charge current: 0A = 0A + -5.8A (4000W @ 3p)
Jan 31 13:15:44 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:44 pv max charge current: 0A
Jan 31 13:15:47 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:47 charge power: 1300W
Jan 31 13:15:51 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:51 charger status: B
Jan 31 13:15:53 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:53 final charge energy: 0kWh
Jan 31 13:15:53 charger evcc[2908]: [lp-1  ] INFO 2021/01/31 13:15:53 stop charging <-
Jan 31 13:15:53 charger evcc[2908]: [lp-1  ] DEBUG 2021/01/31 13:15:53 vehicle soc: 78%
[... 60 lines cut ...]
I configured according to the example for renault

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>
<- paste here

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

premultiply, 2021-02-02:
Eins der Probleme beim NRGkick ist, dass er den Status der Ladesperre beim Anstecken zurücksetzt.
Um das zu Umschiffen gibt es eigentlich einen Workaround im Code, der den Ladevorgang kurz nach dem Ladestart wieder pausiert.

Da der NRGkick auch noch extrem träge ist, ist es wichtig, dass du das Regelintervall sehr hoch einstellst. Bei mir sind z. B. 18 Sekunden notwendig. Sonst macht das System Mist weil es dauernd Einstellungen ändert die der NRGkick noch nicht übernommen hat oder umgekehrt sagt er hätte sie übernommen es aber in Wirklichkeit abgebrochen hat.

Also: Erster Test: Intervall hochsetzen und nochmal probieren. Die ZOE sollte in der Situation nach dem Anstecken eine kurze Zeit lang laden und dann pausieren.

Übrigens ist es keine allzu kluge Idee eine ZOE 3-phasig mit PV-Überschuss laden zu wollen...

martink173, 2021-02-03:
Danke für diesen Hinweis, das probiere ich aus. Wir reden hier also von der Einstellung für "interval" in der Yaml Datei ganz oben.

3-phasig Überschuss, warum nicht?

Cruiserlive, 2021-02-03:
Ich vermute es geht direkt um Nrgkick ohne connect ? Leider bekomme jetzt ich trotz intensiven Versuchen keine Verbindung via Bluetooth (timeout) . Wie bekommt man die korrrekte  Hex Mac Adresse raus ? Hab schon erfolglos 4-5 Scanner installiert. Danke 🙏

premultiply, 2021-02-03:
@martink173 Ja, interval in der Konfig-Datei ganz oben.
3P braucht zu viel Leistung die die Heim-PV i.d.R. nicht ausreichend liefert. Den größten Teil verschenkst du damit ins Netz. Ausserdem ist die Regelung um den Faktor 3 gröber und speziell bei der ZOE sind niedrige 3P-Leistungen besonders ineffizient - wenn der Ladevorgang überhaupt gestartet wird und auch dauerhaft weiterläuft.

@Cruiserlive Standardtools...
```
sudo bluetoothctl
scan on
```

martink173, 2021-02-03:
Bluetoothctl hat mir bei die MAC Adresse korrekt ausgegeben.

Ich hab den Wert für interval auf 20 Sekunden gestellt und dann das Fahrzeug verbunden. Im Log sehe ich dass er schon 8sek nach der ersten Meldung (charge power) zum ersten Mal "out of sync" sagt.

premultiply, 2021-02-03:
Wie sieht die Konfig aus?

martink173, 2021-02-03:
`uri: 0.0.0.0:7070 # uri for ui
interval: 20s # control cycle interval

# log settings
log: debug # error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- type: default
  name: volkszaehler_pv
  power:
      type: http
      uri: http://xxx
      jq: .data.tuples[0][1]
- type: default
  name: volkszaehler_grid
  power:
      type: http
      uri: http://xxx
      jq: .data.tuples[0][1]
  #- name: grid
  #  type: modbus
  #  model: sdm # SDM630
  #  uri: rs485.fritz.box:23
  #  rtu: true # rs485 device connected using ethernet adapter
  #  id: 2
  #  power: Power # default value, optionally override
  #  energy: Sum # default value, optionally override
  #- name: pv
  #  type: ...
  #- name: battery
  #  type: ...
  #- name: charge
  #  type: ...

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: nrgkick
  type: nrgkick-bluetooth
  macaddress: xx:xx:xx:xx:xx:xx 
  pin: 0000
  #- name: wallbe
  #  type: wallbe # Wallbe charger
  #  uri: 192.168.0.8:502 # ModBus address
  #- name: keba
  #  type: ...

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
- type: default
  name: xxx
  title: xxx
  capacity: xx # kWh
  charge:
          type: script
          cmd: /bin/sh -c "echo 50"
          timeout: 3s
  cache: 5m

# site describes the EVU connection, PV and home battery
site:
  title: xxx # display name for UI
  meters:
    grid: volkszaehler_grid # grid meter
    pv: volkszaehler_pv # pv meter
    #battery: battery # battery meter
  prioritySoC: 60 # give home battery priority up to this soc (0 to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Carport # display name for UI
  charger: nrgkick # charger
  meters:
    charge: # charge meter
  vehicle: xxx
  # vehicles: # use if multiple vehicles allowed to charge on this loadpoint
  # - ID.3
  # - e-Up
  mode: pv
  soc:
    # polling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    estimate: false # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 30
    - 50
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge to 100%
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: -1000 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 3m # threshold must be exceeded for this long
    threshold: 400 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 7 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

# mqtt message broker
mqtt:
  # broker: localhost:1883
  # topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>`

premultiply, 2021-02-03:
Bitte mal als Code formatieren, da kann man ja nix erkennen!

martink173, 2021-02-03:
> Bitte mal als Code formatieren, da kann man ja nix erkennen!
> Stimmt die Pin überhaupt? 0000 ist kein gültiger Code. Führende Nullen weglassen.
Ich hab zu voreilig abgesendet, sorry ;)

Verbindung zum Charger funktioniert grundsätzlich, so weit ich das erkennen kann. Aber ich lass das mal weg wenn du sagst.

martink173, 2021-02-03:
> Bitte mal als Code formatieren, da kann man ja nix erkennen!
> Stimmt die Pin überhaupt? 0000 ist kein gültiger Code. Führende Nullen weglassen.
Wenn ich das rausnehme sagt er "invalid pin". Mit 0000 funktioniert die Verbindung.

premultiply, 2021-02-03:
Die Pin muss mit der App übereinstimmen.
Mit falscher Pin kann man nur lesen aber nix schreiben was dem Verhalten entsprechen würde.
Daher: In der App eine gültige Pin setzen (nicht 0000) und diese in EVCC ohne führende Nullen konfigurieren.
EVCC muss währenddessen gestoppt sein!

WICHTIG: Die App danach schließen und nicht mehr verwenden! Es dürfen sich keine zwei Systeme gleichzeitig mit dem NRGkick per Bluetooth verbinden!

martink173, 2021-02-03:
Dass das Gerät nicht von 2 Punkten aus kontrolliert werden kann, ist klar (die App findet das Gerät beim Scan gar nicht wenn die Verbindung aktiv ist).

Das mit dem PIN teste ich noch! Danke erstmal.

martink173, 2021-02-03:
@premultiply Hab den PIN jetzt geändert und das Verhalten scheint jetzt so zu sein wie von dir beschrieben, also anstecken, dann wird kurz geladen, dann je nach PV Strom Verfügbarkeit Laden oder Pausieren. Vielen Dank für die Hilfe!

premultiply, 2021-02-03:
Wunderbar, dann kann hier bitte zu @andig

closed 2021-02-03 by andig

Cruiserlive, 2021-02-03:
> @martink173 Ja, interval in der Konfig-Datei ganz oben.
> 3P braucht zu viel Leistung die die Heim-PV i.d.R. nicht ausreichend liefert. Den größten Teil verschenkst du damit ins Netz. Ausserdem ist [...]

So einfach 👍😚. Mein altes MacBook scheint vom Bluetooth relativ schwach zu sein. Mit einem alten Windowsrechner von meinem Sohn klappt es mittlerweilen 👌. Ladentest folgt morgen. Leider kann ich meinen Wechselrichter SMA tripower 10k von 2012 (nur Bluetooth zum SHM 1.0) nicht einbinden . Kein TCP, modbus usw. oder gibt es hier eine Alternative ? Die Lösung mit HMA hab ich gelesen , aber ich brauch im Ziel eine Lösung für zwei Elektroautos mit unterschiedlichen Ladegeräten (nrkkick) & (innogy Ebox via OCPP)...schönen Abend ps: danke nochmal , erster Durchstich 😎

premultiply, 2021-02-04:
Warum soll der kein Modbus können?
Frag doch mal bei SMA nach. Viele ältere WR haben Modbus RTU-Schnittstellen oder können zumindest nachgerüstet werden.
Da gab es teilweise auch optionale Zusatzboards.

EVCC kann auch mit mehreren Ladepunkten umgehen und es lassen sich auch über die Plugin-Funktion auch die dämlichsten Schnittstellen anbinden.
Warum kaufst du denn auch so „komische“ Gerätschaften? Es gibt doch so viele einfache und robuste Wallboxen mit gescheiten Schnittstellen auf dem Markt...

goebelmeier, 2021-02-04:
> Warum soll der kein Modbus können?
> Frag doch mal bei SMA nach. Viele ältere WR haben Modbus RTU-Schnittstellen oder können zumindest nachgerüstet werden.
Vielleicht findet sich die Antwort schon dort: https://www.sma-sunny.com/servicetipp-welche-sma-wechselrichter-unterstuetzen-modbus-2/

Falls nicht, ist das ja nicht dramatisch, es reicht ja ein auslesbares Gridmeter. Das Auslesen des WR ist ja nur ein optionales Sahnehäubchen.

Cruiserlive, 2021-02-04:
Der Link war hilfreich und anscheinend waren die STP 10000TL-10 modbus fähig bzw. hatten eine RS485 Modul. Er ist allerdings nicht im Netzwerk sichtbar sondern nur via Bluetooth mit dem SHM 1.0 verbunden welcher im Netz zumindestens sichtbar wäre mit ID 2

Gebe dir recht mit komisch, wenn auch ein Technikbomber 😚. Lief über meinen AG welcher den Strom für den Dienstwagen bezahlt (im Büro geht es auch um ein zukünftiges Lastmgmt). Mir hatte der Gedanke gefallen später mal einen öffentlichen Ladepunkt aufzumachen mit Abrechnung. Wie auch immer , das OCPP Interface könnte nochmal interessant werden. Ob man mit dem Interface einen “privaten” Ladevorgang starten kann würde ich mal evaluieren für einen moeglichen EVCC Durchstich. Die APP unterstützt leider nur “geschäftliche” Ladevorgänge. Leider haben wir zu Hause gerade Baustelle 2.0 und mein COO hat ständig neue Ideen 😎