evcc-io/evcc #638: Missing accountID

martink173, 2021-02-02:
**Describe the bug**
The log says:
Feb  2 11:18:16 charger evcc[953]: [main  ] FATAL 2021/02/02 11:18:16 cannot create vehicle 'renault': cannot create type 'renault': missing accountID

**To Reproduce**
Steps to reproduce the behavior:
1. Starting EVCC, it fails
2. 
**Expected behavior**
A clear and concise description of what you expected to happen.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
<- paste here

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
vehicles:
- type: renault
  name: renault
  title: Zoe
  capacity: 42 # kWh
  user: <my username>
  password: <my password>
  vin: <my vin>
  region: de_AT
  cache: 5m

</pre></details>

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>
<- paste here

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

martink173, 2021-02-02:
Update: Mehr Infos unter --log trace hinzugefügt.

thierolm, 2021-02-02:
Hi, das Thema ist schon in bug #636 adressiert! :-)

closed 2021-02-02 by andig

tkempf, 2021-02-02:
Das Problem hierbei ist, dass der EVCC nach Ausgeb der Fehlermeldung gar nicht mehr startet, was bedeutet, dass das System komplett ausfällt. Ich wollte heute früh ein Update von 36 auf 42 machen, aber den EVCC bekomme ich so gar nicht mehr gestartet. 
m.E. wäre es sinnvoll den Fix aus  #636 in eine Zwischenrelease zu veröffentlichen, damit man wenigstens nicht ganz ohne EVCC dahsteht...

tkempf, 2021-02-02:
Hey Andi, sehe gerade die Release 0.43. Du bist echt schneller als die Polizei erlaubt, Danke!
Ich trau mich's ja schon fast nicht schreiben, aber könntest du die kompilierten assets auch bereitstellen, für Leute ohne GO-Environment ... Das wär echt super

tkempf, 2021-02-02:
So, mit dem neuen 0.43 ist der Fehler weg und EVCC läuft wieder und zeigt auch den SOC wieder perfekt an. 
Vielen Dank für den schnellen Fix und die binaries !!!

andig, 2021-02-02:
Die compilierten Assets sind beim Release dabei, ab Go 1.16 in ein paar Tagen auch ohne Release 👍🏻