evcc-io/evcc #643: Wrong color of sun icon in site details

rivengh, 2021-02-04:
**Describe the bug**
The sun icon next to "Erzeugung" under site details is:
* colored grey, when PV Power has a positive value and (see image)
* colored blue, when PV Power has a negative value.

[image]

See assets/js/components/SiteDetails.vue:

```
<div class="col-6 col-md-3 mt-3" v-if="pvConfigured">
	<div class="mb-2 value">
		Erzeugung
		<fa-icon
			icon="sun"
			:class="{
				'text-primary': pvPower < 0,
				'text-muted': pvPower >= 0,
			}"
		></fa-icon>
	</div>
	<h2 class="value">
		{{ fmt(pvPower) }}
		<small class="text-muted">{{ fmtUnit(pvPower) }}W</small>
	</h2>
</div>
```

**Expected behavior**
As the value of PV Power should be positive when power is generated, the sun icon shall be colored as follows:
* grey, when PV Power has a negative value (<0) and
* blue, when PV Power has a positive value (>=0).

**EVCC details:**
```
0.42
```

**Debug Log**
```
[site  ] DEBUG 2021/02/04 08:38:03 ----
[site  ] DEBUG 2021/02/04 08:38:03 pv power: 47W
[site  ] DEBUG 2021/02/04 08:38:03 grid power: 167W
[site  ] DEBUG 2021/02/04 08:38:03 battery power: -6W
[site  ] DEBUG 2021/02/04 08:38:03 battery soc: 0%
[site  ] DEBUG 2021/02/04 08:38:03 site power: 161W
[lp-1  ] DEBUG 2021/02/04 08:38:03 charge power: 0W
[lp-1  ] DEBUG 2021/02/04 08:38:03 charger status: A
```

**evcc.yaml / PV meter**
```
- name: pv
  type: default
  power: # power reading
    type: mqtt # use mqtt plugin
    topic: iobroker/sonnen/0/status/production # mqtt topic
    timeout: 30s # don't use older values
```

andig, 2021-02-04:
> As the value of PV Power should be positive when power is generated, the sun icon shall be colored as follows

That's an assumption. Currently, negative PV is assumed to be generation (similar to negative value at grid meter being export).

Ich glaube ich muss das nochmal dokumentieren, ggf. ist es aktuell auch noch nicht vollständig konsistent.

premultiply, 2021-02-04:
Der Standardweg ist positiv. PV-WR lieferen nie negative Werte.
Nur Batterie-WR (Batterie laden) und Grid (Netzeinspeisung) können negativ werden.

andig, 2021-02-04:
> Der Standardweg ist positiv. PV-WR lieferen ja nie negative Werte. Nur Batterie-WR (Batterie laden) und Grid (Netzeinspeisung) können negativ werden.

Damit sind beide unterschiedlich. Auf einen Weg müssen wir uns festlegen. 

So ists heute:

[image]

premultiply, 2021-02-04:
Ich halte es auf Grund der Erfahrungen mit neuen Anwendern für quasi obligatorisch dass die Standardwerte die die Geräte liefern der gemeinen Erwartungshaltung entsprechen. Sprich: z. B. die PV-WR liefern im Standardfall "Sonne scheint" automatisch positive Werte wenn man sie als Modbus-Device hinzufügt. Also ist das der "richtige" Weg. Sonst müsste man jedem Anwender erklären: "Nein, du muss erstmal noch manuell deine PV-Produktion invertieren".

Die meisten Anwender werden das auch schon so (positiv) konfiguriert haben. Da die PV-Produktion aber für die Berechnung quasi irrelevant ist ist das nie weiter aufgefallen. Bei mir ist das auch so, aber ich hab ehrlich gesagt auch nie auf die Farbe des Symbols geachtet.

goebelmeier, 2021-02-04:
Ich musste den Wert auch extra invertieren, damit er wie gewünscht negativ ist. Ich denke, es gibt wenig bis keine WR die Erzeugung per Default als negativen Wert ausliefern.

rivengh, 2021-02-04:
Damit wären auch die Beispiele für die PV Meter unter https://github.com/andig/evcc-config#meters nicht korrekt, da hier grundsätzlich nicht invertiert wird.

andig, 2021-02-04:
> Ich halte es auf Grund der Erfahrungen mit neuen Anwendern für quasi obligatorisch dass die Standardwerte die die Geräte liefern der gemeinen Erwartungshaltung entsprechen.

Erstmal: das PV Problem ist rein optisch. Funktional irrelevant, Ursache/Wirkung schnell zu identifizieren. Können wir dennoch ändern falls notwendig.

Zweitens: die Batterie WR scheinen- soweit ich die Diskussion bisher verstanden habe- keinem Standard bzgl. +/- zu folgende.

> Was meinst du mit +Eigenverbrauch unter dem WR?

Restleistungsbezug falls der WR keine Erzeugung hat.

Warum sind die Vorzeichen wie sie sind? Weil ich über die rechte Seite eine Summe ziehen kann um auf die linke Seite zu kommen. Was ist euer konkreter Vorschlag? Bzgl. PV? Bzgl. Batterie? Bzgl. Hybrid WR?

premultiply, 2021-02-04:
Mein (bzw. auch "unser" Vorschlag):
Leistung die WR ins Hausnetz einspeisen: positiv.
Leistung die WR/Netz aus dem Hausnetz aufnehmen (Einspeisen/Batterie laden/Eigenverbrauch des Geräts im Standby): negativ.
Alles andere (Verbraucher, insb. Auto laden) auch positiv.

Gegenüber deiner Übersicht wäre somit wirklich nur das Vorzeichen bei den PV-WR zu drehen und damit die Farbgebung des Symbols zu invertieren bzw. ggf. der seltene Fall das EVCC ohne Gridmeter aus der PV-Erzeugung gesteuert wird nochmal zu validieren und ggf. zu invertieren.

premultiply, 2021-02-04:
Bei Hybrid-WR ist analog zu verfahren.
Die meisten Modelle mit denen ich jetzt zu bekommen habe bilden die PV-Erzeugung nur sinnvoll auf den DC-MPPs hab. Die sind ebenfalls positiv. Batterie laden/entladen liegt dann auch oft auf weiteren MPPs getrennt. Davon muss man dann "laden" manuell invertieren weil die Werte immer positiv sind.
Wenn ein Hybrid-WR auch eine sinnvolle AC-Leistung ausgibt dann ist dieser bei Erzeugung (also Leistungsfluss zum Hausnetz hin) meiner bisherigen Erfahrung nach auch positiv.

andig, 2021-02-04:
Würde sich ein Freiwiliger finden, der mal alle Modelle in evcc-config durchflöht und einen Sammel-PR auf die yamls+make erstellt?

premultiply, 2021-02-04:
Kann ich probieren. Kann aber einen Moment dauern. Aber bitte hilf mir wenn ich irgendwas falsch machen sollte.

andig, 2021-02-04:
Das wäre perfekt. Ich könnte den ersten Aufschlag für core machen, dann config und dann mergen wir alles gleichzeitig. 
Wichtig: NUR yaml ändern + make. README etc entstehen dann automatisch.

andig, 2021-02-04:
> Leistung die WR/Netz ins Hausnetz einspeisen (PV-Erzeugung/Batterie entladen/Netzbezug): positiv.
> Leistung die WR/Netz aus dem Hausnetz aufnehmen (Einspeisen/**Batterie laden**/Eigenverbrauch des Geräts im Standby): **negativ**.

> Die meisten Modelle mit denen ich bis jetzt zu bekommen habe bilden die PV-Erzeugung nur sinnvoll auf den DC-MPPs hab. Die sind ebenfalls positiv (und vorzeichenlos). Batterie laden/entladen liegt dann auch oft auf weiteren MPPs getrennt davon. Davon muss man dann "**Batterie laden**" jeweils **manuell invertieren** weil die Werte eben immer positiv/vorzeichenlos sind.

Du widersprichst Dir bei der Batterie? 

Ich habe jetzt folgendes verstanden:
- [ ] Verbraucher (Residual etc): + (keine Änderung/ as-is)
- [ ] Netz (Grid): Bezug + Einspeisung - (keine Änderung/ as-is)
- [ ] PV (Inverter): Erzeugung + Eigenverbrauch - (TODO: Kennzeichen/UI invertieren)
- [ ] Batterie und Hybrid-WR (Battery): Laden - Entladen + (TODO: Kennzeichen/UI invertieren)

Bei beiden TODO Punkten müssen wir damit die Konfiguration und Logik anpassen.

premultiply, 2021-02-04:
Nach deinem Bild oben ist aber die Batterie auch schon heute "korrekt".

rivengh, 2021-02-04:
>     * [ ]  Batterie und Hybrid-WR (Battery): Laden - Entladen + (TODO: Kennzeichen/UI invertieren)

Das ist doch aktuell schon so implementiert, oder nicht? Also Laden - und Entladen +.

andig, 2021-02-04:
Stimmt: https://github.com/andig/evcc/issues/631#issuecomment-770628690

> Batterie laden: negativ
> Batterie entladen: positiv

Oben aktualisiert.

premultiply, 2021-02-04:
Ich hab jetzt soweit mal alle Beispiele in evcc-config durchgesehen aber musste da nichts tun. Dahingehend ist dort alles schon korrekt.

andig, 2021-02-07:
Mit https://github.com/andig/evcc/commit/84bf5feb69ed4836ad3ecdf100bd3a42ebe5ecf4 erledigt.

closed 2021-02-07 by andig